dorssel/usbipd-win #280: Machine interface to usbipd

d0n13, 2022-02-22:
We would like to be able to access usbipd via a software mechanism. Would anybody be interested in us adding a PR for a JSON interface to the commands instead of listing stuff directly to the console as raw text?

It's difficult to parse when trying to access the software from an app. 

Any other preferred mechanisms that people would use instead of JSON?

Thanks
Donie

d0n13, 2022-02-22:
Is it public?

dorssel, 2022-02-22:
Not yet. And far from complete. 2-3 weeks (plus the usual overconfidence factor).

dorssel, 2022-02-22:
PS: I will commit to a PoC branch later this week.

dorssel, 2022-03-06:
@d0n13 
Can you try PR #289 to see if it fits your needs? See description of PR. The artifact is at https://github.com/dorssel/usbipd-win/actions/runs/1942363841.

d0n13, 2022-03-07:
I'm trying to figure out how to use this in my own app. Can you give an examples of use? Or is it tied to powershell?

dorssel, 2022-03-07:
See PR description. Just run `usbipd state` and you will get JSON. You can look at the PowerShell automation object source on how to interpret the different fields.

d0n13, 2022-03-07:
I'll try to get around to doing this sometimes this week. This will make my software simpler if it works nicely...

dorssel, 2022-03-16:
@d0n13 
Did you have a chance to look at this? I will be releasing a new version soon and I want this merged.

d0n13, 2022-03-16:
@dorssel Yes, I just looked at it and compared it to my code to see was anything missing that I might need.

It seems I could rewrite my code and replace all of the USB stuff I do (mostly copied from your code) and replace it with this call. 

The bind and attach are already using your  usbipd command directly so I can use the "usbipd state" in the same way. 

Thank you for doing this. Can you let me know when it's merged and the other changes done recently are merged together as well?

Maybe 2.2 release soon?

dorssel, 2022-03-16:
@d0n13 
Yes, 2.2 is near. Maybe over the weekend (no promises, there are still some minor fixes I want to include).

closed 2022-03-16 by dorssel

DeadlyChambers, 2022-03-17:
I was able to hit usbipd directly from the admin wsl cmd prompt. I was able run the win.exe directly from the wsl like thi
`/mnt/c/Program\ Files/usbipd-win/usbipd.exe list`
[image]

You might want to double check the  service is running.
[image]

PR #289 (2022-03-16): Add PowerShell cmdlets
This PR adds helpers for scripted automation:

1. There is a new command `usbipd state`, which returns JSON for all USB devices. More easily parsed by scripts than the standard `list` output, with more information and without truncating long strings.
2. There is a PowerShell module that exposes a `Get-UsbipdDevice` cmdlet that uses the JSON from (1) and returns it as PowerShell objects, with even more (meta-)information like `IsConnected`, `IsAttached`, etc.

The PowerShell cmdlet is built on top of PowerShellStandard, so it can be used by standard Windows 10 PowerShell  [... cut]