dorssel/usbipd-win #283: [Question] Is there a way to set up a default WSL2 distribution for usbipd-win?

akuropka, 2022-02-24:
I know there's an option to define to which distribution the device should be attached by using the `--distribution <NAME>` option. If this option is not used the default WSL distribution will be used.

Is there a way to define a default distribution for `usbipd-win` which is different to the default distribution in `WSL` to avoid always using `--distribution <NAME>` but keeping the defaults for WSL?

dorssel, 2022-02-24:
Sorry, no. And I don't think there ever will be.

For PowerShell you can make an Alias, see https://docs.microsoft.com/en-us/powershell/module/microsoft.powershell.utility/set-alias?view=powershell-7.2, Example 5

akuropka, 2022-02-24:
Unfortunate… but thanks for clarification.
PS is not an option in this case.

closed 2022-02-24 by akuropka

dorssel, 2022-02-24:
@akuropka 
You can always use a good old .BAT file...

akuropka, 2022-02-24:
Yeah, will be anything like that.
All fine, was just a question. 😉