dorssel/usbipd-win #286: usbipd: warning: The VBoxUsbMon driver is currently not running; a reboot should fix that.

abrhambas01, 2022-02-27:
when issuing a 
`$  usbipd wsl list`

I get this warning, is this some kind of an issue or what ?

[image]

abrhambas01, 2022-02-27:
@dorssel  no output sir.

dorssel, 2022-02-27:
there is always some output, even if it is an error. for example:
```
C:\> sc query VBoxUsbMon

SERVICE_NAME: VBoxUsbMon
        TYPE               : 1  KERNEL_DRIVER
        STATE              : 4  RUNNING
                                (STOPPABLE, NOT_PAUSABLE, IGNORES_SHUTDOWN)
        WIN32_EXIT_CODE    : 0  (0x0)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x0
```

abrhambas01, 2022-02-27:
I get this when I run it in windows terminal 
```
[SC] EnumQueryServicesStatus:OpenService FAILED 1060:

The specified service does not exist as an installed service.

```

dorssel, 2022-02-27:
How did you install `usbipd-win`? msi or winget? was there an error?
Please do a repair install of `usbipd-win`, or uninstall/reinstall.

dorssel, 2022-02-27:
you need to have a command prompt open in wsl. like it says: the wsl instance must be running.

abrhambas01, 2022-02-27:
I think I got it right now thanks @dorssel

dorssel, 2022-02-27:
once the device is attached, usbipd is done. the rest is linux stuff.

abrhambas01, 2022-02-27:
yep , i think they're already attached based on what is said here

[image]

so can I go now to /mnt/e ?

dorssel, 2022-02-27:
You are confusing windows/linux for the drive. Either you use it as a windows drive (no need for usbipd-win) and use it as E: or /mnt/e. Or attach it as a USB device to Linux, in which case it is no longer accessible under Windows and you will have to `sudo mount` it under Linux yourself.

Are you sure you selected the right tool for what you are trying to do? In fact, what exactly is it you are tryin to accomplish?

closed 2022-03-12 by dorssel

GimpMaster, 2024-05-02:
I had this same issue just happen. I upgraded from usbipd 2.X and 4.X. Repair did not work. I had to uninstall, then install again and now i don't get the VboxUsbMon missing error.

Zazmi2, 2024-09-26:
Hi I am getting the same error. I have tried rebooting it, uninstalling and reinstalling it. I am able to bind the device but not attach when doing this: usbipd attach --wsl --busid 1-4 and I get no output when running sc query VBoxUsMon.

I am using a surface pro laptop Windows 11 arm64 processor and I am using WSl2.

[image]

JPHutchins, 2025-06-10:
I got this after upgrading from 4.0.0 -> 5.1.0 (using `winget upgrade` and then restarting as instructed).

```
sc query VBoxUsbMon
[SC] EnumQueryServicesStatus:OpenService FAILED 1060:

The specified service does not exist as an installed service.
```

Resolution matches @JoshuaHintze's suggestion.
```
winget uninstall usbipd
Found usbipd-win [dorssel.usbipd-win]
Starting package uninstall...
Successfully uninstalled
```

```
winget install usbipd
Found usbipd-win [dorssel.usbipd-win] Version 5.1.0
This application is licensed to you by its owner.
Microsoft is not responsible for, nor does it grant any licenses to, third-party packages.
Downloading https://github.com/dorssel/usbipd-win/releases/download/v5.1.0/usbipd-win_5.1.0_x64.msi
  ██████████████████████████████  4.15 MB / 4.15 MB
Successfully verified installer hash
Starting package install...
The installer will request to run as administrator, expect a prompt.
Successfully installed
```

Rood75, 2026-01-06:
> Was facing the same issue, the fix for me was `winget uninstall usbipd` and `winget install usbipd` Somehow using the msi did not work for me. A clean install made it work!

Uninstalling the old MSI version and installing the latest version did the trick.

PR #299 (2022-03-12): Add VBoxUsbMon installation check
Fixes #286