dorssel/usbipd-win #291: SIM800C USB dongle cannot be used

ocalvo, 2022-03-05:
[SIM800C USB dongle](https://www.amazon.com/gp/product/B087Z6F953/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) timeouts when trying to access the device via gammu.

Requisites:
Windows 11 and Linux subsystem for Windows Preview

Repro steps:
1. Connect device
2. Run `usbipd wsl attach --busid=7-1`
3. In WSL run: `sudo chmod 777 /dev/ttyUSB0`
4. Run python test program:
```
import gammu
from gammu.asyncworker import GammuAsyncWorker
import asyncio
import logging
from sys import platform

import sys
import pprint
pp = pprint.PrettyPrinter(indent=4)

async def main():

    # Get the current event loop.
    loop = asyncio.get_running_loop()

    gammu.SetDebugFile(sys.stderr)
    gammu.SetDebugLevel('textall')

    config = dict(Device="/dev/ttyUSB0", Connection="at")

    state_machine0 = gammu.StateMachine()
    state_machine0.SetConfig(0, config)
    state_machine0.Init()
    state_machine = state_machine0
    status = state_machine.GetSMSStatus()
    print(f"Status :{status}")

    state_machine0.Terminate()
    return 0

if __name__ == '__main__':
    event_loop = asyncio.get_event_loop()
    try:
        event_loop.run_until_complete(main())
    finally:
        event_loop.close()
```

USBIP log before running gammu:
```
> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\oscar
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.18.120.30
dbug: UsbIpServer.Server[1000]
      connection closed
dbug: UsbIpServer.Server[1000]
      new connection from 172.18.120.30
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: UsbIpServer.ConnectedClient[1]
      Client 172.18.120.30 claimed device at 7-1 (USB\VID_1A86&PID_7523\A&392D36F6&0&1).
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=64, flags=512, length=64, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 64
trce: UsbIpServer.AttachedClient[1001]
      actual: 18, requested: 64
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=65, flags=512, length=18, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 18
trce: UsbIpServer.AttachedClient[1001]
      actual: 18, requested: 18
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=66, flags=512, length=9, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 512 0 9
trce: UsbIpServer.AttachedClient[1001]
      actual: 9, requested: 9
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=67, flags=512, length=39, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 512 0 39
trce: UsbIpServer.AttachedClient[1001]
      actual: 39, requested: 39
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=68, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 768 0 255
trce: UsbIpServer.AttachedClient[1001]
      actual: 4, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=69, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 770 1033 255
trce: UsbIpServer.AttachedClient[1001]
      actual: 22, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=70, flags=0, length=0, ep=0
dbug: UsbIpServer.AttachedClient[1000]
      Trapped SET_CONFIGURATION: 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 0
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=71, flags=512, length=2, ep=0
trce: UsbIpServer.AttachedClient[1001]
      192 95 0 0 2
trce: UsbIpServer.AttachedClient[1001]
      actual: 2, requested: 2
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=72, flags=0, length=0, ep=0
trce: UsbIpServer.AttachedClient[1001]
      64 161 0 0 0
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 0
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=73, flags=0, length=0, ep=0
trce: UsbIpServer.AttachedClient[1001]
      64 154 4882 45698 0
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 0
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=74, flags=0, length=0, ep=0
trce: UsbIpServer.AttachedClient[1001]
      64 154 9496 195 0
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 0
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=75, flags=0, length=0, ep=0
trce: UsbIpServer.AttachedClient[1001]
      64 164 65535 0 0
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 0
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=76, flags=512, length=2, ep=0
trce: UsbIpServer.AttachedClient[1001]
      192 149 5 0 2
trce: UsbIpServer.AttachedClient[1001]
      actual: 2, requested: 2
```
Log after running gammu:
```
```

dorssel, 2022-03-05:
@ocalvo 
Everything to endpoint 0 (the default control endpoint for any USB device) looks normal. But the requests to endpoint 1 never seems to complete.

Can you give some more details on the device configuration, perhaps the output of `usbview.exe` (part of Windows SDK)?
What is logged on the Linux side (syslog)?

ocalvo, 2022-03-05:
UsbView output:
```
[Port1]  :  USB-SERIAL CH340

Is Port User Connectable:         yes
Is Port Debug Capable:            no
Companion Port Number:            0
Companion Hub Symbolic Link Name: 
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "USB Serial"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: Full (is not SuperSpeed or higher capable)
Device Address:                    0x0D
Open Pipes:                           3

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0110
bDeviceClass:                      0xFF  -> This is a Vendor Specific Device
bDeviceSubClass:                   0x00
bDeviceProtocol:                   0x00
bMaxPacketSize0:                   0x08 = (8) Bytes
idVendor:                        0x1A86 = Nanjing Qinherg Electronics Co., Ltd.
idProduct:                       0x7523
bcdDevice:                       0x0264
iManufacturer:                     0x00
iProduct:                          0x02
     English (United States)  "USB Serial"
iSerialNumber:                     0x00
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x82  -> Direction: IN - EndpointID: 2
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0020 = 0x20 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x02  -> Direction: OUT - EndpointID: 2
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0020 = 0x20 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0008 = 0x08 bytes
bInterval:                         0x01

       ---===>Full Configuration Descriptor<===---

          ===>Configuration Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x02
wTotalLength:                    0x0027  -> Validated
bNumInterfaces:                    0x01
bConfigurationValue:               0x01
iConfiguration:                    0x00
bmAttributes:                      0x80  -> Bus Powered
MaxPower:                          0x31 =  98 mA

          ===>Interface Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x04
bInterfaceNumber:                  0x00
bAlternateSetting:                 0x00
bNumEndpoints:                     0x03
bInterfaceClass:                   0xFF  -> Interface Class Unknown to USBView
bInterfaceSubClass:                0x01
bInterfaceProtocol:                0x02
iInterface:                        0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x82  -> Direction: IN - EndpointID: 2
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0020 = 0x20 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x02  -> Direction: OUT - EndpointID: 2
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0020 = 0x20 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0008 = 0x08 bytes
bInterval:                         0x01
```

dorssel, 2022-03-11:
@ocalvo 
OK, so endpoint 0 (control, config, etc.) works fine.

Endpoint 1 is an interrupt endpoint. It is normal that a single request is made to it; that will only complete if the device has something to alert (which it apparently never does).

Endpoint 2 is bidirectional (it really is 2 endpoints, 0x2 and 0x82). All the way at the end of your log, that endpoint completes 2 requests.

All in all, it looks quite normal. Also: linux reports no errors. What is gammu? Is there any other software you can use to test basic functionality? Because from these logs, nothing appears to be wrong...

dorssel, 2022-03-16:
@ocalvo 
An issue that may be related to yours was just fixed in the master build. Could you give https://github.com/dorssel/usbipd-win/actions/runs/1991685584 a try?

ocalvo, 2022-03-16:
@dorssel Happy to report that the issue no longer repro with [this](https://github.com/dorssel/usbipd-win/actions/runs/1991685584) build.

ocalvo, 2022-03-17:
Fixed with #307

closed 2022-03-17 by ocalvo