dorssel/usbipd-win #295: Can't connect to Oculus Quest through WSL 2

ahmeda14960, 2022-03-09:
Hi all,

I've been trying to use usbip on wsl 2 and I ran into an issue I haven't been able to solve.  I'm running this on a laptop with Version 21H2 (OS BUILD 22000.527), and with the following  WSL configuration:

```
uname -a
Linux Omen 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

```

 I'm trying to connect to an Oculus Quest 2 and I can see that it's detected in Adminstrator Powershell

```
PS C:\windows\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
2-2    HD Pro Webcam C920                                            Not attached
2-3    Logitech USB Input Device, USB Input Device                   Not attached
2-4    Quest 2, Oculus Composite XRSP Interface, Oculus Composit...  Not attached
2-6    HP Wide Vision HD Camera, Camera DFU Device                   Not attached
2-7    Intel(R) Wireless Bluetooth(R)                                Not attached
2-8    USB Input Device                                              Not attached
```

However, when I try to attach to the Oculus I get the following error:

```
C:\windows\system32> usbipd wsl attach --busid 2-4
usbip: error: Attach Request for 2-4 failed - Device busy (exported)

usbipd: error: Failed to attach device with BUSID '2-4'.
```

What's strange is that I'm able to attach to other devices without issues:
```
PS C:\windows\system32> usbipd wsl attach --busid 2-6
PS C:\windows\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
2-3    Logitech USB Input Device, USB Input Device                   Not attached
2-6    HP Wide Vision HD Camera, Camera DFU Device                   Attached - Ubuntu
2-7    Intel(R) Wireless Bluetooth(R)                                Not attached
2-8    USB Input Device                                              Not attached
```

I made sure to allow file access to the Oculus from Windows so I'm not sure what's causing this.  Any advice?

dorssel, 2022-03-09:
@ahmeda14960 

> usbip: error: Attach Request for 2-4 failed - Device busy (exported)

This means the host (Windows) is already exclusively using the device. Possibly some software you installed for it, or a service, or maybe even the driver itself. It just means what it says: "device busy". If you do not intend to use the device on Windows at all, you can use `usbipd bind --force`.

ahmeda14960, 2022-03-09:
So I tried the command
```PS C:\windows\system32> usbipd bind --force --busid 2-4
usbipd: warning: A reboot may be required before the changes take effect. ```

And I rebooted WSL and windows and nothing has changed. Any more tips?

dorssel, 2022-03-09:
What is the output of `usbipd list` now?

ahmeda14960, 2022-03-09:
So it actually took a few minutes but I can see that the oculus is shared now 

```
usbipd list
Connected:
BUSID  DEVICE                                                        STATE
2-2    HD Pro Webcam C920                                            Not shared
2-3    Logitech USB Input Device, USB Input Device                   Not shared
2-4    Oculus Composite XRSP Interface, Oculus Composite ADB Int...  Shared (forced)
2-6    HP Wide Vision HD Camera, Camera DFU Device                   Shared
2-7    Intel(R) Wireless Bluetooth(R)                                Not shared
2-8    USB Input Device                                              Not shared

Persisted:
GUID                                  DEVICE
76427fb0-0856-49cf-8b36-922c74eb2531  Quest 2, Oculus Composite XRSP Interface, Oculus Composit...D
```

However, WSL2 is still not able to detect the Oculus. When switching over to WSL and running a command to check
```
(base) a@Omen:~$ adb devices
List of devices attached

```

It should be listing the name of the oculus after the command above, and it does so on another machine which is purely running Ubuntu without windows.

dorssel, 2022-03-09:
You have only shared (forced) the device without the host keeping it busy. You still need to attach, which should now succeed.

ahmeda14960, 2022-03-09:
I see, I just tried that and got the following output
```
PS C:\windows\system32> usbipd wsl attach --busid 2-4
usbipd: warning: A third-party firewall may be blocking the connection; ensure TCP port 3240 is allowed.
usbip: error: tcp connect
usbipd: error: Failed to attach device with BUSID '2-4'.
```

dorssel, 2022-03-09:
Is the error message not correct?

ahmeda14960, 2022-03-09:
Sorry about, just missed it!

I fixed the issue by killing a VPN that was running into the background, and I'm close to resolving the issue. Now when I run the commands you mentioned before I see the following on powershell

```
PS C:\windows\system32> usbipd bind --force --busid 2-4
usbipd: info: Device with busid '2-4' was already shared.
PS C:\windows\system32> usbipd wsl attach --busid 2-4
PS C:\windows\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
2-2    HD Pro Webcam C920                                            Not attached
2-3    Logitech USB Input Device, USB Input Device                   Not attached
2-4    Quest 2, Oculus Composite XRSP Interface, Oculus Composit...  Attached - Ubuntu
2-6    HP Wide Vision HD Camera, Camera DFU Device                   Not attached
2-7    Intel(R) Wireless Bluetooth(R)                                Not attached
2-8    USB Input Device                                              Not attached
```

Thanks for your help thus far! Unfortunately, what's strange is that even though wsl can detect the 
Oculus is attached, the command to check for adb devices still doesn't turn up any devices, with all the required software.  Any thoughts here?

```
a@Omen:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2833:0183 Oculus Quest 2
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
(base) a@Omen:~$ adb devices
List of devices attached

```

dorssel, 2022-03-09:
> PS C:\windows\system32> usbipd bind --force --busid 2-4
> usbipd: info: Device with busid '2-4' was already shared.

As you can see, this has to be done only once. This permanently (also across reboots) binds the device to usbipd. If you ever want to use the device on Windows again, you will have to manually `usbipd unbind` first.

You now have the device attached the device to WSL. `usbipd-win` has done its job. The rest is Linux stuff, which is outside the scope of `usbipd-win`. Other people have found ways to get `udev` working for `adb`. See for example https://github.com/dorssel/usbipd-win/issues/186#issuecomment-1007361794. Good luck!

closed 2022-03-26 by dorssel