dorssel/usbipd-win #297: HackRF One not attaching to WSL2 (Ubuntu)

fmitch, 2022-03-10:
After following the installation and instructions, my HackRF One USB device isn't attaching to WSL. I'm able to attach a normal USB drive, but the HackRF doesn't show up in `lsusb`. However both devices show up in the `usbipd wsl list` as attached to Ubuntu:

`PS C:\WINDOWS\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-2    HackRF One                                                    Attached - Ubuntu
1-3    ELAN WBF Fingerprint Sensor                                   Not attached
1-4    HP Wide Vision HD Camera                                      Not attached
2-1    Intel(R) Wireless Bluetooth(R)                                Not attached
2-2    USB Mass Storage Device                                       Attached - Ubuntu
3-2    Realtek USB GbE Family Controller                             Not attached
4-5    Realtek USB Audio                                             Not attached
4-7    USB Input Device                                              Not attached`

`~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 014: ID 13fe:1f23 Kingston Technology Company Inc. PS2232 flash drive controller
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

dorssel, 2022-03-10:
@fmitch 
Can you post syslog/dmesg from WSL instance at the time of attach?
Can you run the usbipd server with full logging, see https://github.com/dorssel/usbipd-win/wiki/Troubleshooting, and post the log?

fmitch, 2022-03-10:
From syslog:
```
Mar 10 14:56:38 DESKTOP-HTQC3FS kernel: [ 3450.140979] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
Mar 10 14:56:38 DESKTOP-HTQC3FS kernel: [ 3450.140983] vhci_hcd vhci_hcd.0: devid(65538) speed(3) speed_str(high-speed)
Mar 10 14:56:38 DESKTOP-HTQC3FS kernel: [ 3450.141046] vhci_hcd vhci_hcd.0: Device attached
Mar 10 14:56:39 DESKTOP-HTQC3FS kernel: [ 3450.482137] usb 1-1: new high-speed USB device number 19 using vhci_hcd
Mar 10 14:56:39 DESKTOP-HTQC3FS kernel: [ 3450.651754] usb 1-1: device descriptor read/64, error -62
Mar 10 14:56:39 DESKTOP-HTQC3FS kernel: [ 3450.921684] usb 1-1: device descriptor read/64, error -62
Mar 10 14:56:40 DESKTOP-HTQC3FS kernel: [ 3451.191967] usb 1-1: new high-speed USB device number 20 using vhci_hcd
Mar 10 14:56:40 DESKTOP-HTQC3FS kernel: [ 3451.341970] usb 1-1: device descriptor read/64, error -62
Mar 10 14:56:40 DESKTOP-HTQC3FS kernel: [ 3451.611643] usb 1-1: device descriptor read/64, error -62
Mar 10 14:56:40 DESKTOP-HTQC3FS kernel: [ 3451.731825] usb usb1-port1: attempt power cycle
Mar 10 14:56:40 DESKTOP-HTQC3FS kernel: [ 3451.921947] usb 1-1: new high-speed USB device number 21 using vhci_hcd
Mar 10 14:56:40 DESKTOP-HTQC3FS kernel: [ 3451.922081] usb 1-1: SetAddress Request (21) to port 0
Mar 10 14:56:40 DESKTOP-HTQC3FS kernel: [ 3451.958872] usb 1-1: device descriptor read/8, error -62
Mar 10 14:56:40 DESKTOP-HTQC3FS kernel: [ 3452.106490] usb 1-1: device descriptor read/8, error -62
Mar 10 14:56:41 DESKTOP-HTQC3FS kernel: [ 3453.181729] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
Mar 10 14:56:41 DESKTOP-HTQC3FS kernel: [ 3453.181949] usb usb1-port1: unable to enumerate USB device
```

From usbipd server logging:
```
PS C:\WINDOWS\system32> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\WINDOWS\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.22.117.243
dbug: UsbIpServer.Server[1000]
      connection closed
dbug: UsbIpServer.Server[1000]
      new connection from 172.22.117.243
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: UsbIpServer.ConnectedClient[1]
      Client 172.22.117.243 claimed device at 1-2 (USB\VID_1D50&PID_6089\000000000000000087C867DC2A605E5F).
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=82, flags=512, length=64, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 64
dbug: UsbIpServer.AttachedClient[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 64
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=83, flags=512, length=64, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 64
dbug: UsbIpServer.AttachedClient[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 64
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=84, flags=512, length=64, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 64
dbug: UsbIpServer.AttachedClient[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 64
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=85, flags=512, length=64, ep=0
trce: UsbIpServer.AttachedClient[1001]
[... 110 lines cut ...]
      USBIP_CMD_SUBMIT, seqnum=99, flags=512, length=8, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 8
dbug: UsbIpServer.AttachedClient[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 8
```

Just for kicks, here's the output from `usbip list --remote<WIN_IP>`
```
frost@DESKTOP-HTQC3FS:~$ usbip list --remote=172.22.112.1
Exportable USB devices
======================
 - 172.22.112.1
        1-2: OpenMoko, Inc. : Great Scott Gadgets HackRF One SDR (1d50:6089)
           : USB\VID_1D50&PID_6089\000000000000000087C867DC2A605E5F
           : (Defined at Interface level) (00/00/00)
           :  0 - Vendor Specific Class / Vendor Specific Subclass / Vendor Specific Protocol (ff/ff/ff)

        2-2: Kingston Technology Company Inc. : PS2232 flash drive controller (13fe:1f23)
           : USB\VID_13FE&PID_1F23\078809A32317
           : (Defined at Interface level) (00/00/00)
           :  0 - Mass Storage / SCSI / Bulk-Only (08/06/50)
```

dorssel, 2022-03-11:
@fmitch
That looks really bad:

> USBSUP_XFER_DNR -> ETIME -> -62

DNR == Did Not Respond.
Can you try `usbipd bind --force --busid 1-2` to see if that makes a difference?

PS: the reason that it still reported as attached is probably because Linux "hung up" on the TCP connection, and usbipd-win has a keep-alive timeout of 1 minute or so. So, after a while, the attached state should disappear. Can you confirm?

fmitch, 2022-03-11:
Binding the device didn't make any difference. And the attached state doesn't seem to be going away after 5 minutes.

dorssel, 2022-03-11:
OK, next attempt:
1) install PR #247, artifact is at https://github.com/dorssel/usbipd-win/actions/runs/1798620074
2) `usbipd bind --force --busid 1-2`  (probably will say: already bound)
3) unplug device
4) replug device
5) post result of `usbipd list`
6) try `usbipd wsl attach --busid 1-2`

And about the "Attached" state: it really is attached. However, since Linux is not even able to read the device configuration, it does not report anything with `lsusb`...

fmitch, 2022-03-11:
That seems to have worked! Here's the output from `usbipd list` after unplug/replug:
```
PS C:\WINDOWS\system32> usbipd list
Connected:
BUSID  DEVICE                                                        STATE
1-2    HackRF One                                                    Shared (forced)
1-3    ELAN WBF Fingerprint Sensor                                   Not shared
1-4    HP Wide Vision HD Camera                                      Not shared
2-1    Intel(R) Wireless Bluetooth(R)                                Not shared
3-5    Realtek USB Audio                                             Not shared
3-7    USB Input Device                                              Not shared
5-2    Realtek USB GbE Family Controller                             Not shared
```
The device is now visable and working

dorssel, 2022-03-11:
Thanks for testing!
Just to confirm that PR #247 solved the issue, would you be willing to "break" it again?

1) uninstall `usbipd-win` (this will also unbind everything)
2) install release 2.1.0
3) `usbipd bind --force --busid 1-2`    (this is now necessary again, because of the uninstall)
4) unplug device
5) replug device
6) post result of `usbipd list`
7) try `usbipd wsl attach --busid 1-2`  (I suspect this will now fail again)

dorssel, 2022-03-26:
Closing.
#247 was release with version 2.2.0.

closed 2022-03-26 by dorssel