dorssel/usbipd-win #298: Build instructions

d0n13, 2022-03-11:
Just wondering what tools you need to build the whole project from source? Including installers. I jsut ahve community edition of VS 2022. It doesn't seem to be enough.

dorssel, 2022-03-11:
For the software, you only need .NET SDK 6.
For the installer, you only need *any* VS tools (such as your VS 2022 community edition).

Follow build script as in:
https://github.com/dorssel/usbipd-win/blob/master/.github/workflows/build.yml

d0n13, 2022-03-14:
There is a dependency on .NET 3.5 so when I installed that and turned on the build for the installer it worked in VS 2022

https://dotnet.microsoft.com/en-us/download/dotnet-framework/thank-you/net35-sp1-web-installer

closed 2022-03-14 by d0n13