dorssel/usbipd-win #300: random connection drop

icy-comet, 2022-03-13:
I have a Pi Pico that gets randomly disconnected from Ubuntu (WSL 2 Distro) when using `GNU Screen` or `minicom` on Windows 11.

`minicom` or `GNU Screen` freeze randomly, lock the device for about a minute and then the distro drops the USB connection altogether.

I think this is related to `usbipd-win` because the REPL access works fine if accessed through Windows using `PuTTY` or `Thonny` or through `GNU Screen` with WSL1.

I tried looking at logs but couldn't determine the cause.

Here's the `dmesg` output related to the USB and Pico:
```
[Sun Mar 13 10:44:13 2022] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[Sun Mar 13 10:44:13 2022] vhci_hcd vhci_hcd.0: devid(131073) speed(2) speed_str(full-speed)
[Sun Mar 13 10:44:13 2022] vhci_hcd vhci_hcd.0: Device attached
[Sun Mar 13 10:44:13 2022] vhci_hcd: vhci_device speed not set
[Sun Mar 13 10:44:13 2022] usb 1-1: new full-speed USB device number 4 using vhci_hcd
[Sun Mar 13 10:44:13 2022] vhci_hcd: vhci_device speed not set
[Sun Mar 13 10:44:13 2022] usb 1-1: SetAddress Request (4) to port 0
[Sun Mar 13 10:44:13 2022] usb 1-1: New USB device found, idVendor=239a, idProduct=80f4, bcdDevice= 1.00
[Sun Mar 13 10:44:13 2022] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sun Mar 13 10:44:13 2022] usb 1-1: Product: Pico
[Sun Mar 13 10:44:13 2022] usb 1-1: Manufacturer: Raspberry Pi
[Sun Mar 13 10:44:13 2022] usb 1-1: SerialNumber: E6605838834AC534
[Sun Mar 13 10:44:13 2022] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[Sun Mar 13 10:44:13 2022] input: Raspberry Pi Pico Keyboard as /devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.3/0003:239A:80F4.0003/input/input6
[Sun Mar 13 10:44:13 2022] input: Raspberry Pi Pico Mouse as /devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.3/0003:239A:80F4.0003/input/input7
[Sun Mar 13 10:44:13 2022] input: Raspberry Pi Pico Consumer Control as /devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.3/0003:239A:80F4.0003/input/input8
[Sun Mar 13 10:44:13 2022] hid-generic 0003:239A:80F4.0003: input: USB HID v1.11 Keyboard [Raspberry Pi Pico] on usb-vhci_hcd.0-1/input3
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2954
[Sun Mar 13 10:45:25 2022] vhci_hcd: the urb (seqnum 2954) was already given back
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2955
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2956
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2957
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2958
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2959
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2960
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2961
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2962
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2963
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2964
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2965
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2966
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2967
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2968
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2969
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink->seqnum 2970
[Sun Mar 13 10:45:25 2022] vhci_hcd: urb->status 0
[Sun Mar 13 10:45:25 2022] vhci_hcd: cannot find a urb of seqnum 2919 max seqnum 2971
[Sun Mar 13 10:45:25 2022] vhci_hcd: stop threads
[Sun Mar 13 10:45:25 2022] vhci_hcd: release socket
[Sun Mar 13 10:45:25 2022] vhci_hcd: unlink cleanup rx 2920
[Sun Mar 13 10:45:25 2022] vhci_hcd: disconnect device
[Sun Mar 13 10:45:25 2022] usb 1-1: enqueue for inactive port 0
[Sun Mar 13 10:45:25 2022] usb 1-1: enqueue for inactive port 0
[Sun Mar 13 10:45:25 2022] usb 1-1: USB disconnect, device number 4
```
[... 247 lines cut ...]
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 128
dbug: UsbIpServer.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 128
dbug: UsbIpServer.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 128
dbug: UsbIpServer.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 9
dbug: UsbIpServer.ConnectedClient[1000]
      Unbind or unplug while attached
info: UsbIpServer.ConnectedClient[2]
      Client 172.22.180.193 released device at 2-1 (USB\VID_239A&PID_80F4\E6605838834AC534).
dbug: UsbIpServer.Server[1000]
      connection closed
```

I am new to microcontrollers, so I can't determine if this is hardware related. Any pointers would be much appreciated :)

PS: The `dmesg` output stays same for every disconnect.

dorssel, 2022-03-13:
Here is some analysis. It looks like a lot of URBs are queued and don't complete. This looks like https://github.com/dorssel/usbipd-win/issues/111#issuecomment-977677651; in your case:

>  Unlinking 1914, pending = True, pending count = 18

That means 18 pending URBs, which to me sounds like a problem. Then Linux "aborts" and unlinks all these stale URBs. Interesting is that the first URB to be unlinked is 1914, where the current URB sequence numbers are >2900, so the URB to unlink is a really old one. However, that could be explained by 1914 being on a different (interrupt) pipe than the bulk pipe for normal traffic.

When linux starts unlinking, usbipd-win translates that to an abort for the entire pipe. This is normal, as VBoxUsb does not allow for cancellation of individual URBs. However, this ends with:

> vhci_hcd: cannot find a urb of seqnum 2919 max seqnum 2971

This is definitely not normal. It means that usbipd-win sends a reply for an URB that is not (or no longer) on the list of pending URBs. I don't know if this is wrong on the usbipd-win side or on the Linux side, but it should never happen. I guess this is why Linux decides to give up. Then again, this happens long after things went wrong in the first place: a complete stall on pending URBs.

The usbipd-win log tail starts at URB 2929. Would be nice to see URB 2919 (and even URB 1914). Do you still have the full log?

icy-comet, 2022-03-13:
Here's the full [log](https://github.com/dorssel/usbipd-win/files/8239340/usbip_extensive_log.txt).

dorssel, 2022-03-13:
Endpoint 1 is probably of type interrupt. It has only one URB:

> USBIP_CMD_SUBMIT, seqnum=1914, flags=512, length=64, ep=1

This gets cancelled during Linux cleanup *after* the problems occur, so that's ok.

There is also an endpoint 4:

> USBIP_CMD_SUBMIT, seqnum=1911, flags=512, length=9, ep=4

This is the only URB to it. Don't know what it is for, but it never gets cancelled.

Then Linux quickly submits 16 URBs for endpoint 2. Unusually many, but not wrong per se. It's essentially "overlapped IO" in order to always have a queue of outstanding read requests. Whenever an URB completes, Linux just tops up the queue. So, a pending count around 16 for this endpoint is normal.

Then there is a lot of traffic on endpoint 2. This suddenly stops with a control URB to endpoint 0:

> USBIP_CMD_SUBMIT, seqnum=2950, flags=0, length=0, ep=0

After this, everything gets cancelled and cleaned up. I guess this URB 2950 is the first indication that Linux is trying to solve a problem. Unfortunately, there is nothing right in front of it that indicates any problem. The pending count of 18 at this stage now also makes sense: 16 for the the queue to endpoint 2 + 1 outstanding to endpoint 1 (URB 1914) and 1 to endpoint 4 (URB 1911).

Then the cleanup starts: Linux is unlinking every pending URB, starting with 1914. Weird thing is that 1911 is *not* unlinked. Finally, the error about URB 2919 not being found:

>  vhci_hcd: cannot find a urb of seqnum 2919 max seqnum 2971

Linux did ask to unlink it:

> USBIP_CMD_UNLINK, seqnum=2960, unlink_seqnum=2919

What is odd is that 2919 is (of all the endpoint 2 URBs), the *lowest* seqnum to be unlinked.

TL;DR: everything looks normal until Linux decides it needs to clean up due to some error that isn't obvious. The cleanup itself is weird: 2919 is asked to be unlinked, but shortly after Linux logs it doesn't know about it. And URB 1911 never gets unlinked. Looks like a bookkeeping error, and to be honest, it looks Linux-side.

icy-comet, 2022-03-13:
Thanks @dorssel for looking into this so fast. I'm not so familiar with the USB protocol [ reading up on it now ;) ]. I couldn't have pinpointed the exact points of breakage.

Is there a way to confirm that the issue is on the Linux-side, so that we can go ahead and close the issue? And maybe add a Wiki entry to avoid a duplicate issue after that?

dorssel, 2022-03-13:
I'm working on  better tracing of the URBs in PR #242, but it is not yet ready. In the mean time you could try `usbmon` on the linux side. This allows capture of every URB with `tcpdump` or `tshark`. Unfortunately, `usbmon` is not compiled into the default WSL kernel, so you would have to compile your own following https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel, and enable CONFIG_XXX for `usbmon`. I know at least one user has done so successfully, see https://github.com/dorssel/usbipd-win/issues/154#issuecomment-990332279.

dorssel, 2022-03-13:
Just to be sure, you are running `usbipd-win` version 2.1.0, right?

icy-comet, 2022-03-13:
`2.1.0+14.Branch.master.Sha.0e5a276542476f24b2dc6de8d07c82c9fb8912c7` to be exact.

icy-comet, 2022-03-14:
Hi. Here's the [capture file](https://github.com/dorssel/usbipd-win/files/8246543/capture.zip) of `tshark`. I was able to reproduce the exact connection drop for this.

dorssel, 2022-03-15:
@icy-comet 
Thanks for the capture, really helpful!
I think I found the problem. Could you try PR #307? The installer is at https://github.com/dorssel/usbipd-win/actions/runs/1989341558. This may very well fix it!

icy-comet, 2022-03-16:
Thank You! Yes, [this](https://github.com/dorssel/usbipd-win/actions/runs/1989341558) build seems to have fixed the issue.

PS: `minicom` froze for a moment on start. However, it resumed the connection perfectly and didn't drop it.

PR #307 (2022-03-16): Queue replies per raw endpoint
Fixes #300
closed 2022-03-16 by dorssel