dorssel/usbipd-win #303: CLI error

JPHutchins, 2022-03-14:
Thank you for this work, it is incredible and very useful!

When issuing the command: `usbipd wsl attach --busid 7-4 -d wsl2-usbip` I got exception:

```
Unhandled exception. System.AggregateException: One or more errors occurred. (DiInstallDevice)
 ---> System.ComponentModel.Win32Exception (0x80004005): DiInstallDevice
   at UsbIpServer.NewDev.ThrowOnError(BOOL success, String message) in D:\a\usbipd-win\usbipd-win\UsbIpServer\NewDev.cs:line 21
   at UsbIpServer.NewDev.UnforceVBoxDriver(String originalInstanceId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\NewDev.cs:line 127
   at UsbIpServer.CommandHandlers.Bind(BusId busId, Boolean wslAttach, Boolean force, IConsole console) in D:\a\usbipd-win\usbipd-win\UsbIpServer\CommandHandlers.cs:line 152
   at UsbIpServer.CommandHandlers.UsbIpServer.ICommandHandlers.WslAttach(BusId busId, String distribution, String usbipPath, IConsole console, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\CommandHandlers.cs:line 396
   at UsbIpServer.Program.<>c__DisplayClass10_5.<<Run>b__23>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 372
--- End of stack trace from previous location ---
   at System.CommandLine.Invocation.AnonymousCommandHandler.InvokeAsync(InvocationContext )
   at System.CommandLine.Invocation.InvocationPipeline.<>c__DisplayClass4_0.<<BuildInvocationChain>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass22_0.<<UseParseErrorReporting>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass15_0.<<UseHelp>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass26_0.<<UseVersionOption>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass24_0.<<UseTypoCorrections>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseSuggestDirective>b__23_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass21_0.<<UseParseDirective>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseDebugDirective>b__8_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<RegisterWithDotnetSuggest>b__7_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<CancelOnProcessTermination>b__1_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Invocation.InvocationPipeline.InvokeAsync(IConsole )
   at System.CommandLine.Parsing.ParseResultExtensions.InvokeAsync(ParseResult , IConsole )
   at System.CommandLine.Parsing.ParserExtensions.InvokeAsync(Parser , String[] , IConsole )
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean )
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean )
   at System.Threading.Tasks.Task`1.get_Result()
   at UsbIpServer.Program.Run(IConsole optionalTestConsole, ICommandHandlers commandHandlers, String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 504
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 117
```

Where the source code lead me to reorder the command: `usbipd wsl attach --distribution wsl2-usbip --busid 7-4` and the command works!

I also tried `usbipd wsl attach --busid 7-4 --distribution wsl2-usbip` since I've noticed exceptions from using the single letter flag of the CLI rather than fall flag.  But I will need to investigate to understand the behavior.

JPHutchins, 2022-03-14:
Update:  I got an error again on `usbipd wsl attach --distribution wsl2-usbip --busid 7-4`.  Running the command again worked.  So I suppose this may be unrelated to CLI.  

```
Unhandled exception. System.AggregateException: One or more errors occurred. (DiInstallDevice)
 ---> System.ComponentModel.Win32Exception (0x80004005): DiInstallDevice
   at UsbIpServer.NewDev.ThrowOnError(BOOL success, String message) in D:\a\usbipd-win\usbipd-win\UsbIpServer\NewDev.cs:line 21
   at UsbIpServer.NewDev.UnforceVBoxDriver(String originalInstanceId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\NewDev.cs:line 127
   at UsbIpServer.CommandHandlers.Bind(BusId busId, Boolean wslAttach, Boolean force, IConsole console) in D:\a\usbipd-win\usbipd-win\UsbIpServer\CommandHandlers.cs:line 152
   at UsbIpServer.CommandHandlers.UsbIpServer.ICommandHandlers.WslAttach(BusId busId, String distribution, String usbipPath, IConsole console, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\CommandHandlers.cs:line 396
   at UsbIpServer.Program.<>c__DisplayClass10_5.<<Run>b__23>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 372
--- End of stack trace from previous location ---
   at System.CommandLine.Invocation.AnonymousCommandHandler.InvokeAsync(InvocationContext )
   at System.CommandLine.Invocation.InvocationPipeline.<>c__DisplayClass4_0.<<BuildInvocationChain>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass22_0.<<UseParseErrorReporting>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass15_0.<<UseHelp>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass26_0.<<UseVersionOption>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass24_0.<<UseTypoCorrections>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseSuggestDirective>b__23_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass21_0.<<UseParseDirective>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseDebugDirective>b__8_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<RegisterWithDotnetSuggest>b__7_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<CancelOnProcessTermination>b__1_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Invocation.InvocationPipeline.InvokeAsync(IConsole )
   at System.CommandLine.Parsing.ParseResultExtensions.InvokeAsync(ParseResult , IConsole )
   at System.CommandLine.Parsing.ParserExtensions.InvokeAsync(Parser , String[] , IConsole )
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean )
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean )
   at System.Threading.Tasks.Task`1.get_Result()
   at UsbIpServer.Program.Run(IConsole optionalTestConsole, ICommandHandlers commandHandlers, String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 504
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 117
   ```

dorssel, 2022-03-14:
@JPHutchins 
Thanks for the report!
This is very useful, as you are not the only one suffering from failing `DiInstallDevice`.
Please try the current master build; artifact is at https://github.com/dorssel/usbipd-win/actions/runs/1974527927.
It will not solve the problem, but it does fix the error reporting of `DiInstallDevice` and hopefully give a reason why it fails (see #274). I'm pretty sure it has nothing to do with the CLI.

JPHutchins, 2022-03-15:
Process notes

* MSI from Build #363 prompts to update and updates successfully
* In the output of `usbipd list`, my BLE dongle appears as 
  ```
  7-4    USBIP Shared Device                                           Shared (forced)
  ```
  And the USBPcap warning is displayed at the bottom
  This is not new with Build #363 - I failed to mention it before
* `usbipd wsl attach --distribution wsl2-usbip --busid 7-4` success- note that this is not necessarily different - it was working off and on before
* `usbipd wsl detach --busid 7-4` success
* `usbipd wsl attach --distribution wsl2-usbip --busid 7-4` success
* `usbipd wsl detach --busid 7-4` success
* `usbipd wsl attach --distribution wsl2-usbip --busid 7-4` success
* `usbipd wsl detach --busid 7-4` success
* `usbipd wsl attach --distribution wsl2-usbip --busid 7-4` success
* `usbipd wsl detach --busid 7-4` success

I cannot recreate the error.

dorssel, 2022-03-15:
@JPHutchins 
See https://github.com/dorssel/usbipd-win/issues/270#issuecomment-1068060014.
This may have been accidentally fixed (yes, it's rare, but it happens!).

dorssel, 2022-03-16:
I feel confident that #274 fixed the problem. Can we close this issue?

closed 2022-03-16 by JPHutchins