dorssel/usbipd-win #304: usbip: error: Attach Request for 1-2 failed - Device busy (exported)

Kuhicop, 2022-03-15:
I'm trying to access my phone from WSL:
[image]

I don't know why I'm getting that error:
usbip: error: Attach Request for 1-2 failed - Device busy (exported)

Any ideas?

dorssel, 2022-03-15:
"Device busy" means exactly that. Something on the Windows host is holding an exclusive lock to it. Maybe software, maybe a service, maybe even the native driver itself. If you do not intend to use the device on Windows, you can force binding to the VBoxUsb driver with `usbipd bind --force --busid 1-2`. That may require a reboot, but in the end the device will be permanently claimed by usbipd-win and `wsl attach` should no longer fail.

Kuhicop, 2022-03-15:
Worked :D

[image]

[image]

Thanks <3

closed 2022-03-15 by Kuhicop