dorssel/usbipd-win #313: Fails to install

chollandvoxx, 2022-03-18:
Enclosed is the errors from the log. It looks like VBox.inf may be the culprit, or maybe the dialog box. 
I did not have vbox installed. I tried installing vbox to remove the error, but the error still exists. 

DEBUG: Error 2826:  Control BottomLine on dialog Install extends beyond the boundaries of the dialog to the right by 3 pixels
The installer has encountered an unexpected error installing this package. This may indicate a problem with this package. The error code is 2826. The arguments are: Install, BottomLine, to the right
Info 2898. For Caption textstyle, the system created a 'Segoe UI' font, in 0 character set, of 15 pixels height.
Action 11:54:01: Install. Dialog created
Action ended 11:54:03: Install. Return value 1.
Action 11:54:03: Progress. 
Action start 11:54:03: Progress.
DEBUG: Error 2826:  Control BottomLine on dialog Progress extends beyond the boundaries of the dialog to the right by 3 pixels
The installer has encountered an unexpected error installing this package. This may indicate a problem with this package. The error code is 2826. The arguments are: Progress, BottomLine, to the right
DIFXAPP: INFO: opening HKEY_USERS\S-1-5-21-156100097-4022086903-402207814-33110\Software\Microsoft\Windows\CurrentVersion\DIFxApp\Components\{8A8CBCB2-24E6-5C5D-9554-2DC10D83875B} (User's SID: 'S-1-5-21-156100097-4022086903-402207814-33110') ...
DIFXAPP: INFO:   ENTER:  DriverPackageInstallW
DIFXAPP: INFO:   Installing INF file 'C:\Program Files\usbipd-win\Drivers\VBoxUSB\VBoxUSB.inf' (Plug and Play).
DIFXAPP: INFO:   Could not open file C:\WINDOWS\System32\DriverStore\FileRepository\vboxusb.inf_amd64_dc47f90cefac1d2d\vboxusb.inf. (Error code 0x3: The system cannot find the path specified.)
DIFXAPP: ERROR:  PnP Install failed. (Error code 0x3EE: The volume for a file has been externally altered so that the opened file is no longer valid.)
DIFXAPP: INFO:   Attempting to rollback ...
DIFXAPP: INFO:   No devices to rollback
DIFXAPP: INFO:   Successfully removed '{8A8CBCB2-24E6-5C5D-9554-2DC10D83875B}' from reference list of driver store entry 'C:\WINDOWS\System32\DriverStore\FileRepository\vboxusb.inf_amd64_dc47f90cefac1d2d\vboxusb.inf'
DIFXAPP: INFO:   RETURN: DriverPackageInstallW  (0x3EE)
DIFXAPP: ERROR: encountered while installing driver package 'C:\Program Files\usbipd-win\Drivers\VBoxUSB\VBoxUSB.inf'
DIFXAPP: ERROR: InstallDriverPackages failed with error 0x3EE
DIFXAPP: RETURN: InstallDriverPackages() 1006 (0x3EE)
CustomAction MsiInstallDrivers returned actual error code 1603 (note this may not be 100% accurate if translation happened inside sandbox)
DIFXAPP: INFO: user SID of user performing the install is 'S-1-5-21-156100097-4022086903-402207814-33110'.
DIFXAPP: INFO: This was the component where the install failure occurred. The rollback for this component must have already occurred at the time of failure. No need to rollback again
DIFXAPP: ERROR: Rollback failed with error 0x2

dorssel, 2022-03-19:
@chollandvoxx 
Problem is:

> DIFXAPP: INFO: Could not open file C:\WINDOWS\System32\DriverStore\FileRepository\vboxusb.inf_amd64_dc47f90cefac1d2d\vboxusb.inf. (Error code 0x3: The system cannot find the path specified.)
DIFXAPP: ERROR: PnP Install failed. (Error code 0x3EE: The volume for a file has been externally altered so that the opened file is no longer valid.)

See also https://forums.virtualbox.org/viewtopic.php?f=2&t=99294#p482401.

My guess is either:
1) installer file corruption, try downloading `usbipd-win` again
2) version clash with an earlier (old) installation of VBox; try uninstalling all VBox related drivers that show up for `pnputil /enum-drivers`
3) file system corruption, try `chkdsk`

dorssel, 2022-04-21:
Closing due to lack of activity.

closed 2022-04-21 by dorssel