dorssel/usbipd-win #321: ESP32 (cp210x) detach when trying to flash

Diegorro98, 2022-03-26:
Hi, I've been trying to test the ESP32 in WSL now that cp210x is supported since 0.51.3 release.
I'm using this this setup:

- Windows 11 pro, build 22579.1
- Windows Subsystem for Linux Preview 0.56.2 (from Microsoft Store)
-  WSL kernel 5.10.102.1 (no modified)
- Ubuntu 20.04 over WSL2
- usbipd 2.1.0
- Powershell 7.2.2
- No modified WSL kernel

I attached the ESP32 with usbpid and I get the following by executing `$ dmesg`
```
$ dmesg
...
[ 1601.184053] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[ 1601.185266] vhci_hcd vhci_hcd.0: devid(65542) speed(2) speed_str(full-speed)
[ 1601.186392] vhci_hcd vhci_hcd.0: Device attached
[ 1601.474405] vhci_hcd: vhci_device speed not set
[ 1601.544561] usb 1-1: new full-speed USB device number 6 using vhci_hcd
[ 1601.624393] vhci_hcd: vhci_device speed not set
[ 1601.694383] usb 1-1: SetAddress Request (6) to port 0
[ 1601.730388] usb 1-1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 1601.731178] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1601.731886] usb 1-1: Product: CP2102N USB to UART Bridge Controller
[ 1601.732491] usb 1-1: Manufacturer: Silicon Labs
[ 1601.732918] usb 1-1: SerialNumber: 044718903fe9eb11b63da08787c335f3
[ 1601.735773] cp210x 1-1:1.0: cp210x converter detected
[ 1601.739526] usb 1-1: cp210x converter now attached to ttyUSB0
```
That means that it's working according to [Wiki](https://github.com/dorssel/usbipd-win/wiki/Tested-Devices#all-usb-device-types).

Then I tried to flash esp-idf's blink example with '$ idf.py flash' but it says:
`/dev/ttyUSB0 failed to connect: [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'`

After some research I found some workarounds

-  by running `$ sudo chmod 0666 /dev/ttyUSB0` every time it is attached
- Setting a udev rule with the following content:
     `ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="ea60", MODE:="0666", ENV{ID_MM_DEVICE_IGNORE}="1", ENV{ID_MM_PORT_IGNORE}=">` and running every time too `$ sudo service udev restart` every time I attach the device, which is not pretty comfortable.
If there is any better solution for this, please let me know., but I've seen in [Wiki](https://github.com/dorssel/usbipd-win/wiki/WSL-support#udev)that the last one is the good one, but maybe I'm doing something wrong and I makes me restart the service each time.
And If its the right one, maybe an update that runs `$ sudo service udev restart` after attaching the device could be a good idea.

So, with the obtained permission, I run again esp-idf flash and I get this output:

```
$ idf.py flash
Executing action: flash
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
```
It stays like this like 20 seconds more less and then I get this output:
```
Serial port /dev/ttyUSB0
Traceback (most recent call last):
  File "/home/diego/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB0'
```
And if I run `usbipd wsl list` on powershell, I find that the ESP32 (Silicon Labs CP210x USB to UART Bridge) has been detached.
If `$ dmesg` is executed again after the failed flashing try and I got this:
```
$ dmesg
...
[ 1677.796756] vhci_hcd: unlink->seqnum 570
[ 1677.934506] vhci_hcd: urb->status 0
[ 1677.936030] vhci_hcd: unlink->seqnum 571
[ 1677.936450] vhci_hcd: urb->status 0
[ 1677.938238] vhci_hcd: cannot find a urb of seqnum 566 max seqnum 572
[ 1677.939955] vhci_hcd: stop threads
[ 1677.970111] vhci_hcd: release socket
[ 1677.970491] vhci_hcd: unlink cleanup rx 565
[ 1677.970940] vhci_hcd: the urb (seqnum 565) was already given back
[ 1677.971581] vhci_hcd: unlink cleanup rx 567
[ 1677.971932] vhci_hcd: disconnect device
[ 1677.982592] usb 1-1: USB disconnect, device number 6
[ 1682.994614] cp210x ttyUSB0: failed set request 0x12 status: -110
[ 1682.996149] cp210x ttyUSB0: failed set request 0x0 status: -19
[ 1682.997934] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 1682.998765] cp210x 1-1:1.0: device disconnected
```
An this happens repeatedly.

After some more testing, I flashed a new blink example (+ serial output) on Windows, attached the ESP32 to WSL and ran `$ make monitor` on the original blink example on WSL and I got the output from the ESP32!
Then I tried `$ idf.py monitor`, which it should be almost the same command, but it detach ESP32 from WSL too.
I tried `$ make flash` but it fails again with this output `serial.serialutil.SerialTimeoutException: Write timeout` and its detached.

I hope we can find a solution and change the result status of Silicon Labs CP210x UART Bridge on [Tested Devices Wiki](https://github.com/dorssel/usbipd-win/wiki/Tested-Devices) from :yellow_circle: to :heavy_check_mark:

dorssel, 2022-03-26:
@Diegorro98 
Can you try the brand new 2.2.0 version of `usbipd-win`? It was released minutes ago...
Just so we have the latest debug info?

Diegorro98, 2022-03-26:
Updated to 2.2.0 (and restarted, just in case) and it is working perfectly now!

I also updated the drivers in Windows from [Silicon Labs web](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers), but as usbipd "bypass" Windows so that these drivers are not used, I think that it doesn't really matter.

Thank you so much for you help and your work on usbipd!!

Diegorro98, 2022-03-26:
This permissions problem still remains

> Then I tried to flash esp-idf's blink example with `$ idf.py flash` but it says:
> `/dev/ttyUSB0 failed to connect: [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: [...]

Any idea about it?

dorssel, 2022-03-26:
No, sorry. That's WSL/Linux stuff. Not something that usbipd-win can solve.

Diegorro98, 2022-03-26:
Okey, Thank you anyway!
I'm closing this issue.

closed 2022-03-26 by Diegorro98

Diegorro98, 2022-07-11:
@dorssel, I've found a better better workaround for this problem:
> Then I tried to flash esp-idf's blink example with '$ idf.py flash' but it says: `/dev/ttyUSB0 failed to connect: [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: [...]

`service udev restart` can be run at boot time on the distro by setting at `/etc/wsl.conf` the following content:
```
[boot]
command = "service udev restart"
```
I would like to document it at the wiki but as I'm not a collaborator, I've clone the wiki and modified it in order to document this workaround: https://github.com/Diegorro98/usbipd-win-wiki/commit/7d1d2f5ceedb2e2387a3e614f590526f152198d8