dorssel/usbipd-win #322: Incompatibility with Fortinet Device Guard

dorssel, 2022-03-26:
@dorssel Is it possible that our antivirus software is causing problems with unloading the drivers during --force? We use forticlient. Our software seems to work ok on machines that don't have it installed. But I have not had a wide enough test yet to prove that out. Just wondering what you think? 

All devices that are on the USB have the forticlient driver added to their list of drivers.

[image]

_Originally posted by @d0n13 in https://github.com/dorssel/usbipd-win/issues/270#issuecomment-1075047979_

d0n13, 2022-03-29:
I finally got forticlient removed. Issue remains...

dorssel, 2022-03-29:
To confirm, does the property page indeed list only `winusb.sys`?

d0n13, 2022-03-30:
[image]

Ok, this is probably unrelated to the issue exactly but just noticed it when getting the screenshot.

The device is listed as a portable device now. It's a pixel phone. It used to appear only under USB serial devices. Is that a driver side effect? 
Also noted that it's saying the device requires further installation. I'm not sure if I know what that means.

[image]

The message logged as part of driver loading under "view all events" is as follows. 

```
Device USB\VID_18D1&PID_4EE6&MI_00\7&2d18d080&b&0000 requires further installation.
Driver Management has concluded the process to add Service WUDFWpdMtp for Device Instance ID USB\VID_18D1&PID_4EE6&MI_00\7&2D18D080&B&0000 with the following status: 0.
Driver Management has concluded the process to add Service WinUsb for Device Instance ID USB\VID_18D1&PID_4EE6&MI_00\7&2D18D080&B&0000 with the following status: 0.
```

dorssel, 2022-03-30:
That makes sense. The normal PnP driver installation for the Windows host driver didn't complete yet. That's why it is probably not possible to uninstall and replace the driver with `--force`, because the device is marked for "installation in progress" already. A reboot may help, or else: try disabling the device and reboot and then --force. Or maybe try to uninstall the current drivers manually.

dorssel, 2022-03-31:
Now that we resolved #325 with PR #327, maybe Fortinet wasn't the problem after all? Can we somehow confirm that this issue is actually real?

d0n13, 2022-04-01:
Forticlient is not the issue I don't think either. As I noticed when I removed it.

closed 2022-04-01 by dorssel