dorssel/usbipd-win #323: Can't attach from inside WSL

d0n13, 2022-03-29:
I'm still having issues with devices throwing errors during bind -- force stage. If I run as administrator I get no error and the device shows as shared.

```
> usbipd list
Connected:
BUSID  DEVICE                                                        STATE
...
4-3    Logitech USB Input Device, USB Input Device                   Not shared
4-4    Pixel                                                         Shared
...

Persisted:
GUID                                  DEVICE
```
However, wsl attach does not work and it throws no errors either.

Looking from inside WSL I can see this....

```
$ usbip list --remote=172.17.96.1
Exportable USB devices
======================
 - 172.17.96.1
        4-4: Google Inc. : Nexus/Pixel Device (charging + debug) (18d1:4ee7)
           : USB\VID_18D1&PID_4EE7\FA69H0309127
           : (Defined at Interface level) (00/00/00)
           :  0 - Vendor Specific Class / unknown subclass / unknown protocol (ff/42/01)
```
And trying to attach to it gives the following...

```
$ usbip attach -r 172.17.96.1 -b 4-4
usbip: error: Attach Request for 4-4 failed - Device in error state
```
Is there anyway to understand what error state this is in? And what can I do to help resolve this...

dorssel, 2022-03-29:
Some follow up question/remarks:
1) Are you running the new version 2.2.0?
2) bind without --force doesn't actually do anything with the device. It will always succeed. It is just a registry entry. The error will be seen when attaching, and unfortunately cannot be more specific (due to VBoxUsbMon not expoing that information).
3) `bind --force` is still the way to go. Version 2.2.0 definitely has a more detailed error than 2.1.0 had, so please post it.

d0n13, 2022-03-30:
@dorssel  Thanks for following up.

1. Yes, I built it myself from master a few days ago. I see you have it released now and I can try that version
2. Yes, I understand
3. Problem is there is no error shown. See below

```
PS C:\Users\donie> usbipd unbind -a
PS C:\Users\donie> usbipd bind --force -b 4-4
PS C:\Users\donie> usbipd wsl attach -b 4-4
PS C:\Users\donie> wsl lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
PS C:\Users\donie>
```
It's not exactly helpful. Not sure where to go next with it. Odd thing is, sometimes it works. And works repeatedly and then stops working and I can't figure out what is causing the problem. It's particularly annoying to be so lost :/ 

Is there anyway you could work with us to help us solve it?

d0n13, 2022-03-30:
The usbipd service was running in the foreground as per your debugging instructions. It just logs the following for the above sequence of commands

```
> usbipd server
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\donie.kelly
info: UsbIpServer.ConnectedClient[1]
      Client 172.17.111.161 claimed device at 4-4 (USB\VID_18D1&PID_4EE7\FA69H0309127).
info: UsbIpServer.ConnectedClient[2]
      Client 172.17.111.161 released device at 4-4 (USB\VID_18D1&PID_4EE7\FA69H0309127).
```

Just wondering why it released the device immediately once bound to WSL? Is that something of note?

dorssel, 2022-03-30:
So, the attach is actually working. But then it gets attached immediately again. You'll find the reason in the *Linux* log.

d0n13, 2022-03-30:
@dorssel This is all I can find atm inside the dmesg logs

```
[78281.314526] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[78281.314527] vhci_hcd vhci_hcd.0: devid(262148) speed(3) speed_str(high-speed)
[78281.314554] vhci_hcd vhci_hcd.0: Device attached
[78281.330683] vhci_hcd: connection closed
[78281.330768] vhci_hcd: stop threads
[78281.330770] vhci_hcd: release socket
[78281.330773] vhci_hcd: disconnect device
```

dorssel, 2022-03-30:
@d0n13 
Please run the server with the new capture debugging option, see https://github.com/dorssel/usbipd-win/wiki/Troubleshooting#usb-capture
Let's see if there is any UB traffic at all...

d0n13, 2022-03-30:
@dorssel If you notice at the top of the ticket here it shows the Pixel device as "Shared". Not "Shared (forced)". If I run this on another machine it does show as "Shared (forced)"

The driver for the Pixel does not switch to the VBox driver. 

Here is the capture data. Limited it to 128 bytes. I can see some text to the Pixel device in there and the serial number. Hope it helps to identify the issue.

I am running as administrator on an Enterprise version Windows 10. Have you seen any issues with this version of Windows?

Donie

[usb.zip](https://github.com/dorssel/usbipd-win/files/8381304/usb.zip)

dorssel, 2022-03-30:
@d0n13 
The capture stops quickly after SET_CONFIGURATION. It could be that the device "resets" to a different state then, as if it unplugs/replugs. That is certainly a reason for it to get detached again. Other devices have shown such behavior, which is exactly what --force intends to solve.

dorssel, 2022-03-31:
@d0n13 
Do you think (or can you confirm) that this issue is also fixed by #327 ?

d0n13, 2022-04-01:
This is now resolved by #327

closed 2022-04-01 by d0n13