dorssel/usbipd-win #324: Problem with "Build your own USB/IP enabled WSL2 kernel" : can't find libusbip.so.0 and bzImage

giabattag, 2022-03-30:
Good morning,

I am trying to build the USB/IP kernel on an Ubuntu 20.04 distribution, on Windows 10.

I managed to do everything exactly as on the guide (https://github.com/dorssel/usbipd-win/wiki/WSL-support), but I cannot find the files "libusbip.so.0" and bzImage.

What could I do to solve this problem?

dorssel, 2022-03-30:
Possible duplicate of #131

giabattag, 2022-03-30:
I have tried to redo the three sudo make separately but on the first one it is giving me some errors right now...

Should I delete the Kernel completely and redo the all process?

dorssel, 2022-03-30:
Without the actual error, I don't think anyone can answer this.

giabattag, 2022-03-30:
I did it again as in #131 but the "sudo make -j 4" gives some compilation errors (in ipt_ECN.c, because of "struct ipt_ECN_info" and some macros being not updated  probably, with SET in place of MATCH).

I corrected the errors manually and now I have:

make[2]: *** no rule to make target 'net/netfilter/xt_HL.o', needed by 'net/netfilter/built-in.a'. Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [scripts/Makefile.build:497: net/netfilter] Error 2
make: *** [Makefile:1822: net] Error 2
make: *** Waiting for unfinished jobs....

The make ends but it doesn't really do anything (it stops at AR arch/x86/pci/built-in.a).

dorssel, 2022-03-30:
@giabattag 
That's going horribly wrong. There should be no editing of sources necessary at all.
What is your current kernel version (`uname -a`)?
You did substitute that version in the `git checkout linux-msft-wsl-a.b.c.d` instruction, right? Because it looks like you have a mismatch...

giabattag, 2022-03-30:
Thank you! I enabled the modules in the list in the link (the most similar names that I found):

Device Drivers -> USB Support
Device Drivers -> USB Support -> USB announce new devices
Device Drivers -> USB Support -> USB Modem (CDC ACM) support
Device Drivers -> USB Support -> USB/IP support
Device Drivers -> USB Support -> VHCI HCD
Device Drivers -> USB Support -> Debug messages for USB/IP
Device Drivers -> USB Support -> USB Serial Converter Support
Device Drivers -> USB Support -> USB Serial Converter Support -> USB FTDI Single port Serial Driver

I actually left on the other things that were already on... Should I have switched them all off?

dorssel, 2022-03-30:
These options are on in the default kernel already. You only need to compile your own kernel if you need stuff that isn't enabled by default. What device are you trying to use?

giabattag, 2022-03-30:
"Debug messages for USB/IP" wasn't active actually

I am using an AVR Dragon (AVR32).

giabattag, 2022-03-30:
Because then I also have a Ubuntu 16.04 distro with WSL 2 (same kernel version), where I could recompile de kernel... If that isn't a problem of kernel then

dorssel, 2022-03-30:
No, Windows 10/11 does not matter here. I think AVR should work out of the box, as in: I don't think you need a new kernel.
Current Ubuntu 20.04 should be fine too.

Let's try to debug the problem when you try to attach/use the device. What is failing?

giabattag, 2022-03-30:
It tells: "No USB programmers found."

In Ubuntu 16.04 in another PC it works if I launch the program with sudo for example, but not here.

Also: usbipd wsl list doesn't work for me (he doesn't recognise "usbipd")

dorssel, 2022-03-30:
BTW: I've just compiled the WSL kernel, version 5.10.102.1, from within Ubuntu 20.04 on Windows 10 WSL, with your options. No errors. It did compile `net/netfilter/xt_HL.c` too.

giabattag, 2022-03-30:
On the compilations, I don't really know what happened on my PC then...

giabattag, 2022-03-30:
Oh I see it now, I didnt install the actual software usbipd-win... I came there from the Windows page on WSL and I didn't really see it...
 
I am sorry! Thank you for your helpfulness btw!

giabattag, 2022-03-30:
UPDATE:

I installed usbipd and it works (I had actually a blue screen problem when i unplugged the Dragon from the PC after having attached it to the Ubuntu)

Right now it gives a "Segmentation fault" error, which maybe it is not related to usbip.

dorssel, 2022-03-30:
@giabattag 
Exactly what gives a SEGFAULT, which command/program?

giabattag, 2022-03-31:
I use waf.

The command is ./waf program.

dorssel, 2022-03-31:
So, the attachment of the USB device succeeded? What is the output on Linux when you do `lsusb`?

dorssel, 2022-03-31:
I'm sorry, but I cannot help you further. As far as usbipd-win is concerned: it has done its job. Once the device successfully attaches to Linux/WSL, the remaining issues are within the client OS/software, which is outside the scope of usbipd-win.

Since the original issue is resolved by now, I will close this one. Please feel free to open a new issue for further issues with usbipd-win.

closed 2022-03-31 by dorssel