dorssel/usbipd-win #325: Results of debugging 2.2.0 and running 2.2.0

d0n13, 2022-03-30:
Hi @dorssel 

To try and get to the root of the problems in my installation I have resorted to debug stepping through the code. I noticed something odd. I have this version installed which is latest I believe.

`
2.2.0+72.Branch.master.Sha.00b9e5ec41cb89af4ec628fe83625ad2a50d321d
`

If I run a force bind from the terminal like so and do a list of devices I get the following.

```
usbipd bind --force -b 4-1

BUSID  DEVICE                                                        STATE
4-1    Pixel                                                         Shared

```
This should be showing a (forced) state but it doesn't. The driver attached to the device does not change and the attach does not work.

However, if I setup a profile in the debugger to run the same command and set breakpoints in ICommandHandlers.WslAttach and step through the code until it exits, then the command works and the list shows as 

```
BUSID  DEVICE                                                        STATE
4-1    Pixel                                                         Shared (forced)
```
The follow on attach to wsl then works and the device appears in WSL as expected. 

There seems to be a race condition or something wrong in the code that causes it to fail if running at full speed. Is this possible?

d0n13, 2022-03-30:
If you would like to remote into my machine here to see it happening for real and see if we can solve? We are open to getting extra help from you on this...

dorssel, 2022-03-30:
@d0n13 
RDP? Fine, do you have a (temp) email adress for DM?

d0n13, 2022-03-30:
Let me setup an email and come back to you. Do you have an hourly rate?

d0n13, 2022-03-30:
d0n13.usbip@gmail.com

d0n13, 2022-04-01:
Thank you. This has probably solved all my issues. Will test over the coming days. This particular issue is now resolved.

PR #327 (2022-04-01): Add delay between driver changes
Fixes #325
closed 2022-04-01 by d0n13