dorssel/usbipd-win #328: [Device Support] Ledger Nano S (and X)

antonionardella, 2022-04-01:
Hello here!

I am using a Ledger Nano (S and X) device to store my SSH private keys.
The main use is for public key authentication which works in GNU/Linux as explained in the manufacture's blog: https://blog.ledger.com/ssh/

This is the `ledger-agent` software which is utilized to connect using SSH:
https://github.com/romanz/trezor-agent/blob/master/agents/ledger/ledger_agent.py

I got so far to be able to use `lsusb` in WSL to see the device, but the agent cannot see the dongle to connect

[image]

Is there anything I can do from my side to support the integration (I am not a developer myself though)

Thanks

dorssel, 2022-04-03:
@antonionardella 
Looks like the USB attach worked fine, meaning that the remaining issues are not `usbipd-win` related, but instead WSL related.

Have a look through other (possibly closed) issues on how to ensure that the proper drivers are enabled in the WSL kernel, how `udev` may have to be (re)started to pick up newly attached devices within the distro, how `sudo` may be required to allow access to newly attached devices, etc. Usually these problems are caused by the fact that the default WSL kernel has limited hardware support (you may have to compile your own), and that distros actually run inside a WSL container. These are things that `usbipd-win` cannot solve for you, it has done its job.

akuropka, 2022-04-05:
Just some basic thoughts… it’s obviously an input device. While HID is compiled in, HID_KBD and HID_MOUSE is not.
Regarding rules: https://github.com/LedgerHQ/udev-rules

dorssel, 2022-04-05:
@akuropka 
Interesting... That repo has an open PR: https://github.com/LedgerHQ/udev-rules/pull/8, specifically for hardware wallet use.
This mentions `hidraw`, while the default WSL kernel config has:
```
# CONFIG_HIDRAW is not set
```
Looks like you may have to recompile the kernel with HIDRAW enabled.

dorssel, 2022-04-21:
@antonionardella 
Has this issue been resolved?

dorssel, 2022-05-21:
Closed due to inactivity.

closed 2022-05-21 by dorssel