dorssel/usbipd-win #336: Feature - `attach` devices by specifying `Product:Vendor` pair

CraigHutchinson, 2022-04-08:
# Issue
A device can be attached to any USB port causing the command required to attach to change outside of script control e.g.
`sudo usbip attach -r $HOST -pid 04d8 -vid ec5a`

# Feature Idea
Provide an option to attach devices(s) via `Product:Vendor` ID

# Benefits
- No longer need to supply bus or device index
- Simplify attach multiple devices of same Product+vendor in the same attach call

dorssel, 2022-04-08:
@CraigHutchinson 
Nice idea!

What if multiple devices are connected? In what order will they be attached? As that will determine the bus-number / udev ID on the client side... Order by <?...> bus-id again?

Note the new `usbipd state` command, which allows you to do something similar already. albeit that you need a bit of parsing/looping/etc. in you r script.

But I definitely like this idea!

CraigHutchinson, 2022-04-08:
For a first-pass to get the feature out the door, multiple-devices could be 'ambiguous' error. i.e. Enforce that we always attach a single device. Possibly refering to a device by SerialId could be a secondary option for users with this ambiguity?

CraigHutchinson, 2022-04-08:
And, yes, I see `state` that could be used but requires a bit more processing than ideal.

dorssel, 2022-04-09:
@CraigHutchinson 
To keep implementation simple, I found the following boundary conditions:
- The VID:PID is available from information already retrieved for each device, even for persisted ones that are not currently connected (it is part of what Windows calls the InstanceId).
- Revision number (USB terminology: bcdDevice) is *not* currently available, and cannot be retrieved for persisted devices (unless we change the internal structures & registry format, which I would like to keep as it is)
- A serial number is going to be tough: not all devices have them; and they cannot be retrieved for devices that do have one but are currently in low-power mode. This would require activating the device, which is too much for just "listing" them.

Therefore, I would like to stick to VID:PID only. I don't see a reason to allow only VID (i.e. I see no reason to make PID optional). Do you? And allowing only PID (i.e. optional VID): that would just be silly (different vendors are allowed to have PID collisions).

Windows likes to call this "HardwareId". so I suggest adding `--hardware-id <PID>:<VID>` as alternative to `--busid <BUSID>`.
(BTW: The reason `--busid` is not spelled `--bus-id` is due to the Linux tooling calling it `--busid`).

For the actions I suggest:
1) for `bind` and `wsl attach`, the new option `--hardware-id <VID>:<PID>` will be mutually exclusive with `--busid`. The command will try to find a unique BUSID for VID:PID given all currently connected devices. If it finds (exactly) one, the command behaves the same as if `--busid <BUSID>` would have been specified. If not (zero or multiple) -> error.
2) for `unbind` and `wsl detach`, the new option `--hardware-id <VID>:<PID>` will be mutually exclusive with `--busid` and `--guid`. This time, it will unbind/detach *all* matching devices, including persisted ones. In that sense, it is between `--busid/guid` (exactly one)  and `--all` (exactly all).

I would like a little more input on the "user story"/"use case" for this...

All I can think of is: you are developing for a specific device. You already know the VID:PID. You use multiple development machines, or maybe sometimes a dock/sometimes not. Therefore, you would like to bind/attach the device in one go, without having to find out what the BUSID is for this environment. This would also make scripting (for example on build bots) easier.

Can you explain exactly what you are trying to achieve, or which problem you are trying to solve?

dorssel, 2022-04-09:
@benmcmorran 
Could you have a look at this proposal?

CraigHutchinson, 2022-04-10:
For my use case I have multiple devices of the same product that I can test/develop with. The test engineer may connect the usb to any of a large number of usb ports on the pc and a large pool of devices to test. We wish to simplify the mounting workflow for a single device as priority.

dorssel, 2022-04-11:
@CraigHutchinson 

They do the same as the suggested `usbipd state` parsing: get a full list --> grep for VID:PID --> attach by busid. 

```
$ usbip attach -r server_IP_address -b $(/usr/sbin/usbip list -p -r server_IP_address | grep '0924:3d68' | cut '-d:' -f1 | awk '{print $1}')
```

CraigHutchinson, 2022-04-11:
Yes, but the `usbpd-win` output for `list` is not as  informative as the linux daemon it appears. This could eb raised as a separate ticket or keep related here?

benmcmorran, 2022-04-12:
Thanks for the suggestion @CraigHutchinson! The proposal makes sense to me. Given that the `--hardware-id` flag is just a convenience option for existing functionality, I don't see any issues with restricting `bind` and `wsl attach` to a single unique device. In addition to the use cases already described, I can also see this being useful for simplifying documentation in embedded development scenarios. For example, instead of asking the user to list devices and then select their board, documentation could instead be written as a single command with a well-known VID/PID, like `0483:374B` for ST-Link devices.

dorssel, 2022-04-12:
Alright... it has been decided! Expect a PR soon.

PR #341 (2022-04-20): Add option --hardware-id
Fixes #336
closed 2022-04-20 by dorssel