dorssel/usbipd-win #340: Enhancement Request?

meluvalli, 2022-04-13:
Hello.  I am currently running WSL in "BRIDGE" mode.  (See: https://stackoverflow.com/questions/61002681/connecting-to-wsl2-server-via-local-network#:~:text=There%20is%20another%20mode%20called,its%20own%20network%20card%20each.)

However, I am unable to attach a USB when in this mode :(

The error I get is: usbipd: error: The specified WSL distribution cannot be reached via the WSL virtual switch; try restarting the WSL distribution.

I am needing bridge mode because I am running a HomeKit application that requires bonjour advertisements on the network.   If I leave it in NAT mode, HomeKit cannot see the device even if I open all the ports.  (I think this is because it broadcasts on the 127 address).  

I am not exactly sure how USBIPD works, but would be nice if I could just specify the IP address of the WSL.  

Any thoughts?  Maybe there is another way for me to route all internal WSL to my external IP without bridge mode?  Any ideas to assist would be GREATLY appreciated!

dorssel, 2022-04-13:
@meluvalli 
From the quoted document:

> Open Hyper-V Manager as administrator

And later:

> sudo ip addr add 192.168.0.116/24 dev eth0

In other words, you are just using Hyper-V with a fixed IP address on the client (and presumably host as well). In this case, forget about the `wsl` subcommands. For example, `wsl attach` is just a macro for `bind` + client-side attach. Since the *normal* WSL-switch uses dynamic IP addresses, it made sense to automate it.

Instead, simply use `bind` on the host and then attach from within Linux to the (fixed) IP of the host (as in the README).

I guess this "WSL mode" only works on Window Pro (since it requires Hyper-V)? In any case, you can make it work with the available commands already. I don't think it is common enough to warrant changes to the `wsl` commands. Especially because those were created to simplify the *dynamic* IP addresses of the regular WSL switch.

meluvalli, 2022-04-14:
THANK YOU SO MUCH!  That did it and is working like a charm!

closed 2022-04-14 by meluvalli