dorssel/usbipd-win #343: Feature - attach devices by specifying NAME matching pattern e.g. "*Camera*"

CraigHutchinson, 2022-04-14:
https://github.com/dorssel/usbipd-win/pull/341 simplifies attaching from a VID:PID pair but when/if a the device is a generic device such as a "Integrated Camera" or "USB Camera" it would be nice to be able to easily select the device.

The idea would be to support along the lines of `-n, --name <NAME> Share device with name matching pattern e.g. "*Camera*"` this increase user convenience to even higher levels. For example "Integrated Camera" from any/most manufacturer may be mounted 🥇

As for the 'pattern' style this is open to be regex or otherwise but a simple GLOB style match could suffice depending on the engine available.

(Relates to https://github.com/dorssel/usbipd-win/issues/336)

dorssel, 2022-04-14:
I'm sorry, but this does not seem to be a good idea, it is not similar to VID:PID at all (which certainly *is* a good idea).

Device "names" are unstable/unpredictable due to:
- driver install state (installed/uninstalled)
- driver version dependent
- driver language/locale dependent
- OS/User language/locale dependent
- can be different for devices that are disabled/enabled
- can be different for devices that are in low power mode vs. active

The example you give does not seem for automation/scripting at all, but for a one-off action. Binding your integrated webcam needs to be done only once, and then you might as well look up the busid once. You would have to list the devices anyway, as the name is totally unpredictable for reasons above. And then the busid is stable, it won't ever change for integrated devices.

And if you *really* want this, you can easily wrap this functionality in a script, see https://github.com/dorssel/usbipd-win/wiki/Automation.

To be honest, this sounds like over-engineering, i.e. "lets make this because we can". But I don't think it would actually serve a useful purpose. Making a complex (you even mention GLOB) device selection to *possibly* save a single `list` command for 2% of the users only once does not make sense to me.

CraigHutchinson, 2022-04-14:
I agree with your comments and you are definitely right.

I must thank you for your very quick and informative responses which are  invaluable and greatly appreciated. I'm not expecting to have any further silly ideas anytime soon!

Coming from a standpoint any idea could be a good one it's always good to get these out on the table for others to coin in and discuss. You never know the odd one becomes unexpected gold! this one however isn't worth further thought.

Emteq Ltd, Sussex Innovation Centre, Brighton BN1 9SB
Company Registration: 09708880

This e-mail (including any attachments) is considered confidential and is intended only for the recipient(s) listed above. Any review, use, disclosure, distribution or copying of this e-mail is prohibited except by or on behalf of the intended recipient. If you have received this email in error, please notify me immediately by reply e-mail, delete this e-mail, and do not disclose its contents to anyone. Thank you.

dorssel, 2022-04-14:
@CraigHutchinson 
Thank you for your involvement. I certainly don't want to discourage you from bringing up *any* ideas. You already had a very nice one: VID:PID! So keep them coming!

closed 2022-04-14 by dorssel