dorssel/usbipd-win #345: usb drive not listed by lsblk in wsl

JanBittendorf, 2022-04-14:
This is similar to issue #70. The device is listed by lsusb but not by lsblk:

`
└─$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 090c:2000 Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) Intenso Ultra Line
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`
`
└─$ lsblk
NAME MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda    8:0    0   256G  0 disk
sdb    8:16   0 339.7M  1 disk
sdc    8:32   0   256G  0 disk /
`

But the difference is, that there is no error in [dmesg](https://github.com/dorssel/usbipd-win/files/8490260/usbipd_dmesg.txt).
Also I am using kali linux instead of ubuntu and since linux-tools is unavailable for kali, I am using the usbip package. 

I also have the [tracelog](https://github.com/dorssel/usbipd-win/files/8490281/usbipd_trace.txt), though that seems to be normal as well.

dorssel, 2022-04-14:
This looks like a missing driver in the Linux kernel. The default WSL kernel does not support USB storage devices. Did you compile your own?

dorssel, 2022-05-13:
@JanBittendorf 
Has this issue been resolved?

Korel, 2022-05-16:
> This looks like a missing driver in the Linux kernel. The default WSL kernel does not support USB storage devices. Did you compile your own?

So does this mean this part in the wiki is stale now?
_Recent versions of Windows running WSL kernel 5.10.60.1 or later already include support for common scenarios like USB-to-serial adapters and flashing embedded development boards. If you're trying to do one of these tasks on Ubuntu, you can avoid recompiling the kernel by following the WSL Setup instructions at the top of this page. If you require special drivers, you'll need to build your own kernel for WSL 2._

I have followed the initial setup on WSL2 Ubuntu and haven't compiled the kernel myself yet. I guess there is no escaping it?

dorssel, 2022-05-16:
@Korel 
The USB storage scenario is not part of "USB-to-serial adapters and flashing embedded development board". Apparently it falls under "if you require special drivers...".
For USB storage (and webcams, and WiFi, and Bluetooth, etc.) you still need to compile your own kernel.
See: https://github.com/microsoft/WSL/issues/7770

Korel, 2022-05-16:
Thank you @dorssel . I have built the kernel as you said, .config did not include USB Mass Storage support so I have enabled it. It works for me now!

marcin2x4, 2022-06-04:
I've created issue to Microsoft to create driver so no custom compilation will be required:
https://github.com/microsoft/WSL/issues/8470

dorssel, 2022-06-15:
Closed due to inactivity.

closed 2022-06-15 by dorssel

kapad, 2024-10-26:
For anyone else that made it here, this guide helped me build a custom version of the WSL kernel which supports usb storage devices. 

https://gist.github.com/tatumroaquin/b51d16be877d66de9bd3c1bac94faec9