dorssel/usbipd-win #349: iphone doesnt attach on ubuntu 20.04

someone2008, 2022-04-17:
usbipd wsl attach --busid 3-4
usbipd: info: Using default distribution 'Ubuntu-20.04'.
usbip: error: Attach Request for 3-4 failed - Device busy (exported)

usbipd: error: Failed to attach device with BUSID '3-4'.

dorssel, 2022-04-17:
> Device busy (exported)

Please search through the issues for 'iPhone' and 'Device busy'. This has been answered before.

Hinkiii, 2022-04-25:
Is it possible to always bind/attach to the same device slot?

dorssel, 2022-04-25:
If you mean: attach as soon as the device is connected: see PR #351.

Hinkiii, 2022-04-25:
Well the auto attach feature itself works great but sadly the WSL environment never sees it as being in the same port I guess?

`mynam@Jojo:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 05ac:1281 Apple, Inc. Apple Mobile Device [Recovery Mode]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
mynam@Jojo:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 05ac:1281 Apple, Inc. Apple Mobile Device [Recovery Mode]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
mynam@Jojo:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 05ac:1227 Apple, Inc. Mobile Device (DFU Mode)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

And the usbipd logs

`usbipd wsl attach --auto-attach -b 6-2
usbipd: info: Starting endless attach loop; press Ctrl+C to quit.
Attached
Detached
usbip: error: Attach Request for 6-2 failed - Device not found
Attached
Detached
usbip: error: Attach Request for 6-2 failed - Device not found
Attached
Detached
usbip: error: Attach Request for 6-2 failed - Device not found
Attached`

dorssel, 2022-04-25:
@Hinkiii 
I think you are confusing "Bus 001 Device 004" with the device ID designator. You'll find that the latter is always 1-1 (unless of course you attach multiple devices). It is normal for the Linux VHCI driver to have an incremental number for "Bus XXX Device YYY". This is not WSL related, but happens on any Linux. What is constant is the "1-1" as listed in `/sys/bus/usb/devices`.

dorssel, 2022-05-13:
@Hinkiii 
Has this issue been resolved?

Hinkiii, 2022-05-13:
It's probably worth closing, it wasn't working quite as I wanted it to but it's fine

closed 2022-05-13 by dorssel

WWIA7062, 2024-06-02:
i have the same issue trying to run checkra1n on WSL 2