dorssel/usbipd-win #350: Is there a way to automatically run USBIPD with admin privileges, or else not require admin privileges?

e-t-l, 2022-04-18:
On my Windows machine, I am the sole user and my user profile is the administrator. If I open a regular cmd prompt and use USBIPD, it works fine up until I run `usbipd bind -b 2-2`, which returns an access denied error that says I need administrator privileges. I would like to not have to reopen an elevated cmd every time I use usbipd. Is there a way to work around this? (I have already tried assigning full permissions for the usbipd executable and service to all users - aka my profile, system, the admin group, and the users group).

dorssel, 2022-04-18:
@e-t-l 
You need to `usbipd bind` only once. It is persistent, also across reboots. That must be done as administrator, but attaching does not.

e-t-l, 2022-04-19:
Correct me if I'm wrong, but once a a device is shared with `usbipd bind`, USBIPD has exclusive use of it. My use case is having a personal Windows machine that I use locally, but where I occasionally want to share a USB device with a different machine on my network. So leaving the USB device binded (bound?) to USBIPD 24/7 is not an option.

dorssel, 2022-04-19:
@e-t-l 
That's not how it works. If you use `bind` it is still available to the host until you actually attach. Have you read the documentation?

e-t-l, 2022-04-19:
I did read the documentation, but I've also done some test runs, and my experiences there led to this question. 

When I used `bind` on the USB mouse I had handy, it became unavailable to the host computer (i.e. when I moved the mouse, the cursor did not move). Only after executing `unbind -a` did it become available again. (note: I did actually have to use `bind -f` to successfully bind the mouse. Is it possible that force-binding a device would make it unavailable to the local machine?)

dorssel, 2022-04-19:
```
PS> usbipd bind --help
usbipd-win 2.2.0

Description:
  Registers a single USB device for sharing, so it can be attached to other machines. Unless the --force option is
  used, shared devices remain available to the host until they are attached to another machine.

Usage:
  usbipd bind [options]

Options:
  -b, --busid <BUSID> (REQUIRED)  Share device having <BUSID>
  -f, --force                     Force binding; the host cannot use the device
  -?, -h, --help                  Show help and usage information
```

dorssel, 2022-05-13:
@e-t-l 
Has this issue been resolved?

dorssel, 2022-06-15:
Closed due to inactivity.