dorssel/usbipd-win #352: USB flash drives are not accessible

bengalih, 2022-04-19:
On Kernel Linux 5.10.102.1-microsoft-standard-WSL2.
`usbipd wsl attach` works fine to attach.
dmesg shows device attached:
```
[ 4356.534334] usb 1-1: new high-speed USB device number 6 using vhci_hcd
[ 4356.684337] usb 1-1: SetAddress Request (6) to port 0
[ 4356.721598] usb 1-1: New USB device found, idVendor=0930, idProduct=6545, bcdDevice= 1.00
[ 4356.721602] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4356.721604] usb 1-1: Product: DataTraveler 2.0
[ 4356.721605] usb 1-1: Manufacturer: Kingston
[ 4356.721606] usb 1-1: SerialNumber: 50E549C69355EDA1D980024F
```

lsusb shows device:

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 0930:6545 Toshiba Corp. Kingston DataTraveler 102/2.0 / HEMA Flash Drive 2 GB / PNY Attache 4GB Stick
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

however device is not accessible under any /dev.  Unable to mounted and used as a block device for file access.

please advise.

dorssel, 2022-04-19:
Possible duplicate of #70, #345, etc.

This looks like a missing driver in the Linux kernel. The default WSL kernel does not support USB storage devices. Did you compile your own?

bengalih, 2022-04-19:
Thanks.  When reading https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel I interpreted the following passage incorrectly:

#Recent versions of Windows running WSL kernel 5.10.60.1 or later already include support for common scenarios like USB-to-serial adapters and flashing embedded development boards. If you're trying to do one of these tasks on Ubuntu, you can avoid recompiling the kernel by following the WSL Setup instructions at the top of this page. If you require special drivers, you'll need to build your own kernel for WSL 2.

I had inferred that usb storage would be a "common scenario" and as such thought that my kernel was up-to-date enough that it would qualify.  Re-reading I see that I likely need to compile my own.

I will do that tonight and close this if the problem is resolved.

thanks.

bengalih, 2022-04-20:
Compiling my own seems to work along with the additional USB STORAGE and FS options referenced in #70.
I am not a kernel whiz, but I have a few comments (and also questions) about what is posted @ https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel

1) Everything after Build USB/IP tools will fail because you have done a `cd tools/usb/usbip`.  You have to move back into your root working directory.  This is minor, and most should figure out, but you did a good job of presenting "for dummies" and some may be confused without this direction.

2) I got an error without dwarves installed.  It is mentioned to install this here:
https://github.com/microsoft/WSL2-Linux-Kernel
But you do not include it in your list of prerequisites:
`sudo apt install build-essential flex bison libssl-dev libelf-dev libncurses-dev autoconf libudev-dev libtool`

Questions:
I'm a bit confused on everything that comes after (and includes) the "Build USB/IP tools" section.
I followed the guide exactly by exporting my Ubuntu distro and then importing a new one as wsl2-usbip.

When complete, I wasn't able to use usbipd on the wsl2-usbip distro copy because it would only attach to the default Ubuntu and I couldn't find any syntax to point usbpid to another distro?

Also, I see in the guide that we copy the bzImage to a shared location on Windows and create a .wslconfig which presumably points all distros launched by this user to run the kernel??

So, I launched my default Ubuntu install after creating the .wslconfig file and I seemingly can now see my USB storage.  However, I completed the following 3 sections only under the wsl2-usbip distro:

```
- Build USB/IP tools.
- Copy tools libraries location so usbip tools can get them.
- Install usb.ids so you have names displayed for usb devices.
```

I am confused as to what these 3 steps do and why my Ubuntu distro appears to function now with USB storage despite these only being configured on the wsl2-usbip import.  And, to this point, what, if anything should I need to do in order to use this kernel on another distro (i.e. on another machine).  Can I just copy it to a shared location on that system and update the .wslconfig file?  What if I don't want all distros to use it -how do I configure just one?

Thank you very much!

dorssel, 2022-04-20:
@benmcmorran 
Concerning point 1 and 2: can you have a look (and maybe amend) the kernel build instructions? Maybe the dependencies have changed since Nelson wrote this?

@bengalih

> I couldn't find any syntax to point usbpid to another distro?
 
`usbipd wsl attach --help` would have revealed the `--distribution` option...

> What if I don't want all distros to use it -how do I configure just one?

That's not how WSL works. All distros share the same kernel.

bengalih, 2022-04-21:
> @benmcmorran
Concerning point 1 and 2: can you have a look (and maybe amend) the kernel build instructions? Maybe the dependencies have changed since Nelson wrote this?

I believe the error I got was similar to this:
```
BTF: .tmp_vmlinux.btf: pahole (pahole) is not available
Failed to generate BTF for vmlinux
Try to disable CONFIG_DEBUG_INFO_BTF
```

I have just a few additional comments about the build guide coming from a neophyte user in this.  Take them or leave them, but these are the things which I feel needed better explanation and were a possible source of confusion (for me).   I will also leave it to you to close out this issue - I was going to, but thought you might want it to remain open until any documentation was updated.  One final question before I list the issues:

- Previous to building the kernel as instructed with USB support, the output of dmesg was very succinct as shown above it looked like:

```
[ 4356.534334] usb 1-1: new high-speed USB device number 6 using vhci_hcd
[ 4356.684337] usb 1-1: SetAddress Request (6) to port 0
[ 4356.721598] usb 1-1: New USB device found, idVendor=0930, idProduct=6545, bcdDevice= 1.00
[ 4356.721602] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4356.721604] usb 1-1: Product: DataTraveler 2.0
[ 4356.721605] usb 1-1: Manufacturer: Kingston
[ 4356.721606] usb 1-1: SerialNumber: XXXXXXXX9355EDA1D98XXXXX
```

After the kernel build there are over 6000 lines output in dmesg when attaching a drive, a sampling of is:
```
[  354.960015] usbip_core: usbip_dump_header:282: USBIP_RET_SUBMIT: st 0 al 0 sf 0 #p -1 ec 0
[  354.960016] vhci_hcd: pickup_urb_and_free_priv:26: find urb seqnum 125
[  354.960021] usb 1-1:        devnum(4) devpath(1) usb speed(high-speed)
[  354.960022] usbip_core: usbip_dump_usb_device:94: tt hub ttport 0
[  354.960023] usb 1-1:
[  354.960024] usbip_core: usbip_dump_usb_device:98:   0
[  354.960025] usbip_core: usbip_dump_usb_device:98:   1
[  354.960026] usbip_core: usbip_dump_usb_device:98:   2
[  354.960027] usbip_core: usbip_dump_usb_device:98:   3
```
Is this because the guide references to configure the kernel with:

> Device Drivers -> USB Support -> USB/IP -> Debug messages for USB/IP

If not, do you know what would cause this excessive debug logging?

Here are the (minor) issues with the guide:

1) It looks like this page was two different documents merged together.  One from "WSL Setup" onwards and the second half starting with "Building your own USB/IP enabled WSL 2 kernel".  I say this because there is some duplication that isn't totally clear.  For instance in the former it is advised to install the client tools:
```
sudo apt install linux-tools-virtual hwdata
sudo update-alternatives --install /usr/local/bin/usbip usbip /usr/lib/linux-tools/*/usbip 20

But in the latter this is also instructed:
```
cd tools/usb/usbip
sudo ./autogen.sh
sudo ./configure
sudo make install -j 8
```
Building the tools should not be necessary if you already followed the guide up to that point.  This is why I believe the kernel build guide was tacked on, but the flow of the document isn't clear.  Additionally, this step doesn't seem necessary:

`sudo cp libsrc/.libs/libusbip.so.0 /lib/libusbip.so.0`

Or, at least not if you installed from the package as first instructed.  Perhaps it is necessary if you install from source as instructed in the later half.

2) Just the stuff I already mentioned above in https://github.com/dorssel/usbipd-win/issues/352#issuecomment-1103319022

Thanks again for your support and knowledge!

benmcmorran, 2022-04-22:
@dorssel @bengalih Thanks for the notes about the kernel builds. I probably won't be able to get to it today, but it's on my TODO list to revisit the build instructions and bring them back up to date.

bengalih, 2022-04-22:
@benmcmorran No problems.  I have a few more issues to call out.  This was my first time compiling a kernel on WSL (and maybe only my second or third time compiling one over the past 10-15 years), so forgive me if these things are better known, but these are things that gave me problems.  I actually opened some bugs over at https://github.com/microsoft/WSL.

Specifically https://github.com/microsoft/WSL/issues/8319 and https://github.com/microsoft/WSL/issues/8318

While I found a functional solution to the former, the latter seems to be a more long standing issue/limitation.

However, both of these issues can be mitigated/avoided by ensuring that the build is occurring on the root EXT4 file system and not 9p mounted /mnt/X volumes.

I call this out because if you follow the current guide as written, it instructs to do the following:

`C:\Users\bengalih>wsl --distribution wsl2-usbip  --user bengalih`

doing this will drop you into the user's home *windows* directory, e.g.:

`bengalih@MOBILEONE:/mnt/c/Users/bengalih$`

Which is, mounted as 9p.  Not only will you likely experience the issues I referenced above by not switching to say `/home` to run your `git clone` and `make`, but you will have build times which appear to be 3-4x longer than if you are on the ext4 mount point.

Again, for someone more familiar with the build process they may have avoided these errors, but AFAIK there isn't any official guidance when doing kernel builds that 9p mounts are to be avoided.  After many hours of messing around with this I can definitively say they should be.

dorssel, 2022-05-13:
Documentation will be updated. Original issue has been resolved.

closed 2022-05-13 by dorssel

itxiaoxian, 2025-04-25:
I'm currently facing a problem. When I execute "usbipd list", it shows "1-9  0781:5575  USB Large-capacity Storage Device  Shared (forced)". However, when I run "usbip.exe attach --remote=10.10.10.5 --busid=1-9", it prompts "usbip: error: bus id 1-9 is already in use". How should I solve this issue? I'm using Windows 10 system.