dorssel/usbipd-win #353: USB Webcam detected by WSL2 but still no /dev/video0

Katzeee, 2022-04-20:
On kernel 5.10.102.1
I can find my cameras by lsusb and dmesg but there is no /dev/video*
```bash
xac-lenovo@Xac-lenovo:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1b3f:2247 Generalplus Technology Inc. GENERAL WEBCAM
Bus 001 Device 002: ID 174f:2459 Syntek Integrated Camera
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
```bash
[   39.808193] usb 1-1: new high-speed USB device number 2 using vhci_hcd
[   39.958332] usb 1-1: SetAddress Request (2) to port 0
[   39.998173] usb 1-1: New USB device found, idVendor=174f, idProduct=2459, bcdDevice= 0.08
[   39.998175] usb 1-1: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[   39.998175] usb 1-1: Product: Integrated Camera
[   39.998176] usb 1-1: Manufacturer:
[   39.998176] usb 1-1: SerialNumber: 0001
[   42.276715] vhci_hcd vhci_hcd.0: pdev(0) rhport(1) sockfd(3)
[   42.276717] vhci_hcd vhci_hcd.0: devid(131082) speed(3) speed_str(high-speed)
[   42.276740] vhci_hcd vhci_hcd.0: Device attached
[   42.618364] usb 1-2: new high-speed USB device number 3 using vhci_hcd
[   42.768229] usb 1-2: SetAddress Request (3) to port 1
[   42.800316] usb 1-2: config 1 interface 0 altsetting 0 endpoint 0x83 has an invalid bInterval 32, changing to 9
[   42.801390] usb 1-2: New USB device found, idVendor=1b3f, idProduct=2247, bcdDevice= 1.00
[   42.801392] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   42.801393] usb 1-2: Product: GENERAL WEBCAM
[   42.801394] usb 1-2: Manufacturer: GENERAL
[   49.139456] hv_balloon: Max. dynamic memory size: 3986 MB
```
I compiled my own WSL followed by the instructions in https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel.
```bash
xac-lenovo@Xac-lenovo:/usr/src/4.19.43-microsoft-standard$ sudo make modules_install -j 12
  INSTALL drivers/hid/usbhid/usbhid.ko
  INSTALL drivers/media/dvb-core/dvb-core.ko
  INSTALL drivers/media/mc/mc.ko
  INSTALL drivers/media/v4l2-core/v4l2-dv-timings.ko
  INSTALL drivers/media/v4l2-core/videodev.ko
  INSTALL drivers/net/mii.ko
  INSTALL drivers/net/usb/cdc_ether.ko
  INSTALL drivers/net/usb/cdc_ncm.ko
  INSTALL drivers/net/usb/rndis_host.ko
  INSTALL drivers/net/usb/usbnet.ko
  INSTALL drivers/usb/class/cdc-acm.ko
  INSTALL drivers/usb/common/usb-common.ko
  INSTALL drivers/usb/core/usbcore.ko
  INSTALL drivers/usb/phy/phy-generic.ko
  INSTALL drivers/usb/serial/ch341.ko
  INSTALL drivers/usb/serial/cp210x.ko
  INSTALL drivers/usb/serial/ftdi_sio.ko
  INSTALL drivers/usb/serial/usbserial.ko
  INSTALL drivers/usb/storage/usb-storage.ko
  INSTALL drivers/usb/usbip/usbip-core.ko
  INSTALL drivers/usb/usbip/vhci-hcd.ko
  DEPMOD  5.10.102.1-microsoft-standard-WSL2+
```
```bash
xac-lenovo@Xac-lenovo:/usr/src/4.19.43-microsoft-standard$ sudo make install -j12
sh ./arch/x86/boot/install.sh 5.10.102.1-microsoft-standard-WSL2+ arch/x86/boot/bzImage \
        System.map "/boot"
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.10.102.1-microsoft-standard-WSL2+ /boot/vmlinuz-5.10.102.1-microsoft-standard-WSL2+
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 5.10.102.1-microsoft-standard-WSL2+ /boot/vmlinuz-5.10.102.1-microsoft-standard-WSL2+
update-initramfs: Generating /boot/initrd.img-5.10.102.1-microsoft-standard-WSL2+
W: Couldn't identify type of root file system for fsck hook
run-parts: executing /etc/kernel/postinst.d/unattended-upgrades 5.10.102.1-microsoft-standard-WSL2+ /boot/vmlinuz-5.10.102.1-microsoft-standard-WSL2+
run-parts: executing /etc/kernel/postinst.d/update-notifier 5.10.102.1-microsoft-standard-WSL2+ /boot/vmlinuz-5.10.102.1-microsoft-standard-WSL2+
run-parts: executing /etc/kernel/postinst.d/xx-update-initrd-links 5.10.102.1-microsoft-standard-WSL2+ /boot/vmlinuz-5.10.102.1-microsoft-standard-WSL2+
```
I don't know how to do to fix it. 
please advise.

dorssel, 2022-04-20:
@Katzeee 
> /usr/src/4.19.43-microsoft-standard

Are you compiling the right kernel version? Usually, WSL kernels are built monolithic (i.e. no modules, but everything compiled in: 'Y' instead of 'M'). Maybe you need to manually `modprobe` the correct modules? Or maybe you need to (re)start `udev`?

Katzeee, 2022-04-21:
It is the driver's problem, I didn't choose the right driver. After I selected the right driver, I can find `/dev/video0`.

BTW, i can't use it by `cheese`, though there is `/dev/video0` . But by this way i can:

```bash
ffplay -f video4linux2 -input_format mjpeg -framerate 30 -video_size 640*480 /dev/video0
```

Thank you for your help.

closed 2022-04-21 by Katzeee

1184197761, 2022-05-05:
> 这是司机的问题，我没有选择正确的司机。选择正确的驱动程序后，我可以找到 .`/dev/video0`
> 
> 顺便说一句，我不能使用它，尽管有.但通过这种方式，我可以：`cheese``/dev/video0`
> 
> ```shell
> ffplay -f video4linux2 -input_format mjpeg -framerate 30 -video_size 640*480 /dev/video0
> ```
> 
> 感谢您的帮助。
Can you tell me which driver select? Thanks!

Katzeee, 2022-05-06:
> > It is the driver's problem, I didn't choose the right driver. After I selected the right driver, I can find .`/dev/video0`
> > BTW, i can't use it by` cheese` , though there is`/dev/video0 ` . But by this way i can:
> > ```shell
> > ffplay -f video4linux2 -input_format mjpeg -framerate 30 -video_size 640*480 /dev/video0
> > ```
> > 
> >   
> > Thank you for your help.
> 
> Can you tell me which driver to select? Thanks!

I selected following drivers for usb,

```
Device Drivers->USB support[*]
Device Drivers->USB support->Support for Host-side USB[M]
Device Drivers->USB support->Enable USB persist by default[*]
Device Drivers->USB support->USB Modem (CDC ACM) support[M]
Device Drivers->USB support->USB Mass Storage support[M]
Device Drivers->USB support->USB/IP support[M]
Device Drivers->USB support->VHCI hcd[M]
Device Drivers->USB support->VHCI hcd->Number of ports per USB/IP virtual host controller(8)
Device Drivers->USB support->Number of USB/IP virtual host controllers(1)
Device Drivers->USB support->USB Serial Converter support[M]
Device Drivers->USB support->USB Serial Converter support->USB FTDI Single Port Serial Driver[M]
Device Drivers->USB support->USB Physical Layer drivers->NOP USB Transceiver Driver[M]
Device Drivers->Network device support->USB Network Adapters[M]
Device Drivers->Network device support->USB Network Adapters->[Deselect everything you don't care about]
Device Drivers->Network device support->USB Network Adapters->Multi-purpose USB Networking Framework[M]
Device Drivers->Network device support->USB Network Adapters->CDC Ethernet support (smart devices such as cable modems)[M]
Device Drivers->Network device support->USB Network Adapters->Multi-purpose USB Networking Framework->Host for RNDIS and ActiveSync devices[M]
```
and these are for webcam,

```
Device Drivers->Multimedia support[M]
Device Drivers->Multimedia support[M]->Media core support->Video4Linux core[M]
Device Drivers->Multimedia support[M]->Media core support->Media Controller API[*]
Device Drivers->Multimedia support[M]->Media drivers->Media USB Adapters[*]
Device Drivers->Multimedia support[M]->Media drivers->Media USB Adapters[*]->USB Video Class(UVC)[M]
Device Drivers->Multimedia support[M]->Media drivers->Media USB Adapters[*]->UVC input evnets device support[*]
```

These are enough for me, but if you still can't use the cam, you should try select more drivers under `Multimedia support`.

For more details, see https://github.com/Katzeee/ReadingNotes/blob/master/about-programing/wsl2-using-usb.md

1184197761, 2022-05-06:
> I selected following drivers for usb,
>
> ```
> Device Drivers->USB support[*]
> Device Drivers->USB support->Support for Host-side USB[M]
> Device Drivers->USB support->Enable USB persist by default[*] [...]

I am glad to receive your reply, and you sincerely suggest that I have given me a lot, thank you again!

1184197761, 2022-05-06:
Glad to hear from you, and your sincere advice helped me a lot, thanks again!

> I selected following drivers for usb,
>
> ```
> Device Drivers->USB support[*]
> Device Drivers->USB support->Support for Host-side USB[M]
> Device Drivers->USB support->Enable USB persist by default[*] [...]

Glad to hear from you, and your sincere advice helped me a lot, thanks again!

EstelleChabanel, 2022-10-19:
> I selected following drivers for usb,
>
> ```
> Device Drivers->USB support[*]
> Device Drivers->USB support->Support for Host-side USB[M]
> Device Drivers->USB support->Enable USB persist by default[*] [...]

Hi, 

I have the same problem. I don't understand how/where do you select the driver ?

Thank you

Katzeee, 2022-10-19:
> Hi,
>
> I have the same problem. I don't understand how/where do you select the driver ?
>
> Thank you

https://gist.github.com/cerebrate/d40c89d3fa89594e1b1538b2ce9d2720
This may help you, run ‘make menuconfig’ then you would see the options.