dorssel/usbipd-win #355: One or more errors occurred. (DeviceIoControl returned error ERROR_NOT_SUPPORTED)

coldalfred, 2022-04-23:
Hi dorssel,

Thanks for your hard work.

I'm trying to connect it to a remote Ubuntu Server kernel 5.4.0-109 with a installed usbip client using " apt install linux-virtual " but
I get this error :

Client - Ubuntu 20.04.4 LTS command : usbip attach -r 192.168.20.2 -b 1-4
Server - Windows 10 command : usbipd bind -b 1-4 --force

usbipd list:
Connected:
BUSID  DEVICE                                                        STATE
1-1    USB Input Device                                              Not shared
1-4    USB Input Device, Nano X                                 Shared (forced)

UsbIpServer.Server[1000]
      new connection from 10.0.200.2
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.AggregateException: One or more errors occurred. (DeviceIoControl returned error ERROR_NOT_SUPPORTED)
       ---> System.ComponentModel.Win32Exception (50): DeviceIoControl returned error ERROR_NOT_SUPPORTED
         --- End of inner exception stack trace ---
         at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean )
         at System.Threading.Tasks.Task.Wait(Int32 , CancellationToken )
         at System.Threading.Tasks.Task.Wait()
         at UsbIpServer.ConfigurationManager.RestartingDevice.Dispose() in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 427
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 262
dbug: UsbIpServer.Server[1000]
      connection closed

Hope you can help me

dorssel, 2022-04-23:
@coldalfred 
It looks like your hub does not support IOCTL_USB_HUB_CYCLE_PORT. Maybe we can work around that, but first I like to find out *why* it does not support that.

1) Can you find this hub (the one that your device is connected to) in Device Manager? And then, what drivers are installed *for that hub*? Are those the default Microsoft hub drivers, or some special brand drivers?

2) Or else it could be a USB filter driver. When you use `usbipd bind` *without* `--force`, does usbipd warn about (possibly) incompatible filter drivers? If so, which one?

dorssel, 2022-04-23:
@coldalfred 
Had another look. It is still relevant to get the answers to the questions above, but ... you found a bug too!
Thanks for the report!

Turns out that IoControl() does not throw a Win32Exception on failure, but instead an AggregateException with a Win32Exception as InnerException. That's why it wasn't caught properly. There are more occasions in the code where that is done wrong, so I will fix those.

It means that your attach will probably work, but without a port cycle (the failure was meant to be caught and ignored). But some non-compliant devices really need a port cycle to work properly. So attach *with* port cycle is always better. Therefore, it is still relevant to find out why your hub does not support it...

dorssel, 2022-04-23:
@coldalfred 
OK, that was easy enough. It was only in that one spot.
I've created PR #356, the installer is at https://github.com/dorssel/usbipd-win/actions/runs/2212009642. You may want to give that a try. But it is still worthwhile investigating why your hub does not support port cycling.

coldalfred, 2022-04-23:
I'm so grateful with you, thanks for this hard work, IT WORKS

If you need more debug from me I can do it.

Yes I have this filter but I didn't remember how I installed it 
usbipd: warning: USB filter 'TsUsbFlt' is known to be incompatible with this software; 'bind --force' will be required

If you know how I can uninstall it would be great.

After test your update msi, the new error was:
"  usbip: debug: usbip.c:129:[run_command] running command: `attach'
   libusbip: error: udev_device_new_from_subsystem_sysname failed
   usbip: error: open vhci_driver "

with a simple " modprobe vhci-hcd " it works

coldalfred, 2022-04-23:
> @coldalfred It looks like your hub does not support IOCTL_USB_HUB_CYCLE_PORT. Maybe we can work around that, but first I like to find out _why_ it does not support that.
>
> 1. Can you find this hub [...]

[image]

> 2. Or else it could be a USB filter driver. When you use `usbipd bind` _without_ `--force`, does usbipd warn about (possibly) incompatible filter drivers? If so, which one?

usbipd: warning: USB filter 'TsUsbFlt' is known to be incompatible with this software; 'bind --force' will be required

@dorssel

dorssel, 2022-04-23:
@coldalfred 
Yes, that's it. "Ts" stands for Terminal Server. The TsUsbFlt.sys filter driver does something very similar to VBoxUsb. You probably (just like I once did!) tried to use RDP+RemoteFx to forward your device. Unfortunately, RemoteFx only works when connecting to a Server edition of Windows; it is unavailable in Hyper-V for Windows Pro.

See https://docs.microsoft.com/en-us/answers/questions/56512/usb-mouse-over-remotefx.html for which group policy (which you probably enabled at some point in the past) to disable/unconfigure to remove TsUsbFlt (NOTE: it will require a reboot too).

Fun fact: Lack of RemoteFx in Hyper-V for Windows Pro was the original reason for me to develop usbipd-win.

dorssel, 2022-04-23:
@coldalfred 
Thanks for confirming PR #356. I will merge it, which will automatically close this issue.

closed 2022-04-23 by dorssel

dorssel, 2022-04-23:
@coldalfred 
> I'm so grateful with you, thanks for this hard work, IT WORKS

Thanks for your kind support. Much appreciated!

PR #356 (2022-04-23): Fix handling synchronous Wait and AggregateException
Fixes #355