dorssel/usbipd-win #362: Return codes

d0n13, 2022-04-27:
Looking though the code (well I was looking for a reason) I notice all lot of commands return ExitCode.Failure which is fine but I'm particularly looking for a reason as I'm calling usbipd from code.

Any possibility of extending the range of return codes to include things like WSL not running or wrong version etc? Just so it can be picked up easier.

The problem I have is that I run all usbipd commands inside a windows service (as admin) and it communicates back to the client apps (not running as admin) so I need to communicate the reason for the failure ideally. Getting an exit code of 1 for everything is not ideal...

Would be a nice to have.

dorssel, 2022-04-27:
@d0n13 
Doable. It's on the backlog now.

For the time being:
All messages are sent to stderr and have the format: `usbipd: error: XXX`. In case of an error (when ExitCode.Failure is returned), the last message indicates the final failure. It may be preceded by one or more `usbipd: info: XXX` and/or `usbipd: warning: XXX` messages. Redirection and parsing of stderr could solve it for now.

d0n13, 2022-04-27:
Yes. I realise it went to std err once I got blank messages from std out :) I solved the issue I was having anyway by logging the message to event viewer as I’m in a service. 

Be nice to se Eir improved in the future though. No pressure

closed 2022-04-27 by d0n13