dorssel/usbipd-win #363: "usbip: error: already used bus id: #-#"

e-t-l, 2022-04-30:
I'm trying to attach a USB game controller but I'm receiving the error in the title. Output below:

```
> usbip list -r 100.72.247.44
Intel Corp.
Exportable USB devices
======================
 - 100.72.247.44
        1-2: Microsoft Corp. : Xbox One Controller (Firmware 2015) (045e:02dd)
           : USB\VID_045E&PID_02DD\7EED850A438F
           : Vendor Specific Class / unknown subclass / unknown protocol (ff/47/d0)
           :  0 - Vendor Specific Class / unknown subclass / unknown protocol (ff/47/d0)
           :  1 - Vendor Specific Class / unknown subclass / unknown protocol (ff/47/d0)
           :  2 - Vendor Specific Class / unknown subclass / unknown protocol (ff/47/d0)

> usbip attach -r 100.72.247.44 -b 1-2 -t
usbip: error: already used bus id: 1-2

> usbip port
Imported USB devices
====================

> usbip detach -p 1-2
usbip: error: failed to detach
```

I have no idea what "already used bus id" could mean in this case. To my knowledge I don't have any local USB buses with that id (running `usbip list -l` does not show any busid of `1-2`, and running the attach command with the `-s` arg produces the same error). As you can see from `usbip port` there do not appear to be any currently attached devices.

Anyone know what's going on here?

dorssel, 2022-04-30:
@e-t-l 
What distribution is this? How/where did you get this `usbip` version? The option `-t` is (rather undocumented) used to specify a port different from 3240. It requires an argument, but you do not provide one. Since `usbipd-win` uses the standard port, it isn't needed at all.

The Linux source does not contain such an error (see: https://elixir.bootlin.com/linux/v5.17.5/source/tools/usb/usbip/src), so that's why I wonder what usbip version this is...

e-t-l, 2022-04-30:
Good questions (to clarify, it gives the same error with or without the `-t` flag). I'm using [@cezanne/usbip-win](https://github.com/cezanne/usbip-win). So, this is all being run on Windows, but I've already performed all the steps necessary to make that compatible, e.g. installing the driver, disabling signature enforcement, etc. USBIP has been working exactly as expected for everything except the attach command.

dorssel, 2022-04-30:
@e-t-l 
Oh, OK. I don't know anything about Windows clients. Can't you ask over there? They seem to have an issue list... This looks like a client-side issue.

dorssel, 2022-04-30:
@e-t-l 
The project you mentioned does not seem to be active. Have you tried a more recent fork, e.g. https://github.com/vadimgrn/usbip-win2?

dorssel, 2022-04-30:
@e-t-l 
OK, I did a quick "Use the source, Luke".
Error is from: https://github.com/cezanne/usbip-win/blob/95ccf0a8863b0eeb75dd383cb5ac863c825286f2/userspace/src/usbip/usbip_attach.c#L287
Caused by: https://github.com/cezanne/usbip-win/blob/95ccf0a8863b0eeb75dd383cb5ac863c825286f2/userspace/src/usbip/usbip_attach.c#L150

This means it is the good old "device busy". That means that something is already using it, most likely something on you host (service/software/driver). Look through the issue list for "Device busy" on how to deal with it.

e-t-l, 2022-04-30:
Thank you! That gave me the idea to force-bind it with `usbipd bind -f -b 1-2` on the server PC, and after that I was able to attach it!

closed 2022-04-30 by e-t-l