dorssel/usbipd-win #364: Missing /usr/include/libudev.h

morpheuslord, 2022-05-01:
what do i do about this i tried installing this in the kali wsl for wifi pentesting but it seems it has missing packages what should I do about it please help.

dorssel, 2022-05-01:
What command gives this error?

morpheuslord, 2022-05-01:
the make file and its always showing that something is missing and is aborting what the hell is wrong with my wsl

dorssel, 2022-05-01:
What make file? You really have to provide more info on what you are doing and what you are trying to achieve. I'm sorry, but with so little information it is not possible to troubleshoot your problem.

morpheuslord, 2022-05-02:
i will tell in detail what i was trying to do
first: completed the prerequisite installed all the packages mentioned
second: built the mainconfig file
thrid completed the: autoconfig and config files in USB directory
forth tried to compile in the main make file
end result somekind of stupid error

a request can you please make a youtube vid on the kernal installation inside a wsl machine and how to install the USB module inside the wsl machine doing this even with a detailed document is tough.

dorssel, 2022-05-02:
What distro are you using? The instructions are for Ubuntu (20.04, I believe).

morpheuslord, 2022-05-08:
kali wsl

dorssel, 2022-05-09:
Since the kernel is shared between all WSL instances, the quickest way to the exit is to also install a Ubuntu distro. The kernel you will build there (and install as the WSL kernel) will then also be used by kali. Alternatively, you will have to install the required dependencies for building a kernel on kali, but you're on your own then as I know nothing about kali.

morpheuslord, 2022-05-10:
Is there any video reference for this

dorssel, 2022-05-10:
https://www.google.com/search?q=site%3Ayoutube.com+how+to+compile+wsl+kernel

dorssel, 2022-06-15:
Closed due to inactivity.

closed 2022-06-15 by dorssel