dorssel/usbipd-win #365: Updating driver forces reboot of system, without offering a chance to postpone.

akulbe, 2022-05-01:
I updated this driver in the process of running `winget upgrade --all` and it forcibly rebooted the system. While it was a "graceful" reboot (in the sense that it wasn't just like hitting the reset button), there was absolutely no chance to postpone it.

I'm not sure if this is something that can be controlled, or not, but I had 12 VMs running, as well as multiple containers. (this is a beefy machine)

I did not get a chance to shut them down beforehand. That's **_not_** cool.

This was the entry that showed up in Event Viewer.

```
The process msiexec.exe has initiated the restart of computer KRAKEN on behalf of user NT AUTHORITY\SYSTEM for the following reason: No title for this reason could be found
 Reason Code: 0x80030002
 Shutdown Type: restart
 Comment: The Windows Installer initiated a system restart to complete or continue the configuration of 'usbipd-win'.
```

dorssel, 2022-05-02:
This is (a very unfortunate, I agree) Winget issue. Please report it there. You'll find there are already multiple issues reported.

There is no way for the installer to alter Winget's behavior. All we can do is warm the user. That's why the Readme tells you to use `--interactive`. And the release notes also told you (you did read those before installing, right?)

So, when using Winget: use `--interactive` to prevent auto reboot, or use the MSI installer directly without Winget.

dorssel, 2022-05-16:
A workaround has been put in place to prevent winget from forceReboot until winget gets proper reboot handling support.

closed 2022-05-16 by dorssel