dorssel/usbipd-win #366: Import error on external BTRFS drives from WSL2

saberman888, 2022-05-01:
Hi I'm trying to attach a btrfs raid array composed of 3 drives. I have them binded, but when I try to import them through WSL2 I get an import error.  
I'm using Ubuntu with kernel: Linux 5.10.102.1-microsoft-standard-WSL2. dmesg doesn't really show anything other than messages for existing drives. Is this just not working because my setup is unsupported or what am I doing wrong?

dorssel, 2022-05-02:
What's the exact 'import error'. Any details?

dorssel, 2022-05-04:
That looks like missing drivers. Did you compile your own kernel? You should, as the default WSL kernel does not contain USB storage drivers.

saberman888, 2022-05-04:
I followed the directions in the wiki, nothing has changed so far...

```
[    0.733099] scsi 0:0:0:0: Direct-Access     Msft     Virtual Disk     1.0  PQ: 0 ANSI: 5
[    0.733889] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    0.734919] sd 0:0:0:0: [sda] 536870912 512-byte logical blocks: (275 GB/256 GiB)
[    0.734922] sd 0:0:0:0: [sda] 4096-byte physical blocks
[    0.735041] sd 0:0:0:0: [sda] Write Protect is off
[    0.735043] sd 0:0:0:0: [sda] Mode Sense: 0f 00 00 00
[    0.735433] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    1.027854] sd 0:0:0:0: [sda] Attached SCSI disk
[    1.112042] EXT4-fs (sda): mounted filesystem with ordered data mode. Opts: discard,errors=remount-ro,data=ordered
[    1.267454] clocksource: tsc: mask: 0xffffffffffffffff max_cycles: 0x33cbd063299, max_idle_ns: 440795360766 ns
[    1.267968] clocksource: Switched to clocksource tsc
[    1.297456] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    1.847569] Adding 4194304k swap on /swap/file.  Priority:-2 extents:3 across:4210688k
[    2.807732] scsi 0:0:0:1: Direct-Access     Msft     Virtual Disk     1.0  PQ: 0 ANSI: 5
[    2.808225] sd 0:0:0:1: Attached scsi generic sg1 type 0
[    2.808918] sd 0:0:0:1: [sdb] 536870912 512-byte logical blocks: (275 GB/256 GiB)
[    2.808920] sd 0:0:0:1: [sdb] 4096-byte physical blocks
[    2.808980] sd 0:0:0:1: [sdb] Write Protect is off
[    2.808982] sd 0:0:0:1: [sdb] Mode Sense: 0f 00 00 00
[    2.809163] sd 0:0:0:1: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    2.812100] sd 0:0:0:1: [sdb] Attached SCSI disk
[    2.831117] EXT4-fs (sdb): mounted filesystem with ordered data mode. Opts: discard,errors=remount-ro,data=ordered
[    2.850752] FS-Cache: Duplicate cookie detected
[    2.850756] FS-Cache: O-cookie c=0000000012abcc79 [p=00000000f1b5e498 fl=222 nc=0 na=1]
[    2.850758] FS-Cache: O-cookie d=000000001399b15f n=0000000036e88cc6
[    2.850758] FS-Cache: O-key=[10] '34323934393337353734'
[    2.850764] FS-Cache: N-cookie c=00000000cc590e6b [p=00000000f1b5e498 fl=2 nc=0 na=1]
[    2.850764] FS-Cache: N-cookie d=000000001399b15f n=00000000f3cd1ff1
[    2.850765] FS-Cache: N-key=[10] '34323934393337353734'
[   49.152486] hv_balloon: Max. dynamic memory size: 13052 MB
[   61.735744] WSL2: Performing memory compaction.
[  146.110757] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[  146.110759] vhci_hcd vhci_hcd.0: devid(196612) speed(5) speed_str(super-speed)
[  146.110799] vhci_hcd vhci_hcd.0: Device attached
[  146.467468] usb 2-1: SetAddress Request (2) to port 0
[  146.467475] usb 2-1: new SuperSpeed Gen 1 USB device number 2 using vhci_hcd
[  146.502464] usb 2-1: New USB device found, idVendor=1058, idProduct=25ed, bcdDevice=10.31
[  146.502466] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  146.502467] usb 2-1: Product: My Book 25ED
[  146.502468] usb 2-1: Manufacturer: Western Digital
[  146.502468] usb 2-1: SerialNumber: 575834324438304334384553
[  147.786069] vhci_hcd vhci_hcd.0: pdev(0) rhport(1) sockfd(3)
[  147.786071] vhci_hcd vhci_hcd.0: devid(196611) speed(5) speed_str(super-speed)
[  147.786110] vhci_hcd vhci_hcd.0: Device attached
[  148.137472] usb 2-2: SetAddress Request (3) to port 1
[  148.137478] usb 2-2: new SuperSpeed Gen 1 USB device number 3 using vhci_hcd
[  148.172625] usb 2-2: New USB device found, idVendor=0bc2, idProduct=203b, bcdDevice=18.01
[  148.172627] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  148.172628] usb 2-2: Product: Expansion SW
[  148.172629] usb 2-2: Manufacturer: Seagate
[  148.172630] usb 2-2: SerialNumber: 00000000NAC839HL
[  148.929261] vhci_hcd vhci_hcd.0: pdev(0) rhport(2) sockfd(3)
[  148.929263] vhci_hcd vhci_hcd.0: devid(196610) speed(5) speed_str(super-speed)
[  148.929300] vhci_hcd vhci_hcd.0: Device attached
[  149.277466] usb 2-3: SetAddress Request (4) to port 2
[  149.277473] usb 2-3: new SuperSpeed Gen 1 USB device number 4 using vhci_hcd
[  149.312914] usb 2-3: New USB device found, idVendor=1058, idProduct=25a3, bcdDevice=10.29
[  149.312916] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  149.312917] usb 2-3: Product: Elements 25A3
[  149.312918] usb 2-3: Manufacturer: Western Digital
[  149.312919] usb 2-3: SerialNumber: 31454858384E4D5A
[  149.979454] vhci_hcd vhci_hcd.0: pdev(0) rhport(3) sockfd(3)
[  149.979457] vhci_hcd vhci_hcd.0: devid(196609) speed(5) speed_str(super-speed)
[  149.979530] vhci_hcd vhci_hcd.0: Device attached
[  150.327445] usb 2-4: SetAddress Request (5) to port 3
[  150.327451] usb 2-4: new SuperSpeed Gen 1 USB device number 5 using vhci_hcd
[  150.363375] usb 2-4: New USB device found, idVendor=1058, idProduct=25ee, bcdDevice=40.09
[  150.363378] usb 2-4: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[  150.363379] usb 2-4: Product: My Book 25EE
[  150.363379] usb 2-4: Manufacturer: Western Digital
[  150.363380] usb 2-4: SerialNumber: 5758313244343030524C4E5A
[  182.740633] WSL2: Performing memory compaction.
```

dorssel, 2022-05-04:
The fact that syslog does not report new drives (sdX) means the USB storage drivers are not there. Did you compile them as modules? Then you may need to load them with modprobe. Search through the other issues for more help.

saberman888, 2022-05-07:
Alright, well thanks for the help. I'll keep looking around. Sorry if I sounded a bit crass in my last few messages

dorssel, 2022-05-07:
@saberman888 
NP.
This is what a successful attach *with* drivers looks like:
> May  8 00:02:40 oakdale-ubuntu kernel: [66059.533945] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
May  8 00:02:40 oakdale-ubuntu kernel: [66059.533951] vhci_hcd vhci_hcd.0: devid(196628) speed(5) speed_str(super-speed)
May  8 00:02:40 oakdale-ubuntu kernel: [66059.533996] vhci_hcd vhci_hcd.0: Device attached
May  8 00:02:40 oakdale-ubuntu kernel: [66059.794122] usb 2-1: SetAddress Request (2) to port 0
May  8 00:02:40 oakdale-ubuntu kernel: [66059.794136] usb 2-1: new SuperSpeed USB device number 2 using vhci_hcd
May  8 00:02:40 oakdale-ubuntu kernel: [66059.818525] usb 2-1: New USB device found, idVendor=0951, idProduct=1666, bcdDevice= 1.10
May  8 00:02:40 oakdale-ubuntu kernel: [66059.818535] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May  8 00:02:40 oakdale-ubuntu kernel: [66059.818537] usb 2-1: Product: DataTraveler 3.0
May  8 00:02:40 oakdale-ubuntu kernel: [66059.818539] usb 2-1: Manufacturer: Kingston
May  8 00:02:40 oakdale-ubuntu kernel: [66059.818540] usb 2-1: SerialNumber: 408D5C162FCCE460B976BFF8
May  8 00:02:40 oakdale-ubuntu kernel: [66059.849275] usb-storage 2-1:1.0: USB Mass Storage device detected
May  8 00:02:40 oakdale-ubuntu kernel: [66059.849806] scsi host1: usb-storage 2-1:1.0
May  8 00:02:40 oakdale-ubuntu kernel: [66059.849891] usbcore: registered new interface driver usb-storage
May  8 00:02:40 oakdale-ubuntu kernel: [66059.852252] usbcore: registered new interface driver uas
May  8 00:02:41 oakdale-ubuntu kernel: [66060.881001] scsi 1:0:0:0: Direct-Access     Kingston DataTraveler 3.0 PMAP PQ: 0 ANSI: 6
May  8 00:02:41 oakdale-ubuntu kernel: [66060.881398] sd 1:0:0:0: Attached scsi generic sg2 type 0
May  8 00:02:41 oakdale-ubuntu kernel: [66060.883372] sd 1:0:0:0: [sdb] 121110528 512-byte logical blocks: (62.0 GB/57.8 GiB)
May  8 00:02:42 oakdale-ubuntu kernel: [66060.888394] sd 1:0:0:0: [sdb] Write Protect is off
May  8 00:02:42 oakdale-ubuntu kernel: [66060.888401] sd 1:0:0:0: [sdb] Mode Sense: 45 00 00 00
May  8 00:02:42 oakdale-ubuntu kernel: [66060.889707] sd 1:0:0:0: [sdb] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
May  8 00:02:43 oakdale-ubuntu kernel: [66062.611269]  sdb: sdb1 sdb2
May  8 00:02:43 oakdale-ubuntu kernel: [66062.618166] sd 1:0:0:0: [sdb] Attached SCSI removable disk

As you can see: right after logging of the serial number, the usb-storage driver takes over. If you don't see that (i.e., if there is no further logging after the serial number), then the storage driver is not present.

dorssel, 2022-05-13:
@saberman888 
Has this issue been resolved?

closed 2022-06-15 by dorssel