dorssel/usbipd-win #367: ubuntu released 17/4 and kernel 5.10.102.1 broke the usbipd

dariofinardi, 2022-05-03:
After the update the usbipd doesn't work anymore on both Windows 11 and Windows 10.
No way to install the proper linux-tools version.
I tried  5.4.0.109-generic as last available from "virtual" package.
I tried 5.11.0-22-generic.

dorssel, 2022-05-03:
@dariofinardi 
If you updated the Linux tools, then you also have to run the `update-alternatives` command again. See the wsl support wiki.

dariofinardi, 2022-05-03:
Yes, I did but I still receive:

root@pc: usbipd wsl

WARNING: usbipd not found for kernel 5.10.102.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.10.102.1-microsoft-standard-WSL2
    linux-cloud-tools-5.10.102.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2

dorssel, 2022-05-03:
You're supposed to run `usbipd` on the Windows side.

dariofinardi, 2022-05-03:
Yes, I did on Windows and able to bind and connect the device to WSL.

So, the usbipd on Linux is unuseful, only a legacy placeholder?
I shouldn't be worried about that WARNING, isn't it?

dariofinardi, 2022-05-03:
Now it's clear.

One more question, because I'm curious: what's the usbipd on Linux side?
After that we shall close the deal :-) and I'll test on my device.

dorssel, 2022-05-03:
That's for when you run Linux on your bare metal host and want to share the devices.

johnpettingill, 2022-05-03:
After following wiki instructions for usbip in Ubuntu, I get
(WARNING: usbip not found for kernel 5.10.102.1-microsoft) 

And on the windows side I can list and share drives, but when I try to attach one with  usbipd wsl attach --busid<> I get an error 

usbipd: error: WSL 'usbip' client not correctly installed. See https://github.com/dorssel/usbipd-win/wiki/WSL-support for the latest instructions.

I can find the usbip files in /usr/library & /usr/local on Ubuntu. Do the wiki instructions on that page only apply to Ubuntu 20.04?

I have kernel 5.10.102.1 on Ubuntu 22.04, running in WSL2, Windows 10

dariofinardi, 2022-05-04:
> That's for when you run Linux on your bare metal host and want to share the devices.

Now it's clear.
I confirm that WSL2 runs as expected.

I think that the wiki may be a bit improved with some more details to avoid doubts.

dorssel, 2022-05-06:
@benmcmorran 
This is not the first time WSL users get confused about the difference between `usbipd` (server/Windows side) and `usbip` (client/Linux side). Maybe we should add it to the WSL wiki. Since it is `usbipd wsl` that does "magic" for both sides. Hyper-V users are more aware of the difference.

dorssel, 2022-05-13:
Documentation will be updated. Original issue has been resolved.

closed 2022-05-13 by dorssel