dorssel/usbipd-win #368: Auto attach script error

Uldiniad, 2022-05-05:
I have seen the following error occasionally when using the -a parameter:
`/mnt/c/Program Files/usbipd-win/wsl-scripts/auto-attach.sh: line 70: 0008: value too great for base (error token is "0008")`

I can reproduce frequently with one of my devices

Uldiniad, 2022-05-05:
Not sure if relevant but the same device that I can reproduce this on frequently also does not create /dev/ttyUSB0 as it is expected to. I can create a separate issue if wanted

dorssel, 2022-05-05:
I'll try to solve this over the weekend. We'll see if these are related (I think they are). If not, let's create a separate issue only then.

Thanks for the report.

dorssel, 2022-05-06:
@Uldiniad 
Could you test the installer at https://github.com/dorssel/usbipd-win/actions/runs/2284505248? That should fix it.

Uldiniad, 2022-05-06:
Seems to have fixed the error. The auto attach script no longer exits.

Uldiniad, 2022-05-06:
Is that 2.3.1 package going to be the same as the one in the releases? I installed via winget originally. Can I continue using it after installing the above?

closed 2022-05-06 by Uldiniad

Uldiniad, 2022-05-06:
> Not sure if relevant but the same device that I can reproduce this on frequently also does not create /dev/ttyUSB0 as it is expected to. I can create a separate issue if wanted

Also, this device seems to switch between 2 bus IDs even though it remains connected to the same port. Is this expected or normal for some devices? Should I open a ticket if these quirks are related this project?

dorssel, 2022-05-06:
@Uldiniad 
No, this will be in the next release, which will probably by 2.4.0. The ".1" at the end is an artifact from GitVersion auto-numbering. You are now on an (unsigned) development build. When 2.4.0 will be released, `winget upgrade usbipd` should pick it up automatically.

dorssel, 2022-05-06:
@Uldiniad 
Do you mean 2 different BUSIDs on the Windows side? That would be odd... Maybe if the device is sometimes seen as HighSpeed (USB 2) and sometimes as SuperSpeed (USB 3)...

Uldiniad, 2022-05-06:
Good question and theory, I'll keep an eye out for that. If that does not explain it I will open a separate issue

PR #372 (2022-05-06): Fix auto-attach
- fixes base-10 parsing for numbers with leading zeros
- allow auto-attach on already attached devices

Fixes #368