dorssel/usbipd-win #370: Custom installer

d0n13, 2022-05-06:
Hi @dorssel 

I'm trying to create a custom installer for usbipd which is integrated with our own project. I'm having some difficulties getting it to work. I can get everything to install ok but getting error 2 (file or path not found) when I do a force bind. The drivers seem to be in the correct place in system32/drivers but the forced driver swap is reporting the error above.

It seems the drivers are not getting installed correctly. Is there any way to verify the drivers are installed ok from the registry etc?

Also:
Going through your Wix install code I notice you have a reference to two items that I cannot understand. The two items below are referencing a Group called Usbipd and PowerShell but I cannot find the ComponentGroup with those names. Any help would be appreciated as it may be a reason for the above error?

<ComponentGroupRef Id="Usbipd" />
<ComponentGroupRef Id="PowerShell" />

Thanks for any help you can push my way...
Donie

dorssel, 2022-05-07:
The drivers are installed using a standard WiX extension; here: https://github.com/dorssel/usbipd-win/blob/7d12fb5e3aa5cbc747557a8df05bd9ca3119292b/Installer/Drivers.wxs#L32
There is not much control over how it does that. I have noticed that sometimes a reboot is required, even if the final installer doesn't say so. Apparently, the WiX extension is not always correct about the reboot requirement. Reboots seem to be required  more often when the driver is updated, and less often when they are installed for the first time. But again, the reboot requirement is not always reflected correctly by the MSI installer; and (again) there is nothing that can be done to influence this behavior. Maybe it is not even WiX's fault, as Windows often has a "have you tried rebooting (TM)" policy...

One more remark, "bind --force" does not use the System32 drivers per se. In fact, it always points to the drivers in the `usbipd-win` installation directory. This *should* always succeed (even if the System32 driver failed to install properly), but may fail for the same reasons that the installer fails to install the drivers without reboot requirement: Windows somehow decided a reboot is required. (see: https://github.com/dorssel/usbipd-win/blob/7d12fb5e3aa5cbc747557a8df05bd9ca3119292b/Usbipd/NewDev.cs#L89-L108). Now that I look at this: this should actually not contain a hardcoded path, but instead use the (possibly different) install path. Will fix that later, but it has nothing to do with your issue.

The `Usbipd` component is harvested by WiX here: https://github.com/dorssel/usbipd-win/blob/7d12fb5e3aa5cbc747557a8df05bd9ca3119292b/Installer/Installer.wixproj#L56.
The `PowerShell` component is harvested by WiX here: https://github.com/dorssel/usbipd-win/blob/7d12fb5e3aa5cbc747557a8df05bd9ca3119292b/Installer/Installer.wixproj#L65

BTW: How are you incorporating the installer? Do you call `MsiExec.exe`, or do you use the MSI API?

d0n13, 2022-05-07:
I’ve copied some of your six code into mine. As the machines are under control of IT we don’t need to install powershell etc. I also have added dependencies from our service to yours. All that works ok. Just the drivers throw the exception when I do my install. I have not modified your Drivers.wix file at all and just include it. You are probably right about the reboot. 

However, I’ve noticed a message in even viewer regarding and error with the driver install. Signing I think. I’ll get that message for you later if you don’t mind looking at it. 

Thanks.

d0n13, 2022-05-08:
```
Code integrity determined that the image hash of a file is not valid.  The file could be corrupt due to unauthorized modification or the invalid hash could indicate a potential disk device error.

File Name:	\Device\HarddiskVolume3\Windows\System32\DRVSTORE\VBoxUSBMon_AED4B03805DD8EEC2ADB76A9E4D6717788476392\VBoxUSBMon.sys	
```

```
[-](https://github.com/dorssel/usbipd-win/issues/370#) <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
[-](https://github.com/dorssel/usbipd-win/issues/370#) <System>
  <Provider Name="Microsoft-Windows-Security-Auditing" Guid="{54849625-5478-4994-a5ba-3e3b0328c30d}" /> 
  <EventID>5038</EventID> 
  <Version>0</Version> 
  <Level>0</Level> 
  <Task>12290</Task> 
  <Opcode>0</Opcode> 
  <Keywords>0x8010000000000000</Keywords> 
  <TimeCreated SystemTime="2022-05-08T16:53:46.8117800Z" /> 
  <EventRecordID>87665</EventRecordID> 
  <Correlation /> 
  <Execution ProcessID="4" ThreadID="30640" /> 
  <Channel>Security</Channel> 
  <Computer>BTG-GZGTDB3</Computer> 
  <Security /> 
  </System>
[-](https://github.com/dorssel/usbipd-win/issues/370#) <EventData>
  <Data Name="param1">\Device\HarddiskVolume3\Windows\System32\DRVSTORE\VBoxUSBMon_AED4B03805DD8EEC2ADB76A9E4D6717788476392\VBoxUSBMon.sys</Data> 
  </EventData>
  </Event>
```

@dorssel This is the error that is being logged to Event Viewer. I took the drivers from version 2.3.0 of your build and I wonder do I need to do something in windows to update those drivers correctly?

d0n13, 2022-05-09:
Just looking at your code here that you referenced... and I'm wondering if this is my issue as I'm installing usbipd alongside my app in a different directory. there will be no path to \Program Files\usbipd-win\Drivers\. Ho can this be changed to pick up where it's actually installed? From the usbip registry key where you added the install directory?

https://github.com/dorssel/usbipd-win/blob/7d12fb5e3aa5cbc747557a8df05bd9ca3119292b/Usbipd/NewDev.cs#L89-L108

dorssel, 2022-05-09:
So, you are bitten by my remark:

> Now that I look at this: this should actually not contain a hardcoded path, but instead use the (possibly different) install path. Will fix that later, but it has nothing to do with your issue.

Fortunately, I fixed it already in master, via PR #373. That will use the APPLICATIONFOLDER registry path.

dorssel, 2022-05-09:
PS: The windows event is about VBoxUSBMon, not VBoxUSB. I think you can ignore that.

d0n13, 2022-05-09:
@dorssel I just checked out master to make the modification in a PR and noticed you had fixed it :) Thank you. I have built the latest version and my basic testing have proved this change has fixed the problem.

Aside from the fact I put the wrong path in the registry when installing :/ That too is now fixed on my side.
Thanks for the support...
Donie

closed 2022-05-09 by d0n13