dorssel/usbipd-win #371: feature request - WSL auto attach based on VID:PID

ebabcock72, 2022-05-06:
Hello! I saw this PR from a few days ago #351 which enables auto-attach of a device. This is really more of an auto-reattach (that name was considered, if you look through the comments on the PR). My request is to make this even more automatic, and allow a single setup step once and then never needing to run another command.

Basically, I'd want to specify somewhere (a config file, or by running a command, or something) that any future device that matches a given `idVendor` and/or `idProduct` (VID:PID) automatically would get attached to WSL. Similar to how `udev` rules work in Linux. I have a `udev` rule that looks for a specific `idVendor` and `idProduct` and assigns it a certain name in `/dev`. 

This would allow people who use a certain device frequently (like if they are programming a certain MCU or something) to not have to worry about entering extra commands every time they attach their device. They'd just plug it in, and Linux would automatically have access to it.

I did see that you can combine the `--auto-attach` with the `--hardware-id` to automatically _reattach_ based on the VID:PID. This is **almost** what I am looking for, but you still need to manually run that command for each session and it sits there in a loop. Is there a way to have it always be waiting from a fresh boot in the background?

dorssel, 2022-05-07:
@ebabcock72 
This is not something that belongs in `usbipd-win`, as attaching is really a client-side thing. The convenience commands `usbipd wsl xxx` are special in the sense that that relay the command to run on WSL *client side*. But they are commands, not services.

Your request would require a *server side* service, that somehow monitors the running state of one or more WSL instances, and then magically runs the *client side* attach upon plugging in a device (possible even auto-binding it first). And of course all of this on a *per user* basis (as WSL instances are per user). This is all perfectly reasonable, but it is a totally different service compared to what `usbipd-win` actually does. And it would be very WSL-specific, which `usbipd-win` in itself is not. So, if this is ever made, it would be a new project altogether that would *use* `usbipd-win`, but will not be *part* of it.

This request (and the declining reasons) have been mentioned before: https://github.com/dorssel/usbipd-win/issues/56#issuecomment-1073231849

dorssel, 2022-05-13:
As indicated, this will not be implemented in `usbipd-win` since it belongs in a separate project. Now there is such a project (see #382), this issue will be closed.

closed 2022-05-13 by dorssel