dorssel/usbipd-win #382: Windows Gui Tool

andrewleech, 2022-05-11:
Hello, I just wanted to say a huge thanks for all your work here @dorssel, this is a fantastic piece of work that's making my day to day life as an embedded design engineer much easier. 
In the past I have used virtualhere to achieve similar goals, but it was always difficult to install and keep running reliably. I did however really like having a gui tool to manage the connected state of devices, and provide auto-attach based on USB port used and/or device unique ID.  So I built one, which I thought I'd share here in case others find it useful too!

https://gitlab.com/alelec/wsl-usb-gui#wsl-usb-gui
https://gitlab.com/alelec/wsl-usb-gui/-/releases
[image]

This is simply an orchestration tool for usbipd-win, a copy of which is bundled in the installer (I hope you don't mind me including it?).
The installer also makes a best-effort at updating & converting WSL if needed, as well as install the other tools required (assuming the user has ubuntu installed). So this should be a one-click installer for new users to get WSL USB devices up and running.

While the app is running it monitors for USB connect/disconnect events in windows to auto refresh and manage the auto-attach feature. The auto-attach can be configured to monitor:
* a particular device (based on the windows unique device ID)
* the specific USB port used (so you can tag a port as WSL use and anything plugged in gets forwarded)
* or both (so a particular device only gets forwarded when plugged into specific port)

The priv elevation required for first connect of new devices is handed automatically as needed.

You can also rename a device, to add your own label for it within the manager app. This is particulary helpful if you've got a bunch of devices that all show as "USB Serial Port". This rename applies a new lable within the app only, it doesn't affect windows / device manager / linux in any way.

andrewleech, 2022-05-11:
I've only just caught-up on https://github.com/dorssel/usbipd-win/issues/56 and acknowledge there's a number of limitations in my app here, mainly that it only work for the current windows users default WSL instance.  
That being said, once a USB device has been attached it does so at a kernel level, so the device is also available to WSL docker containers etc (if you bind mount `/dev`) so I've found it quite flexible in that regard. 
I don't have any support for remote clients / hyper-v etc as I wanted to keep the UI simple for now.

My main motivation for writing this app windows side was to make it easy for less-technical / non-linux users to get up and running with a gui tool that did not need linux/wsl gui set up and running (many people do not have this yet) and to deal with the admin escalation needed for first connect - linux tools can only see usb devices that have been connected manually previously from windows side.

dorssel, 2022-05-11:
@andrewleech 
Awesome!

This is exactly what I meant in https://github.com/dorssel/usbipd-win/issues/56#issuecomment-962325059.
And I actually appreciate it being a separate project, as the tooling/skills/issues will be quite distinct.
This is even more so since this has the auto-detect functionality mentioned in https://github.com/dorssel/usbipd-win/issues/371#issuecomment-1120202537 and https://github.com/dorssel/usbipd-win/issues/56#issuecomment-1073231849.

I see that it is now very WSL-centric. If you add 'bind' and 'unbind', it would be useful for Hyper-V/remote users as well...

dorssel, 2022-05-11:
@andrewleech 
Next note: I will leave this issue open for others to find. Maybe you can change the title to "Available: xxxx" or "Announcement: xxxx", so it does not look like a request.

I will at some point change the README and wiki to refer to your project, after which the issue will be migrated to a "show-and-tell" discussion, if you're OK with that.

dorssel, 2022-05-11:
@andrewleech 
Final note (sorry for the legalese):
`usbipd-win` is a derivative work, based on GPL 2 VBox drivers. That means `usbipd-win` must be GPL 2 too. Not by my choice (I prefer MIT licenses), but by the license terms of the VBox drivers. See: https://github.com/dorssel/usbipd-win/blob/master/Drivers/README.md.

I'm not a lawyer, but...
I don't know if your work qualifies as a derivative work. If it does, then it too must be GPL 2 licensed.

I'm perfectly fine that you incorporate the installer (it's free, and open source after all). But that certainly constitutes binary distribution. And that has certain implications, such as the requirement to make available the source code. I'm fine with the way it is, but there are more authors (including Microsoft), and then there is Oracle...

andrewleech, 2022-05-11:
> I see that it is now very WSL-centric. If you add 'bind' and 'unbind', it would be useful for Hyper-V/remote users as well...

I'm assuming this just binds the USB device to the usbipd service, such that remote clients can see / attach them as needed? 
Certainly this would be easy to do in code... I'll just have to think about how to represent it on the gui, ig two more buttons won't fit where the existing ones are without reorganising something, and I'd want to show bound devices differently to attached. The difference between the two would be confusing to some people too. I'm certainly not saying no to this feature (it's well worth it), this is more of a train of thought of what I'll want to solve.

> I don't know if your work qualifies as a derivative work. If it does, then it too must be GPL 2 licensed.
I also am not a lawyer, though I've spent a lot of time in the past working with / in / around these licences. I'm pretty sure it's technically not a derivative work as it's interacting with your binaries on a command line level - it does not include any code directly, nor link in any code (statically or dynamically). That being said, as you mention, I'm distributing the msi - the source of which I won't have properly declared currently. Regardless of all of this - it would certainly simplify any of these discussions if I simply match the GPL2 licence here, I'm certainly not opposed to doing that, that would also make it easier to reference this projects licence/s directly as well :-)

dorssel, 2022-05-12:
I'll suggest and describe the bind/unbind use case on your project. Seems only logical.

And about the license. I also believe yours is not a derivative work. And of course you should release your code under whatever license you whish. I don't think it is an issue at all distributing a GPL work alongside a MIT work (as long as they are not derived). Just take care of the source requirement. A README similar to mine (referring to the releases page of usbipd-win, where source tarballs are offered right next to the msi) should be sufficient.

andrewleech, 2022-05-12:
> And about the license. I also believe yours is not a derivative work. And of course you should release your code under whatever license you whish. I don't think it is an issue at all distributing a [...]

I think GPL will fit my gui project just fine, it's mainly libraries / embedded things I like to keep MIT etc so I can also use them at work easily. Standalone apps / tools are good as GPL with the extra protections that entails. I've updated my project LICENCE file as such and added references / links to the readme which should tick this off. These files are now also included in the installed app folder. Thanks for bringing this up too, while I make efforts to give credit where due, I do often forget to include LICENCE files in installers etc.

dorssel, 2022-05-21:
Added reference to https://github.com/dorssel/usbipd-win/blob/master/README.md, converting to "Show-and-tell" discussion.