dorssel/usbipd-win #384: WSL2 Instance Debian Kernel 5.10.x

silver2row, 2022-05-13:
Hello,

I am trying to use your source per MS support on their dev. blog. Anyway, I am coming across issues now for a Card Reader and SD Card. 

The card reader is plugged into the dev. desktop via USB 3.0. 

Here:

```

usbipd: info: Using default distribution 'Debian'.
usbipd: error: WSL 'usbip' client not correctly installed. See https://github.com/dorssel/usbipd-win/wiki/WSL-support for the latest instructions.

```

That is the error that arises. I looked for a particular apt install via this repo. and came up empty handed. I do not think it is available online via apt for Debian. Is this correct?

Seth

P.S. I installed the usbipd-win via the .msi available. Should I use the .tar file for WSL2 instances?

silver2row, 2022-05-13:
This is what I could not find: ` linux-tools-virtual `

Seth

dorssel, 2022-05-13:
From the wiki page:

> For other distributions a different usbip client package may be required.

https://packages.debian.org/search?keywords=usbip

silver2row, 2022-05-13:
Hello Sir...thank you. I just installed usbip via apt. I will check to see if it works. When it does, this obvious lack of reading for this issue will be closed. 

Seth

silver2row, 2022-05-13:
You rule!

closed 2022-05-13 by silver2row

oiwg, 2022-06-23:
> From the wiki page:
> 
> > For other distributions a different usbip client package may be required.
> 
> https://packages.debian.org/search?keywords=usbip

This link *really* needs to get into the documentation at https://docs.microsoft.com/en-us/windows/wsl/connect-usb ..... perhaps with a table of client package names for the most popular distributions?

dorssel, 2022-06-23:
@oiwg 
If so, please consider filing an issue at https://github.com/microsoft/WSL. This project does not control that documentation, and I am not Microsoft.