dorssel/usbipd-win #390: unable load firmware

rinrin-, 2022-05-23:
Hi:
  I found that when usb wifi adapter attached to wsl2, the kernel cannot load  firmware from filesystem. 
System Version:
Microsoft Windows [Version 10.0.22000.651]
Kernel Version:
Linux version 5.10.102.1-microsoft-standard-WSL2
Repro Steps:
run the following command in cmd:
C:\> usbipd wsl attach -b 1-X
after usb driver init, load another mod
# modprobe carl9170
firmware carl9170-1.fw can not be loaded. kernel log after enable dynamic debug shows: 
[ 719.134445] usb 1-1: new high-speed USB device number 4 using vhci_hcd
[ 719.284377] usb 1-1: SetAddress Request (4) to port 0
[ 719.349176] usb 1-1: New USB device found, idVendor=0846, idProduct=9001, bcdDevice= 1.06
[ 719.349185] usb 1-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[ 719.349189] usb 1-1: Product: USB2.0 WLAN
[ 719.349191] usb 1-1: Manufacturer: ATHER
[ 719.349193] usb 1-1: SerialNumber: 12345
[ 719.494320] usb 1-1: reset high-speed USB device number 4 using vhci_hcd
[ 719.644405] usb 1-1: SetAddress Request (4) to port 0
[ 719.839554] firmware_class: __allocate_fw_priv: fw-carl9170-1.fw fw_priv=000000009a54a0a0
[ 719.839579] usb 1-1: loading /lib/firmware/updates/5.10.102.1-microsoft-standard-WSL2/carl9170-1.fw failed for no such file or directory.
[ 719.839585] usb 1-1: loading /lib/firmware/updates/carl9170-1.fw failed for no such file or directory.
[ 719.839590] usb 1-1: loading /lib/firmware/5.10.102.1-microsoft-standard-WSL2/carl9170-1.fw failed for no such file or directory.
[ 719.839594] usb 1-1: loading /lib/firmware/carl9170-1.fw failed for no such file or directory.
[ 719.839598] usb 1-1: Direct firmware load for carl9170-1.fw failed with error -2
[ 719.839603] firmware_class: __free_fw_priv: fw-carl9170-1.fw fw_priv=000000009a54a0a0 data=0000000000000000 size=0
[ 719.839608] usb 1-1: firmware not found.

but firmware is present in /lib/modules:
wsl2 ~ # ls -l /lib/firmware/carl9170-1.fw
-rw-r--r-- 1 root root 13388 4月26日 12:21 /lib/firmware/carl9170-1.fw

BTW, I checked  firmware load code, it shows kernel read file from init namespace
(drivers/base/firmware_loader/main.c)
/* load firmware files from the mount namespace of init */ rc = kernel_read_file_from_path_initns(path, fw_priv->offset, &buffer, msize, file_size_ptr, READING_FIRMWARE);
the "/init" file is mounted with 9p protocol:
tools on /init type 9p (ro,relatime,dirsync,aname=tools;fmask=022,loose,access=client,msize=65536,trans=fd,rfd=7,wfd=7)

dorssel, 2022-05-23:
The problem is that the *kernel* needs to be able to load the firmware, not the distribution. So even if you have installed `linux-firmware`, then it is still not available in the filesystem that the kernel is using. WSL works by having a *single* kernel running on a hidden VM. The distros are just like a docker, with their own filesystem, but all sharing the same kernel.

I'm sorry to refer you back to WSL support, but I have no clue how you can add files to the "WSL root", which is where the kernel is looking for these firmware files.

mkeydevelop, 2022-07-02:
Copy firmware file to C:\Windows\System32\lxss\tools, and add "/tools/firmware" into firmware loading path can avoid this error

rinrin-, 2022-07-13:
> Copy firmware file to C:\Windows\System32\lxss\tools, and add "/tools/firmware" into firmware loading path can avoid this error

Awesome it works! just quick reference for .wslconfig

```
[wsl2]
kernelCommandLine="firmware_class.path=/tools/firmware"
```
or shell command
`echo -n "/tools/firmware" > /sys/module/firmware_class/parameters/path`

rinrin-, 2022-07-25:
new version of wsl2(0.61.8.0) changes "/init" mount behavior
`rootfs on /init type rootfs (rw,size=3036424k,nr_inodes=759106)`

MohamSayed, 2022-08-06:
### Trying to mount AR9271, it doesn't give clear indication of is the issue. 
**Hint:** Tried all above solution
Ä 2117.050328Å usb 1-1: USB disconnect, device number 5
Ä 2124.833932Å vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
Ä 2124.834779Å vhci_hcd vhci_hcd.0: devid(196618) speed(3) speed_str(high-speed)
Ä 2124.835690Å vhci_hcd vhci_hcd.0: Device attached
Ä 2125.181212Å usb 1-1: new high-speed USB device number 6 using vhci_hcd
Ä 2125.341205Å usb 1-1: SetAddress Request (6) to port 0
Ä 2125.386704Å usb 1-1: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
Ä 2125.387784Å usb 1-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
Ä 2125.388702Å usb 1-1: Product: USB2.0 WLAN
Ä 2125.389178Å usb 1-1: Manufacturer: ATHEROS
Ä 2125.389644Å usb 1-1: SerialNumber: 12345
Ä 2125.392163Å usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.dev.0.fw requested
Ä 2125.393319Å usb 1-1: Direct firmware load for ath9k_htc/htc_9271-1.dev.0.fw failed with error -2
Ä 2125.394511Å usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
Ä 2125.395472Å usb 1-1: Direct firmware load for ath9k_htc/htc_9271-1.4.0.fw failed with error -2
Ä 2125.396546Å usb 1-1: ath9k_htc: Firmware htc_9271.fw requested
Ä 2125.397469Å usb 1-1: Direct firmware load for htc_9271.fw failed with error -2
Ä 2125.398370Å usb 1-1: no suitable firmware found!
Ä 2125.399172Å usb 1-1: ath9k_htc: Failed to get firmware htc_9271.fw
Ä 2125.400131Å usb 1-1: ath9k_htc: USB layer deinitialized

MohamSayed, 2022-08-11:
> ### Trying to mount AR9271, it doesn't give me clear indication of what is the issue.
> **Hint:** Tried all above solution Ä 2117.050328Å usb 1-1: USB disconnect, device number 5 Ä 2124.833932Å vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3) Ä 2124.834779Å vhci_hcd vhci_hcd.0: devid(196618) speed(3) speed_str(high-speed) Ä 2124.835690Å vhci_hcd vhci_hcd.0: Device attached Ä 2125.181212Å usb 1-1: new high-speed USB device number 6 using vhci_hcd Ä 2125.341205Å usb 1-1: SetAddress Request (6) to port 0 Ä 2125.386704Å usb 1-1: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08 Ä 2125.387784Å usb 1-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48 Ä 2125.388702Å usb 1-1: Product: USB2.0 WLAN Ä 2125.389178Å usb 1-1: Manufacturer: ATHEROS Ä 2125.389644Å usb 1-1: SerialNumber: 12345 Ä 2125.392163Å usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.dev.0.fw requested Ä 2125.393319Å usb 1-1: Direct firmware load for ath9k_htc/htc_9271-1.dev.0.fw failed with error -2 Ä 2125.394511Å usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested Ä 2125.395472Å usb 1-1: Direct firmware load for ath9k_htc/htc_9271-1.4.0.fw failed with error -2 Ä 2125.396546Å usb 1-1: ath9k_htc: Firmware htc_9271.fw requested Ä 2125.397469Å usb 1-1: Direct firmware load for htc_9271.fw failed with error -2 Ä 2125.398370Å usb 1-1: no suitable firmware found! Ä 2125.399172Å usb 1-1: ath9k_htc: Failed to get firmware htc_9271.fw Ä 2125.400131Å usb 1-1: ath9k_htc: USB layer deinitialized

## Finally, I managed to make it work💪 by embedding the firmware into the kernel
[image]
[image]

rtritto, 2022-08-19:
@MohamSayed
How can I enable the interface _wlan0_?

I have `htc_9271-1.4.0.fw` in _/lib/firmware/ath9k_htc_.

dmesg
```console
vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
vhci_hcd vhci_hcd.0: devid(65544) speed(3) speed_str(high-speed)
vhci_hcd vhci_hcd.0: Device attached
usb 1-1: new high-speed USB device number 2 using vhci_hcd
usb 1-1: SetAddress Request (2) to port 0
usb 1-1: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
usb 1-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
usb 1-1: Product: UB91C
usb 1-1: Manufacturer: ATHEROS
usb 1-1: SerialNumber: 12345
```

iwconfig
```console
lo        no wireless extensions.
bond0     no wireless extensions.
dummy0    no wireless extensions.
tunl0     no wireless extensions.
sit0      no wireless extensions.
eth0      no wireless extensions.
```

MohamSayed, 2022-08-23:
I followed this example: https://wiki.gentoo.org/wiki/Radeon#Built-in_kernel

1. Put the firmware file directory (ath9k_htc) in the kernel source code directory. Can be found in /lib/firmware/  just copy it!
[image]
2.  Disable this option
[image]
3. Set value to the firmware file we copied at step **1** .  Second value, points to /lib/firmware/
[image]

4. Enabling the module
[image]
[image]

dorssel, 2022-11-02:
@rinrin- 
You re-opened this issue when a new version WSL2 changed the behavior.
Are you able to load firmware again? Do the instructions provided by @MohamSayed solve your problem?

dorssel, 2022-12-17:
Stale

closed 2022-12-17 by dorssel

super100pig, 2024-01-20:
> @rtritto I followed this example: https://wiki.gentoo.org/wiki/Radeon#Built-in_kernel
> 
> 1. Put the firmware file directory (ath9k_htc) in the kernel source code directory. Can be found in /lib/firmware/  just copy it!
>    [image]
> 2. Disable this option
>    [image]
> 3. Set value to the firmware file we copied at step **1** .  Second value, points to /lib/firmware/
>    [image]
> 4. Enabling the module
>    [image]
>    [image]

Copy the firmware files into the kernel source code directory is not necessary. Only the firmware in /lib/firmware will be built in the kernel. Just make sure that the firmware in /lib/firmware.

thardie, 2025-01-27:
I tried the method of compiling into the kernel, and when the driver tried to load the firmware, I get a kernel panic.

```
[   19.281714] dvb-usb: no frontend was attached by 'TurboSight TBS 5520SE'
[   19.291022] BUG: kernel NULL pointer dereference, address: 00000000000000fb
[   19.291524] #PF: supervisor write access in kernel mode
[   19.291877] #PF: error_code(0x0002) - not-present page
[   19.292136] PGD 0 P4D 0
[   19.292257] Oops: 0002 [#1] SMP NOPTI
[   19.292535] CPU: 2 PID: 1822 Comm: systemd-udevd Tainted: P           O      5.15.167.4-microsoft-standard-WSL2+ #16
[   19.293200] RIP: 0010:si2183_probe+0xb8/0x2c0 [si2183]
[   19.293543] Code: 05 90 c0 49 8d 7e 08 45 31 ed e8 83 e2 63 f3 49 89 9e 60 03 00 00 48 8b 55 50 48 8d 7b 20 49 8b 44 24 08 48 c7 c6 30 04 90 c0 <48> 89 10 49 8b 04 24 4c 89 30 41 0f b6 44 24 10 41 88 86 75 05 00
[   19.294595] RSP: 0018:ffffaa8c03a3fc88 EFLAGS: 00010246
[   19.294885] RAX: 00000000000000fb RBX: ffff9e14ac30cc00 RCX: 0000000000000000
[   19.296080] RDX: ffff9e14939d2800 RSI: ffffffffc0900430 RDI: ffff9e14ac30cc20
[   19.296407] RBP: ffff9e147e25b720 R08: 0000000000000000 R09: ffffaa8c03a3f948
[   19.296846] R10: 0000000000000001 R11: 0000000000000001 R12: ffffaa8c0440fab8
[   19.298149] R13: 0000000000000000 R14: ffff9e147e91d000 R15: ffffffffc09020f0
[   19.298550] FS:  00007fa4805ce8c0(0000) GS:ffff9e1b1e680000(0000) knlGS:0000000000000000
[   19.298918] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   19.299215] CR2: 00000000000000fb CR3: 0000000110270004 CR4: 0000000000370ea0
[   19.299585] Call Trace:
[   19.299708]  <TASK>
[   19.299827]  ? __die_body.cold+0x1a/0x1f
[   19.300012]  ? page_fault_oops+0xae/0x250
[   19.300205]  ? dev_printk_emit+0x4e/0x69
[   19.300352]  ? exc_page_fault+0x5d/0x100
[   19.300492]  ? asm_exc_page_fault+0x22/0x30
[   19.300641]  ? si2183_probe+0xb8/0x2c0 [si2183]
[   19.300827]  ? si2183_probe+0x9d/0x2c0 [si2183]
[   19.301124]  i2c_device_probe+0xad/0x210
[   19.301291]  really_probe.part.0+0xa2/0x290
[   19.302267]  driver_probe_device+0x1e/0x90
[   19.302638]  __driver_attach+0x86/0x120
[   19.302770]  ? __device_attach_driver+0xc0/0xc0
[   19.302989]  bus_for_each_dev+0x75/0xc0
[   19.303174]  bus_add_driver+0x11e/0x1d0
[   19.303497]  driver_register+0x8f/0xf0
[   19.303771]  ? 0xffffffffc08b0000
[   19.303996]  i2c_register_driver+0x3f/0xb0
[   19.305665]  do_one_initcall+0x41/0x1d0
[   19.305842]  ? kmem_cache_alloc_trace+0x2f/0x390
[   19.306046]  do_init_module+0x4c/0x240
[   19.306218]  __do_sys_finit_module+0xa0/0xe0
[   19.306411]  do_syscall_64+0x35/0xb0
[   19.306561]  entry_SYSCALL_64_after_hwframe+0x6c/0xd6
[   19.306775] RIP: 0033:0x7fa480cc888d
[   19.306974] Code: 5b 41 5c c3 66 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 73 b5 0f 00 f7 d8 64 89 01 48
[   19.307922] RSP: 002b:00007ffe2192c478 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[   19.308300] RAX: ffffffffffffffda RBX: 000055c8ee1aadc0 RCX: 00007fa480cc888d
[   19.308660] RDX: 0000000000000000 RSI: 00007fa480e60441 RDI: 000000000000000f
[   19.308952] RBP: 0000000000020000 R08: 0000000000000000 R09: 0000000000000002
[   19.309473] R10: 000000000000000f R11: 0000000000000246 R12: 00007fa480e60441
[   19.309837] R13: 000055c8ee32c390 R14: 000055c8ee1ab420 R15: 000055c8ee330a10
[   19.310208]  </TASK>
[   19.310333] Modules linked in: si2183(O+) i2c_mux dvb_usb_tbs5520se(O) dvb_usb(O) dvb_core(O) videobuf2_vmalloc(O) videobuf2_memops(O) videobuf2_common(O) mc(O) rc_core zfs(PO) spl(O) configfs
[   19.311218] CR2: 00000000000000fb
[   19.311364] ---[ end trace f2dc0168c8324c79 ]---
[   19.311585] RIP: 0010:si2183_probe+0xb8/0x2c0 [si2183]
[   19.311796] Code: 05 90 c0 49 8d 7e 08 45 31 ed e8 83 e2 63 f3 49 89 9e 60 03 00 00 48 8b 55 50 48 8d 7b 20 49 8b 44 24 08 48 c7 c6 30 04 90 c0 <48> 89 10 49 8b 04 24 4c 89 30 41 0f b6 44 24 10 41 88 86 75 05 00
[   19.312710] RSP: 0018:ffffaa8c03a3fc88 EFLAGS: 00010246
[   19.312914] RAX: 00000000000000fb RBX: ffff9e14ac30cc00 RCX: 0000000000000000
[   19.313213] RDX: ffff9e14939d2800 RSI: ffffffffc0900430 RDI: ffff9e14ac30cc20
[   19.313548] RBP: ffff9e147e25b720 R08: 0000000000000000 R09: ffffaa8c03a3f948
[   19.313816] R10: 0000000000000001 R11: 0000000000000001 R12: ffffaa8c0440fab8
[   19.314162] R13: 0000000000000000 R14: ffff9e147e91d000 R15: ffffffffc09020f0
[   19.314469] FS:  00007fa4805ce8c0(0000) GS:ffff9e1b1e680000(0000) knlGS:0000000000000000
[   19.314809] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   19.315048] CR2: 00000000000000fb CR3: 0000000110270004 CR4: 0000000000370ea0
[   19.315318] Kernel panic - not syncing: Fatal exception
[   19.335692] Kernel Offset: 0x32000000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
```

lars18th, 2025-03-14:
> new version of wsl2(0.61.8.0) changes "/init" mount behavior `rootfs on /init type rootfs (rw,size=3036424k,nr_inodes=759106)`

Whats the solution now?
```
[wsl2]
kernelCommandLine="firmware_class.path=/tools/firmware"
```
Doesnt' work with recent kernels.