dorssel/usbipd-win #394: auto attach multiple vid:pid on connect?

briskspirit, 2022-05-27:
Hey, thank you for your great work!

I have a question about auto attach:
My device might have 3 different vid:pid combo depending on state: active, bootloader, DFU mode. Software connects to device in active state and puts it into bootloader mode. if failed - into DFU. So is there a good way of doing auto attach? 
The only idea that comes into my mind is to write a python script that checks output of usbipd wsl list and statuses and executes appropriate command. Or I am missing something and there is a better way ? Thank you.

briskspirit, 2022-05-27:
foound.
closed with #371

closed 2022-05-27 by briskspirit

rrakso, 2024-07-29:
I also found this useful https://github.com/dorssel/usbipd-win/discussions/910#discussioncomment-8960963