dorssel/usbipd-win #395: Packages not found

willzyx-hub, 2022-05-27:
Hello dorssel i wanna to ask you that in wiki it says to install linux-tools-virtual but however it doesn't exist on all distro so would you like to give me the correct package name

here the log :

willy@AsuTek-VBL-X515 ~ $ sudo emerge --ask linux-tools-virtual

These are the packages that would be merged, in order:

Calculating dependencies... done!

emerge: there are no ebuilds to satisfy "linux-tools-virtual".

emerge: searching for similar names...
emerge: Maybe you meant any of these: app-text/linuxdoc-tools, sec-policy/selinux-googletalk, sec-policy/selinux-virt?

also

without installing it it doesn't work like this

C:\Users\willi>wsl --list --verbose
  NAME      STATE           VERSION
  Gentoo    Running         2

C:\Users\willi>usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-1    1a2c:4c5e  USB Input Device                                              Not shared
1-5    13d3:5a11  USB2.0 VGA UVC WebCam                                         Not shared
1-6    0781:5567  USB Mass Storage Device                                       Not shared
1-10   8087:0aaa  Intel(R) Wireless Bluetooth(R)                                Not shared
2-3    0250:3412  USB Input Device                                              Not shared

Persisted:
GUID                                  DEVICE

C:\Users\willi>usbipd wsl attach -b 1-6
usbipd: error: No default WSL distribution exists.

dorssel, 2022-05-27:
@willzyx-hub 
Yes, the Wiki states that this differs per distribution. And I don't know all distributions. Besides, you didn't tell me which distribution you use...

> usbipd: error: No default WSL distribution exists.

This is true, the list from `wsl --list --verbose` does not indicate a default (it would be marked with a '*'). So you will either have to indicate the intended distribution with `usbipd list --distribution Gentoo` (see `usbipd list --help`) or mark the distribution as default with `wsl --set-default Gentoo` (see `wsl --help`).

dorssel, 2022-05-27:
See https://www.google.com/search?q=gentoo+usbip

willzyx-hub, 2022-05-27:
> See https://www.google.com/search?q=gentoo+usbip

Ok i will take a try and i will inform you when it's Done

willzyx-hub, 2022-05-27:
> See https://www.google.com/search?q=gentoo+usbip

however it doesn't work

willy@AsuTek-VBL-X515 ~ $ sudo emerge --ask --verbose --autounmask-write --getbinpkg=n usbip

These are the packages that would be merged, in order:

Calculating dependencies... done!

!!! All ebuilds that could satisfy "usbip" have been masked.
!!! One of the following masked packages is required to complete your request:
- net-misc/usbip-5.15::gentoo (masked by: ~amd64 keyword)

For more information, see the MASKED PACKAGES section in the emerge
man page or refer to the Gentoo Handbook.

dorssel, 2022-05-27:
Great, so you found the answer! Can the issue be closed?

willzyx-hub, 2022-05-27:
> Great, so you found the answer! Can the issue be closed?

Ah i guess so