dorssel/usbipd-win #399: Usbipd worked this morning.  Now 'usbipd: error: WSL 'usbip' client not correctly installed'

quixotictech-dev, 2022-05-30:
After some struggle I finally got usbipd working correctly with my WSL2 install.  However,  it t seems to have been shortlived.  After reinstalling I still experience this error which is just bizarre.  This morning I was using ADB through Ubuntu but now I can't launch the program.

dorssel, 2022-05-30:
What distro? Ubuntu? Did you run updates? Maybe a new `usbip` was installed and you have to re-run `update-alternatives`...

Jordan-Anjie-Lu, 2022-06-14:
Encountered a similar issue. In my instance, `sudo update-alternatives --install /usr/local/bin/usbip usbip /usr/lib/linux-tools/*/usbip 20` wasn't installing properly as my distribution somehow had done...something to itself. See image for reference:
[image]

Fixed it locally by locating the new location of the file (in my installation, files in `/usr/lib/linux-tools/5.4.0-117-generic` were pointing to files in `/usr/lib/linux-tools-5.4.0-117/`) and modifying the above `sudo` command to match. 
[image]

dorssel, 2022-06-14:
@Jordan-Anjie-Lu 
See https://github.com/dorssel/usbipd-win/issues/391#issuecomment-1135924535

This happens if you have multiple versions of usbip installed. Can you try the command in the link to see if that solves the problem. If so, I will update the wiki.

Jordan-Anjie-Lu, 2022-06-15:
@dorssel Looks like the command in that link works. Thank you for creating this project!

dorssel, 2022-06-15:
I've updated the WSL wiki instructions.

@quixotictech-dev 
Has your issue been solved?

dorssel, 2022-08-23:
Closing (stale).

closed 2022-08-23 by dorssel