dorssel/usbipd-win #402: Possible to share devices to AWS using usbipd-win?

d0n13, 2022-06-01:
Hi all

Trying to share a device to AWS linux install using usbipd and I'm wondering if anyone has got this to work over an SSH tunnel? I have tried to create the tunnel using the following but I'm not having any luck so far. 

```
ssh -i ./usbip-server.pem -Nf -R 3240:localhost:3240 ubuntu@ec2-xx.xx.xx.xx.eu-west-1.compute.amazonaws.com
```

Apart from the tunnel from being setup wrong will this even work? I know usbipd is already listening at port 3240 on the local machine so I wonder can I do something else to configure this to operate?

d0n13, 2022-06-02:
Found a good guide here for Ubuntu 20.04 and it worked. 
https://askubuntu.com/questions/1303403/how-to-install-usbip-vhci-hcd-drivers-on-an-aws-ec2-ubuntu-kernel-version
Bit tricky to setup but the instructions are ok. Just remember you have to be root on the AWS side to execute the attach...

First created the SSH tunnel as follows:
```
 ssh -i .\ubuntu20.04.pem -R 2001:localhost:3240 ubuntu@ec2-xx-xx-xx-xx.eu-west-1.compute.amazonaws.com
```

Shared an iPhone on the Windows PC
```
PS C:\Users\donie> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-10   27c6:533c  Goodix fingerprint                                            Not shared
1-11   0bda:5510  Integrated Webcam, Camera DFU Device                          Not shared
1-14   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not shared
3-5    0bda:4014  Realtek USB Audio                                             Not shared
3-7    046d:c52b  Logitech USB Input Device, USB Input Device                   Not shared
4-2    0bda:8153  Realtek USB GbE Family Controller                             Not shared
6-1    05ac:12a8  Apple Mobile Device USB Composite Device                      Shared (forced)
```
Then attach on AWS side.

```
ubuntu@ip-xx-xx-xx-xx:~$ sudo usbip --tcp-port 2001 attach -r 127.0.0.1 -b 6-1
usbip: info: using port 2001 ("2001")
ubuntu@ip-172-31-35-88:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Installed libimobiledevice and was able to connect to the phone and retrieve details.
Thanks @dorssel for the tips. Were very helpful

dorssel, 2022-06-03:
@d0n13 
Is it OK if I close this by converting it to a "show and tell" discussion?
And change the title to "How to share devices to Amazon Web Services (AWS)"?

d0n13, 2022-06-03:
Yes. Please do. I was waiting to see what u needed to do