dorssel/usbipd-win #408: pyaudio support?

automateyournetwork, 2022-06-06:
Hello! I have followed the instructions and my USB microphone is listed in WSL2 Ubuntu 

lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0d8c:0005 C-Media Electronics, Inc. Blue Snowball

However pyaudio does not pick up the USB microphone / I can't seem to record audio 

pyaudio 'works' and it does try to record - I am looking for guidance on how to tell ALSA about the USB microphone 

Here is the log from my simply python script 

ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5220:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM sysdefault
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5220:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM sysdefault
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5220:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM dmix
Recording
Finished recording

(The WAV file is 'empty'/silent) 

Some other information - arecord -l does not find any soundcard

arecord -l
arecord: device_list:276: no soundcards found...

When I modprobe for snd-usb-audio I get a FATAL error that the module is not found

sudo modprobe snd-usb-audio
modprobe: FATAL: Module snd-usb-audio not found in directory /lib/modules/5.10.102.1-microsoft-standard-WSL2

Now that I've attached my USB microphone to WSL2 Ubuntu - "how" do I get pyaudio to detect it as my input?

Thanks in advance

dorssel, 2022-06-06:
Did you compile your own kernel? You should. USB audio drivers are not in the default WSL kernel. See https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel.

automateyournetwork, 2022-06-07:
Thank you for the reply I will try this!

automateyournetwork, 2022-06-08:
I followed the instructions up to here

cp arch/x86/boot/bzImage /mnt/c/Users/<user>/usbip-bzImage

There is no bzImage file ? All of the other steps were followed but I can't find the bzImage file - please advise thank you!

dorssel, 2022-06-09:
Does https://github.com/dorssel/usbipd-win/discussions/404 help?

automateyournetwork, 2022-06-09:
Hello - yes I have the same symptoms as #404. I've tried it again from scratch and have the same issue 

The only thing of note is after this step 

sudo make -j 8 && sudo make modules_install -j 8 && sudo make install -j 8

I get the following errors during this step

CC      lib/radix-tree.o
net/ipv4/netfilter/ipt_ECN.c:26:46: warning: ‘struct ipt_ECN_info’ declared inside parameter list will not be visible outside of this definition or declaration
   26 | set_ect_ip(struct sk_buff *skb, const struct ipt_ECN_info *einfo)
      |                                              ^~~~~~~~~~~~
net/ipv4/netfilter/ipt_ECN.c: In function ‘set_ect_ip’:
net/ipv4/netfilter/ipt_ECN.c:30:44: error: dereferencing pointer to incomplete type ‘const struct ipt_ECN_info’
   30 |  if ((iph->tos & IPT_ECN_IP_MASK) != (einfo->ip_ect & IPT_ECN_IP_MASK)) {
      |                                            ^~
net/ipv4/netfilter/ipt_ECN.c: At top level:
net/ipv4/netfilter/ipt_ECN.c:45:47: warning: ‘struct ipt_ECN_info’ declared inside parameter list will not be visible outside of this definition or declaration
   45 | set_ect_tcp(struct sk_buff *skb, const struct ipt_ECN_info *einfo)
      |                                               ^~~~~~~~~~~~
net/ipv4/netfilter/ipt_ECN.c: In function ‘set_ect_tcp’:
net/ipv4/netfilter/ipt_ECN.c:55:14: error: dereferencing pointer to incomplete type ‘const struct ipt_ECN_info’
   55 |  if ((!(einfo->operation & IPT_ECN_OP_SET_ECE) ||
      |              ^~
net/ipv4/netfilter/ipt_ECN.c:55:28: error: ‘IPT_ECN_OP_SET_ECE’ undeclared (first use in this function); did you mean ‘IPT_ECN_OP_MATCH_ECE’?
   55 |  if ((!(einfo->operation & IPT_ECN_OP_SET_ECE) ||
      |                            ^~~~~~~~~~~~~~~~~~
      |                            IPT_ECN_OP_MATCH_ECE
net/ipv4/netfilter/ipt_ECN.c:55:28: note: each undeclared identifier is reported only once for each function it appears in
net/ipv4/netfilter/ipt_ECN.c:57:28: error: ‘IPT_ECN_OP_SET_CWR’ undeclared (first use in this function); did you mean ‘IPT_ECN_OP_MATCH_CWR’?
   57 |      (!(einfo->operation & IPT_ECN_OP_SET_CWR) ||
      |                            ^~~~~~~~~~~~~~~~~~
      |                            IPT_ECN_OP_MATCH_CWR
net/ipv4/netfilter/ipt_ECN.c: In function ‘ecn_tg’:
net/ipv4/netfilter/ipt_ECN.c:81:11: error: dereferencing pointer to incomplete type ‘const struct ipt_ECN_info’
   81 |  if (einfo->operation & IPT_ECN_OP_SET_IP)
      |           ^~
net/ipv4/netfilter/ipt_ECN.c:81:25: error: ‘IPT_ECN_OP_SET_IP’ undeclared (first use in this function); did you mean ‘IPT_ECN_OP_MATCH_IP’?
   81 |  if (einfo->operation & IPT_ECN_OP_SET_IP)
      |                         ^~~~~~~~~~~~~~~~~
      |                         IPT_ECN_OP_MATCH_IP
net/ipv4/netfilter/ipt_ECN.c:82:24: error: passing argument 2 of ‘set_ect_ip’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   82 |   if (!set_ect_ip(skb, einfo))
      |                        ^~~~~
      |                        |
      |                        const struct ipt_ECN_info *
net/ipv4/netfilter/ipt_ECN.c:26:60: note: expected ‘const struct ipt_ECN_info *’ but argument is of type ‘const struct ipt_ECN_info *’
   26 | set_ect_ip(struct sk_buff *skb, const struct ipt_ECN_info *einfo)
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
net/ipv4/netfilter/ipt_ECN.c:85:26: error: ‘IPT_ECN_OP_SET_ECE’ undeclared (first use in this function); did you mean ‘IPT_ECN_OP_MATCH_ECE’?
   85 |  if (einfo->operation & (IPT_ECN_OP_SET_ECE | IPT_ECN_OP_SET_CWR) &&
      |                          ^~~~~~~~~~~~~~~~~~
      |                          IPT_ECN_OP_MATCH_ECE
net/ipv4/netfilter/ipt_ECN.c:85:47: error: ‘IPT_ECN_OP_SET_CWR’ undeclared (first use in this function); did you mean ‘IPT_ECN_OP_MATCH_CWR’?
   85 |  if (einfo->operation & (IPT_ECN_OP_SET_ECE | IPT_ECN_OP_SET_CWR) &&
      |                                               ^~~~~~~~~~~~~~~~~~
      |                                               IPT_ECN_OP_MATCH_CWR
net/ipv4/netfilter/ipt_ECN.c:87:25: error: passing argument 2 of ‘set_ect_tcp’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   87 |   if (!set_ect_tcp(skb, einfo))
      |                         ^~~~~
      |                         |
      |                         const struct ipt_ECN_info *
net/ipv4/netfilter/ipt_ECN.c:45:61: note: expected ‘const struct ipt_ECN_info *’ but argument is of type ‘const struct ipt_ECN_info *’
   45 | set_ect_tcp(struct sk_buff *skb, const struct ipt_ECN_info *einfo)
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
net/ipv4/netfilter/ipt_ECN.c: In function ‘ecn_tg_check’:
net/ipv4/netfilter/ipt_ECN.c:98:11: error: dereferencing pointer to incomplete type ‘const struct ipt_ECN_info’
   98 |  if (einfo->operation & IPT_ECN_OP_MASK)
      |           ^~
net/ipv4/netfilter/ipt_ECN.c:98:25: error: ‘IPT_ECN_OP_MASK’ undeclared (first use in this function); did you mean ‘IPT_ECN_IP_MASK’?
   98 |  if (einfo->operation & IPT_ECN_OP_MASK)
      |                         ^~~~~~~~~~~~~~~
      |                         IPT_ECN_IP_MASK
net/ipv4/netfilter/ipt_ECN.c:104:27: error: ‘IPT_ECN_OP_SET_ECE’ undeclared (first use in this function); did you mean ‘IPT_ECN_OP_MATCH_ECE’?
  104 |  if ((einfo->operation & (IPT_ECN_OP_SET_ECE|IPT_ECN_OP_SET_CWR)) &&
      |                           ^~~~~~~~~~~~~~~~~~
[... cut ...]
      |                                              ^~~~~~~~~~~~~~~~~~
      |                                              IPT_ECN_OP_MATCH_CWR
net/ipv4/netfilter/ipt_ECN.c: At top level:
net/ipv4/netfilter/ipt_ECN.c:116:23: error: invalid application of ‘sizeof’ to incomplete type ‘struct ipt_ECN_info’
  116 |  .targetsize = sizeof(struct ipt_ECN_info),
      |                       ^~~~~~
cc1: some warnings being treated as errors

But otherwise I do all of the other steps as described but there is no bzImage file in /arch/x86/boot 

Any ideas?

Thanks

dorssel, 2022-06-10:
Google: https://stackoverflow.com/questions/62705192/building-own-kernel-and-loading-it

dorssel, 2022-08-23:
Closing (stale).

closed 2022-08-23 by dorssel