dorssel/usbipd-win #409: usb flash drive listed by lsusb , not by lsblk, even after kernel compilation

snowgato, 2022-06-06:
Hello,
I'd like to use a flash drive with wsl2.
I've installed ubuntu20-04. Kernel is 5.10.102.1-microsoft-standard-WSL2
I've compiled the kernel including almost all possible builtin USB modules.
The compilation works fine. 
I also typed:
sudo apt install linux-tools-5.4.0-77-generic hwdata
sudo update-alternatives --install /usr/local/bin/usbip usbip /usr/lib/linux-tools/5.4.0-77-generic/usbip 20 
I succeed to attach both USB3 flash drive and ssd , but none is listed by lsblk.
Below is the .config used to compile the kernel  
[config_wsl.txt](https://github.com/dorssel/usbipd-win/files/8847333/config_wsl.txt)
Thanks very much for your help

snowgato, 2022-06-06:
Well, actually, I made a mistake on the .wslconfig file name, that I first named .wslconf .
Now that I have correctly named the .wslconfig file specifying the linux kernel, lsblk shows both the USB3 flash drive and SSD drive.
Nevertheless, I don't succeed to mount it .
With the flash drive connected:

fdisk -l:

Disk /dev/sdd: 460.27 GiB, 494206451712 bytes, 965246976 sectors
Disk model:  SanDisk 3.2Gen1
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device     Boot Start       End   Sectors   Size Id Type
/dev/sdd1          32 965246975 965246944 460.3G  c W95 FAT32 (LBA)

sudo mount /dev/sdd1 /mnt/UltraFit/ -t vfat -o rw
Seems stalled, needs 20-30 seconds to finish
cd /mnt/UltraFit/
ls 
then no output...

dmesg:
[dmesg.txt](https://github.com/dorssel/usbipd-win/files/8848019/dmesg.txt)

Thanks for your help

dorssel, 2022-06-06:
This is a known issue with SanDisk Ultra Fit drives, see https://github.com/dorssel/usbipd-win/wiki/Tested-Devices. Seems unfixable. It may work on a USB2 port.

snowgato, 2022-06-06:
Hello, thanks for your answer.
Is it the same with Sandisk external usb ssd like SanDisk ExtremeSSD ?

dorssel, 2022-06-06:
Don't know, what's you device ID? The Extreme Pro has been reported working.

snowgato, 2022-06-06:
lsusb:
Bus 001 Device 003: ID 0781:558c SanDisk Corp. Extreme SSD

fdisk -l:
Disk /dev/sde: 931.53 GiB, 1000204140544 bytes, 1953523712 sectors
Disk model: Extreme SSD
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 1048576 bytes
Disklabel type: dos
Disk identifier: 0x0073415a

Device     Boot Start        End    Sectors   Size Id Type
/dev/sde1        2048 1953523711 1953521664 931.5G  7 HPFS/NTFS/exFAT

$ sudo mount -t exfat -o rw /dev/sde1 /mnt/ExtremeDiskSSD/
FUSE exfat 1.3.0
stalls and then 
ERROR: failed to read cluster 0x4.

Seems it doesn't work neither with SanDisk ExtremeSSD
ERROR: failed to read boot sector.
Disk /dev/sde: 931.53 GiB, 1000204140544 bytes, 1953523712 sectors
Disk model: Extreme SSD
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 1048576 bytes
Disklabel type: dos
Disk identifier: 0x0073415a

Device     Boot Start        End    Sectors   Size Id Type
/dev/sde1        2048 1953523711 1953521664 931.5G  7 HPFS/NTFS/exFAT

dorssel, 2022-06-06:
Is this on a USB2 or USB3 port?
Can you do a capture as explained in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting#usb-capture?

dorssel, 2022-06-06:
FYI: Here's some interesting report about SanDisk stalls: https://www.downtowndougbrown.com/2018/12/usb-mass-storage-with-embedded-devices-tips-and-quirks/#comment-465860

snowgato, 2022-06-17:
It doesn't work neither plugging the usb key in a USB 2 port.
I've captured data as explained in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting#usb-capture:

PS C:\Users\cedric> usbipd server Logging:LogLevel:Default=Trace 'usbipd:PcapNg:Path=usbipd_debug_sandisk.pcapng' usbipd:PcapNg:SnapLength=1024
 Anyway the trace file is far greater than expected ( 350 ko), I truncated it.
Below the first 1024 bytes in usbipd_debug_sandisk_truncated.pcapng in zip file

[usbipd_debug_sandisk_truncated.zip](https://github.com/dorssel/usbipd-win/files/8930825/usbipd_debug_sandisk_truncated.zip)

dorssel, 2022-06-17:
@snowgato 
The SnapLength is *per packet*. You really need to capture all packets. I recommend a SnapLength=128 and then send the *entire* file (which is no doubt still quite big, but should be < 1MB).

And please note the privacy warning; don't send it if the drive contains personal/private/sensitive information.

snowgato, 2022-06-17:
Ok, here is the entire file
[usbipd_debug_sandisk.zip](https://github.com/dorssel/usbipd-win/files/8930892/usbipd_debug_sandisk.zip)

dorssel, 2022-06-17:
@snowgato 
Everything looks fine until the log file suddenly stops. In fact, it is a bad sign that it indeed suddenly stops, because it ends in the middle of a USB record. Do you see any errors on the `usbipd` console?

What I see from the log:

- 460GB drive, SanDisk Ultra Fit, serial 010162312029691bfa8a94f1b9ce846...
- USB 3.2Gen1
- Partition 1: bootable FAT32
- Partition 2: SanDisk Secure, or something with 'Nouveau'?
- Partition 3: (don't know, maybe this is just data within Partition 2).

All reads succeed, and then the log file is suddenly cut off. There is no indication of any errors in the log.

snowgato, 2022-06-18:
Actually,
It does work but so slowly, that it is just as if it doesn't work.
It takes around 3 to 5 minutes to list files in directory.
When I try a simple file copy, it copies 128k every minute more or less...
I've had the same problem with a samsung flash drive may it be plugged in a USB2 or USB3 port.
The following output is obtained by connecting the SanDisk UltraFit to a USB3 port:

PS C:\Users\cedric> usbipd server Logging:LogLevel:Default=Trace 'usbipd:PcapNg:Path=usbipd_debug_sandisk.pcapng' usbipd:PcapNg:SnapLength=128
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = 'usbipd_debug_sandisk.pcapng'
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:SnapLength = 128
dbug: Usbipd.Server[1000]
      2.3.0+42.Branch.master.Sha.3d9f5c5acc4e133ab8147684ad1463cbaec43240
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\cedric
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.Server[1000]
      new connection from 172.23.134.81
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.23.134.81
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: Usbipd.ConnectedClient[1]
      Client 172.23.134.81 claimed device at 1-13 (USB\VID_0781&PID_5583\010162312029691BFA8A94F1B9CE846A11A45EA90FAB7AFE2702A80D33689
3A436E200000000000000000000049D33CDFF8A2700835581073A2AAE9C).
dbug: Usbipd.AttachedClient[1000]
      Trapped SET_CONFIGURATION: 1
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_UNDERRUN -> EREMOTEIO -> -121
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      Trapped CLEAR_FEATURE: 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_UNDERRUN -> EREMOTEIO -> -121
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      Trapped CLEAR_FEATURE: 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_UNDERRUN -> EREMOTEIO -> -121
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      Trapped CLEAR_FEATURE: 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_UNDERRUN -> EREMOTEIO -> -121
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      Trapped CLEAR_FEATURE: 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_UNDERRUN -> EREMOTEIO -> -121
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      Trapped CLEAR_FEATURE: 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_UNDERRUN -> EREMOTEIO -> -121
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      Trapped CLEAR_FEATURE: 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_UNDERRUN -> EREMOTEIO -> -121
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      Trapped CLEAR_FEATURE: 129
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.PcapNg[1000]
      Flushing
[usbipd_debug_sandisk.zip](https://github.com/dorssel/usbipd-win/files/8934493/usbipd_debug_sandisk.zip)

snowgato, 2022-06-26:
Hello,
I have the same problem with a Kingston micro usb key DataTraveler 3.2 Gen 1 DTMC3G2/256GB, it is very slow to list directories in the usb key and also to copy files to the usb key

Bus 001 Device 002: ID 0951:1666 Kingston Technology DataTraveler 100 G3/G4/SE9 G2

closed 2022-07-21 by snowgato

dorssel, 2022-07-22:
@snowgato 
Were you able to solve the issue?
I've had a look again at the captures. The delay between the requests seem to be on the Linux side. This delay is what makes everything so slow. Maybe you have CONFIG_USBIP_DEBUG enabled and it is just logging too much?
That's just a guess ... I can't find anything wrong with the capture otherwise.

snowgato, 2022-07-24:
It finally works recompiling the kernel without USBIP debug messages !!
I can copy to and from USB2 or even USB3 plugged SanDisk UltraFit, it takes between 1 minute and 4 seconds to copy a 1GB file.
It also still takes 20 seconds to list for the first time the top directory in the flash drive.
Anyway it does work and that's cool !
Thanks !