dorssel/usbipd-win #410: PC crashes when using usbipd-win

d0n13, 2022-06-07:
Hi all

Just wondering how to go about debugging this. I'm using libimobiledevice on AWS and using a tunnel to communicate with usbipd-win on the pc. That is ok and working and I can retrieve data from the iPhones using it from AWS.

However, when I do a file transfer from AWS to the device using ideviceinstaller -I mobileapp.ipa it runs for a few seconds and then the whole PC blue screens BSOD.

[image]

Anyone seen this behaviour before and know how to even start to debug what might be going wrong? The fact it crashes on the PC side leads me to believe there is a issues on the driver side possibly?

Thanks
Donie

dorssel, 2022-06-07:
See https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xef--critical-process-died
Is there anything in the post-mortem Windows EventLog that tells you *which* process died?
Maybe a driver can cause this too, in which case it may be related to #248.

d0n13, 2022-06-08:
Just to note that sending a small ipa file works where the larger one causes the crash. Just fyi

dorssel, 2023-08-13:
Version 3.1.0 (https://github.com/dorssel/usbipd-win/releases/tag/v3.1.0) was released. This should fix the BSOD. Can this be confirmed?

closed 2023-09-03 by dorssel