dorssel/usbipd-win #412: Read TCP Keep Alive settings from environment variables.

kaplan2539, 2022-06-09:
Even with the latest version 2.3.0 we are seeing random connection drops.
We hope to increase stability by using a higher value for the TCP/IP keep alive time parameter.
[Cezanne's usbip-win project](https://github.com/cezanne/usbip-win) allows to configure the TCP/IP keep alive timeout via an enviornment variable (c.f. https://github.com/cezanne/usbip-win/wiki/Deployment-Tips#handling-abortive-disconnection).

I'm planning to make PR to add a similar feature to this project.

dorssel, 2022-06-09:
I've noticed your PR.
What values are you using to fix your problem? If those are better defaults than we already have, then I much rather use that than making this configurable at all.

kaplan2539, 2022-06-09:
Wow, thanks for the *fast* reply!

We are seeing the same symptoms as described in https://github.com/dorssel/usbipd-win/issues/300
The connection is closed on the Linux (WSL2) client side.
Collecting logs and captures will take me some time.

We started with winusbipd v2.1.0 and custom built kernel based on the official MS WSL2 5.4 (only adding support for usbip and FTDI usb serial adapters, no other config changes). Moving to windusbipd v.2.3.0 and upgrading our kernel to 5.10 made the problem occur less often but we still see it from time to time.

To be honest I haven't though much about the impact of making this configurable via environment variables - I've just copied the idea from https://github.com/cezanne/usbip-win/wiki/Deployment-Tips#handling-abortive-disconnection.

Regarding the default values: What we are trying to verify on our side is, weather a keepalive time-out of 30 seconds instead of 10.

kaplan2539, 2022-06-09:
Actually I found some logs (still usbip v.2.1 / Linux 5.4): 
[log-usbipd.txt](https://github.com/dorssel/usbipd-win/files/8870638/log-usbipd.txt)
[log.txt](https://github.com/dorssel/usbipd-win/files/8870641/log.txt)

No capture though...

kaplan2539, 2022-06-10:
Thank you for looking into this. So it is not the usbip client causing the connection to close, but the usbipd server on the Windows host site?

dorssel, 2022-06-10:
From this log data: yes.

dorssel, 2022-08-20:
I've just updated the KeepAlive settings slightly in PR #434 to fix Windows 2012 R2. This should not affect existing behavior on Windows 10 or newer (but you never know).

However, I think we already concluded that your issue was not caused by KeepAlive after all.

Also, your existing PR #413 is not how I prefer to see configuration settings applied (via the environment). That is hard to control for services. Configuration of service options can be done via the command line already, see https://github.com/dorssel/usbipd-win/blob/master/Usbipd/PcapNg.cs#L27 as an example.

Do you still want KeepAlive configurable?
I certainly *do* welcome contributions, so I'll leave #413 open for you to amend.

If not, can we close this issue and the PR?

kaplan2539, 2022-08-29:
Hi Frans,
I am still not 100% sure the random connections drops are cured by higher TCP keep-alive timeout values.
I am perfectly OK of not making these values configurable over environment variables and I'll close my PR.
But what do you think of increasing the values for `tcpKeepAliveRetryCount` from 5 to 10 and for `tcpKeepAliveTime` from 10 to 30?

dorssel, 2022-08-29:
@kaplan2539 
I've created a new PR #442, could you give that a try? Installer is at https://github.com/dorssel/usbipd-win/actions/runs/2951839137.

It allows you to specify `usbipd:TcpKeepAliveTime` and `usbipd:TcpKeepAliveInterval`, both in milliseconds.

TcpKeepAliveTime is the time after the last received packet to start sending keep-alives. The retry count is fixed at 10 times, to retain backward compatibility with Windows Server 2012 R2 and Windows 8.1. The interval between retries is TcpKeepAliveInterval. Effectively, the defaults will recognize a dead connection after 10.000 + 10 * 500 = 15.000 ms = 15 s.

I don't want to increase the defaults; they are mostly used to detect WSL shutdowns.

closed 2022-09-28 by dorssel

kaplan2539, 2023-02-09:
Hi and sorry for the radio silence. If I would kike to try out different `TcpKeepAliveTime` settings, I guess I need to specify them in the Windows Registry? Where exactly? in `\\HKLM\SOFTWARE\usbipd-win\TcpKeepAliveTime` and `\\HKLM\SOFTWARE\usbipd-win\TcpKeepAliveTime`? How do I know the values are getting picked up by `usbipd-win`? I can't find anything in the logs - do I need to enable debug logs and how would I do that?

Sorry for the many questions - I've searched the Wiki and Github issues, and also tried to read the source but I'm not familiar with `Microsoft.Extensions.Configuration`

kaplan2539, 2023-02-09:
Ok, I believe I figured it out by starring at the code long enough, is the following a valid example?
```usbipd server usbipd:TcpKeepAliveTime=60000 usbipd:TcpKeepAliveInterval=1000```

dorssel, 2023-02-09:
That's the correct syntax, yes. You can add ```Logging:LogLevel:Default=Trace``` to verify; the actual TCP keepalive settings are then logged at program start. See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting#logging.

kaplan2539, 2023-02-10:
Thanks Frans! We're still seeing connection drops. I'll open a new issue to discuss.

PR #442 (2022-09-28): Make TCP Keep-Alive configurable
Closes #412