dorssel/usbipd-win #414: usbip not found in Ubuntu created in Windows 10

rileyle, 2022-06-13:
I created a standard WSL2 Ubuntu VM under Windows 10. I have now upgraded to Windows 11, and I am trying to get usbipd-win and usbip working with my existing WSL2 Ubuntu. 

The usbipd-win end is working, listing devices.

In Ubuntu I have done

> sudo apt install linux-tools-virtual hwdata
> sudo update-alternatives --install /usr/local/bin/usbip usbip /usr/lib/linux-tools/*/usbip 20

and shut down / restarted the VM.

When I run usbip in Ubuntu, I get:

> WARNING: usbip not found for kernel 5.10.102.1-microsoft

I take this to mean that USB support isn't in my kernel. Would creating a fresh Ubuntu in Windows 11 fix this, or do I need to compile a kernel?

dorssel, 2022-06-13:
kernel 5.10.102.1-microsoft is fine, it has USBIP support. Depending on the type of device you want to use you may still have to compile your own kernel. The default WSL kernel has only a limited number of USB drivers built-in.

Regarding:

> WARNING: usbip not found for kernel 5.10.102.1-microsoft

What is the output of (executing in Windows): `wsl --list --verbose`?

rileyle, 2022-06-13:
> PS C:\Users\lriley> wsl --list --verbose
>   NAME      STATE           VERSION
> * Ubuntu    Stopped         2

To clarify, I created the VM as an Ubuntu WSL2 machine under Windows 10. When I upgraded to Windows 11, I installed WSL from the Windows Store (for WSLg functionality).

rileyle, 2022-06-13:
Also, `usbipd wsl list` gives a warning:

> usbipd: warning: Unknown USB filter 'tdevflt' may be incompatible with this software; 'bind --force' may be required.

dorssel, 2022-06-13:
That filter appears to be from Cortex XDR. It just means that dynamically attaching will probably not work. But there is a workaround using `usbipd bind --force` for that. It is an unrelated issue.

closed 2022-06-13 by dorssel