dorssel/usbipd-win #418: usb stick write very slow with usbipd

snowgato, 2022-07-01:
Hello,
I am using wsl2 with compiled kernel including usb specific options set.
I can attach usb sticks, but it is so slow to list directories and copy files.
With small files, it's ok, but with bigger files from hundred Mb to Gb, it last for several minutes to succeed or it even fails.
I have the same problem with sanDisk UltraFit and newly bought
Kingston DataTraveler 100 G3- DT100G3/218GB USB 3.0 128 GB  
For Kingston usb stick,  lsusb:
Bus 001 Device 002: ID 0951:1666 Kingston Technology DataTraveler 100 G3/G4/SE9 G2
Kingston Data Traveler 100 is listed in the supported devices
Is it possible that usbipd doesnt support the latest usb keys ?
Any help to solve this would be greatly appreciated,
Thanks !

closed 2022-07-21 by snowgato

dorssel, 2022-07-22:
Did you try this in a USB2 port? Does that help?

snowgato, 2022-07-24:
It finally works recompiling the kernel without USBIP debug messages.
I can copy to and from USB2 or even USB3 plugged SanDisk UltraFit, it takes between 1 minute and 4 seconds to copy a 1GB file. 
It also still takes 20 seconds to list for the first time the top directory in the flash drive.
Anyway it does work and that's cool !
Thanks !