dorssel/usbipd-win #419: Failed to build my own kernel for USB3 Vision Camera

sszxc, 2022-07-05:
Hello,

Thanks for the usbipd-win firstly.

I want to connect my USB3 Vision Camera to WSL2, so I followed the instructions to build my own kernel. 

Here is the problem. When I send `sudo make -j 8`, the output:

```
  STUBCPY drivers/firmware/efi/libstub/skip_spaces.stub.o
  STUBCPY drivers/firmware/efi/libstub/tpm.stub.o
  STUBCPY drivers/firmware/efi/libstub/vsprintf.stub.o
  STUBCPY drivers/firmware/efi/libstub/x86-stub.stub.o
  AR      drivers/firmware/efi/libstub/lib.a
  AR      drivers/firmware/efi/built-in.a
  CC      drivers/vhost/iotlb.o
  CC      drivers/hid/hid-generic.o
drivers/hv/dxgkrnl/dxgmodule.c:392:39: error: initializer element is not constant
 const int DXGK_VMBUS_VERSION_OFFSET = DXGK_VMBUS_CHANNEL_ID_OFFSET +
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/hv/dxgkrnl/dxgmodule.c:395:41: error: initializer element is not constant
 const int DXGK_VMBUS_VGPU_LUID_OFFSET = DXGK_VMBUS_VERSION_OFFSET +
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~
drivers/hv/dxgkrnl/dxgmodule.c:398:41: error: initializer element is not constant
 const int DXGK_VMBUS_GUESTCAPS_OFFSET = DXGK_VMBUS_VERSION_OFFSET +
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:280: recipe for target 'drivers/hv/dxgkrnl/dxgmodule.o' failed
make[3]: *** [drivers/hv/dxgkrnl/dxgmodule.o] Error 1
scripts/Makefile.build:497: recipe for target 'drivers/hv/dxgkrnl' failed
make[2]: *** [drivers/hv/dxgkrnl] Error 2
scripts/Makefile.build:497: recipe for target 'drivers/hv' failed
make[1]: *** [drivers/hv] Error 2
make[1]: *** Waiting for unfinished jobs....
  CC      drivers/firmware/memmap.o
  AR      drivers/platform/x86/built-in.a
  AR      drivers/platform/built-in.a
  AR      drivers/hid/built-in.a
  AR      drivers/firmware/built-in.a
  AR      drivers/vhost/built-in.a
Makefile:1822: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

I've tried different configurations but all end up with different errors. Maybe it's because I have added some unnecessary modules? Can you tell me which ones are required for USB3 Vision Camera?

BTW, I'm running WSL2, Ubuntu18.04, 5.10.60.1-microsoft-standard-WSL2.

Thanks for your help in advance.

sszxc, 2022-07-05:
Sorry to bother, from [here](https://github.com/microsoft/WSL/issues/7558) I found that the above problem has been fixed by newer version of the kernel. Now I can successfully build my own kernel. But I'm still confused how to determine which modules are necessary...

sszxc, 2022-07-05:
Specifically, when I send `attach`, I can hear the sound of the USB device disconnecting, the device disappear from the list, and a few seconds later there is the sound of the USB device plugging in.

dorssel, 2022-07-05:
Have a look at https://github.com/dorssel/usbipd-win/wiki/Troubleshooting. It may give more info on why the immediate detach happens. Also check `/var/log/syslog` on the Linux side of things (you may have to install/start `rsyslog` for that, though).

dorssel, 2022-08-23:
@sszxc 
Any news?

dorssel, 2022-09-28:
Closing due to lack of activity.

closed 2022-09-28 by dorssel