dorssel/usbipd-win #422: vhci_driver error on WSL 5.10.102.1

skeptical, 2022-07-13:
Hi, no joy even though my setup seems to conform to the stated requirements.  I've read through the wiki and various related issues and seem to be getting the same attach-detach behaviour as #323, but using current usbpid-win 2.3.0.

Using ```wsl attach --auto-attach``` or ```--hardware-id``` fails similarly. Will attempt to provide a _usb-capture_ log if that would help.

On Win10:
```
C:\Users\Administrator>systeminfo
...
OS Name:                   Microsoft Windows 10 Home
OS Version:                10.0.19043 N/A Build 19043
...

C:\Users\Administrator>wsl --list -v
  NAME      STATE           VERSION
* Debian    Running         2

C:\Users\Administrator>usbipd -h
usbipd-win 2.3.0
...

C:\Users\Administrator>usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
1-5    0461:4d8a  USB Input Device                                              Not attached
1-6    0461:4d0f  USB Input Device                                              Not attached
1-11   0bda:0151  USB Mass Storage Device                                       Not attached
2-5    0955:2222  USB Mass Storage Device                                       Not attached

C:\Users\Administrator>usbipd wsl attach --busid 1-11
libusbip: error: udev_device_new_from_subsystem_sysname failed
usbip: error: open vhci_driver
usbip: error: query
usbipd: error: Failed to attach device with busid '1-11'.
```
on the linux side:
```sh
# installed on debian 10 using 'sudo apt-get install usbip hwdata'

:~$ uname -a
Linux DESKTOP-....... 5.10.102.1-microsoft-standard-WSL2 #1 SMP Wed Mar 2 00:30:59 UTC 2022 x86_64 GNU/Linux

:~$ sudo usbipd --version
usbipd (usbip-utils 2.0)

:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

# kernel appears to be configured for usbip:
:~$ cp /proc/config.gz config.gz
:~$ gunzip config.gz
:~$ cat config | grep -E ".+USB.*="
CONFIG_USB_NET_DRIVERS=y
CONFIG_USB_USBNET=y
CONFIG_USB_NET_CDCETHER=y
CONFIG_USB_NET_CDC_NCM=y
CONFIG_USB_HID=y
CONFIG_USB_OHCI_LITTLE_ENDIAN=y
CONFIG_USB_SUPPORT=y
CONFIG_USB_COMMON=y
CONFIG_USB_ARCH_HAS_HCD=y
CONFIG_USB=y
CONFIG_USB_ANNOUNCE_NEW_DEVICES=y
CONFIG_USB_DEFAULT_PERSIST=y
CONFIG_USB_AUTOSUSPEND_DELAY=2
CONFIG_USB_ACM=y
CONFIG_USBIP_CORE=y
CONFIG_USBIP_VHCI_HCD=y
CONFIG_USBIP_VHCI_HC_PORTS=8
CONFIG_USBIP_VHCI_NR_HCS=1
CONFIG_USB_SERIAL=y
CONFIG_USB_SERIAL_CH341=y
CONFIG_USB_SERIAL_CP210X=y
CONFIG_USB_SERIAL_FTDI_SIO=y
```
and the usbipd trace log:
```log
C:\Users\Administrator>usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = ''
dbug: Usbipd.Server[1000]
      2.3.0+42.Branch.master.Sha.3d9f5c5acc4e133ab8147684ad1463cbaec43240
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\Administrator
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.Server[1000]
      new connection from 172.22.224.213
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.22.224.213
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: Usbipd.ConnectedClient[1]
      Client 172.22.224.213 claimed device at 1-11 (USB\VID_0BDA&PID_0151\20060413092100000).
info: Usbipd.ConnectedClient[2]
      Client 172.22.224.213 released device at 1-11 (USB\VID_0BDA&PID_0151\20060413092100000).
dbug: Usbipd.ConnectedClient[1000]
      Unbind or unplug while attached
dbug: Usbipd.Server[1000]
      connection closed
```

dorssel, 2022-07-13:
Looks like your kernel supports USBIP alright, and it is compiled in statically (not as a module); this looks good.
Which package+version of the Linux-side `usbip` tool did you install? It looks like a version mismatch there...

What does (from Windows) `wsl -u root usbip version` say?

skeptical, 2022-07-14:
Great, there's hope then...

package:
```sh
:~$ dpkg -s usbip
...
Source: linux (4.9.320-2)
Version: 2.0+4.9.320-2
Depends: usbutils, libc6 (>= 2.7), libudev1 (>= 183), libwrap0 (>= 7.6-4~)
```
usbip shows the same usb-utils version string:
```
C:\Users\Administrator>wsl -u root usbip version
usbip (usbip-utils 2.0)
```

dorssel, 2022-07-14:
The `usbip` is too old (4.9).
Look similar to https://github.com/dorssel/usbipd-win/issues/376#issuecomment-1122832383.

What distro are you using?

E.g., Ubuntu 18.04 comes with a default 4.x kernel, and therefore the default package that includes `usbip` also is from a 4.x kernel. However, WSL always installs a 5.x kernel, and therefore you also need a 5.x `usbip`. For Ubuntu 18.04, it can be solved by installing a "hwe" version of the tools, which is in fact compiled against a 5.x kernel. If your distro does not have such a thing, you can always compile your own `usbip` as explained in https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel.

skeptical, 2022-07-15:
Right, it began to slowly dawn on me too after actually reading the "4.9.320" in my last post :-P

I had installed debian using ```wsl --install -d debian``` (which for some reason installs debian 9 with old kernel) instead of installing debian through the MS app store, which gives you debian 11 with new kernel (why the inconsistency? who knows, but issue is discussed [here](https://superuser.com/a/1722975)).

After fixing this, things look better
```sh
:~$ dpkg -s usbip
...
Source: linux (5.10.127-1)
Version: 2.0+5.10.127-1
...
```
and attaching is as advertised...
```
C:\Users\Administrator>usbipd wsl attach --busid 1-11
C:\Users\Administrator>usbipd wsl attach --busid 2-5
C:\Users\Administrator>usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
1-5    0461:4d8a  USB Input Device                                              Not attached
1-6    0461:4d0f  USB Input Device                                              Not attached
1-11   0bda:0151  USB Mass Storage Device                                       Attached - Debian
2-5    0955:2222  USB Mass Storage Device                                       Attached - Debian
```
in linux
```sh
:~$ sudo usbip port
Imported USB devices
====================
Port 00: <Port in Use> at High Speed(480Mbps)
       Realtek Semiconductor Corp. : Mass Storage Device (Multicard Reader) (0bda:0151)
       1-1 -> usbip://172.25.128.1:3240/1-11
           -> remote bus/dev 001/011
Port 01: <Port in Use> at High Speed(480Mbps)
       Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) : Flash Drive (090c:1000)
       1-2 -> usbip://172.25.128.1:3240/2-2
           -> remote bus/dev 002/002
Port 08: <Port in Use> at Super Speed(5000Mbps)
       NVIDIA Corp. : unknown product (0955:2222)
       2-1 -> usbip://172.25.128.1:3240/2-5
           -> remote bus/dev 002/005
```
but I must be missing something because nowhere in linux do I see any device to access (/dev/sd? etc). Is there an additional mounting step that has to happen?

dorssel, 2022-07-15:
Now you're running into #352. You need to compile in USB storage support.

closed 2022-07-15 by skeptical